cesanta/mongoose #1961: mqtt broker ，when more than 1018 clients report an error

zhousongsong, 2023-01-13:
- My goal is: _describe_
- My actions were: _describe_
- My expectation was: _describe_
- The result I saw: _provide logs or description_
- My question is:  mqtt broker ，when more than 1018 clients report an error
```
1ecbef 1 mongoose.c:4359:accept_conn    1 accept failed, errno 24
1ecbef 1 mongoose.c:4359:accept_conn    1 accept failed, errno 24
1ecbef 1 mongoose.c:4359:accept_conn    1 accept failed, errno 24
1ecbef 1 mongoose.c:4359:accept_conn    1 accept failed, errno 24
1ecbef 1 mongoose.c:4359:accept_conn    1 accept failed, errno 24
1ecbf0 1 mongoose.c:4359:accept_conn    1 accept failed, errno 24
1ecbf0 1 mongoose.c:4359:accept_conn    1 accept failed, errno 24
1ecbf0 1 mongoose.c:4359:accept_conn    1 accept failed, errno 24
1ecbf0 1 mongoose.c:4359:accept_conn    1 accept failed, errno 24
1ecbf0 1 mongoose.c:4359:accept_conn    1 accept failed, errno 24
1ecbf0 1 mongoose.c:4359:accept_conn    1 accept failed, errno 24
^C1ecbf0 1 mongoose.c:4359:accept_conn    1 accept failed, errno 24
1ecbf0 1 mongoose.c:4359:accept_conn    1 accept failed, errno 24
MG_EV_CLOSE... 
unsub: /js/test/1018
```

# Environment

- mongoose version: 7.8
- Compiler/IDE and SDK: Qt version 5.12
- Target hardware/board: 
- Connectivity chip/module: 
- Target RTOS/OS (if applicable): ubuntu22.04

zhousongsong, 2023-01-13:
```
#ifndef MG_ENABLE_POLL
#define MG_ENABLE_POLL 1
#endif

#ifndef MG_ENABLE_EPOLL
#define MG_ENABLE_EPOLL 1
#endif
```

zhousongsong, 2023-01-13:
linux system cmd
```
 ulimit -n 10240
```

closed 2023-01-13 by zhousongsong

scaprile, 2023-01-13:
Mongoose is not an MQTT broker, it is a Network Library and what you are doing is to evaluate an example of using its API in your platform.