cesanta/mongoose #1962: On Windows 10, the maximum number of connections to the mqtt server is 64. I need a larger number of connections

zhousongsong, 2023-01-14:
- My goal is: _describe_
- My actions were: _describe_
- My expectation was: _describe_
- The result I saw: _provide logs or description_
- My question is:  On Windows 10, the maximum number of connections to the mqtt server is 64. I need a larger number of connections
```
mqtt connected ... 
sub topic: "/js/0060/response/*"
mqtt connected ... 
sub topic: "/js/0061/response/*"
mqtt connected ... 
sub topic: "/js/0062/response/*"
mqtt connected ... 
sub topic: "/js/0063/response/*"
mqtt connected ... 
sub topic: "/js/0064/response/*"

340504 2 mqtt_broker.cpp:193:mqtt_handl PINGREQ 0x324 -> PINGRESP
```

# Environment

- mongoose version: 7.8
- Compiler/IDE and SDK: minGw 64
- Target hardware/board: _?_
- Connectivity chip/module: _?_
- Target RTOS/OS (if applicable): windows 10

scaprile, 2023-01-14:
Mongoose is a Network Library and what you are doing is to evaluate an example of using its API in your platform.
Who is not allowing more connections ? Mongoose ? Your OS ? Do the event handler receive the connection request or is your OS rejecting it ? Did you sniff your network ? Does TCP connections establish ? Did you enable debugging and/or verbose logging in Mongoose ?
How are you using the software, there is no cpp code provided by Cesanta (nor a file named "mqtt_broker") ? What you paste is not a Mongoose log.

zhousongsong, 2023-01-16:
What is the network model of Mongoose in Windows？
select?

zhousongsong, 2023-01-16:
mqtt_broker.cpp
```
#include "mqtt_broker.h"
#include <QDebug>
#include <thread>
#include <functional>
#include <iostream>
#include <list>

//#define DEFAULT_LISTEN_ON	"mqtt://0.0.0.0:18083"
#define DEFAULT_LISTEN_ON	"mqtt://0.0.0.0:1885"
//#define DEFAULT_LISTEN_ON	"mqtt://192.168.16.243:1883"
struct sub {
    uint8_t qos;
    struct sub *next;
    struct mg_connection *c;
    struct mg_str topic;
};

struct mg_mgr mgr;
static bool bExitThread = false;
static struct sub *s_subs = NULL;

CMqttBroker *CMqttBroker::m_pMqttBorker = nullptr;

static void signal_handler(int signo) {
    std::ignore = signo; //忽略未使用变量
    bExitThread = true;
    exit(0);
}

CMqttBroker::CMqttBroker()
{
    m_pMqttBorker = this;
    connect(this, &CMqttBroker::sendMqttResponseStatic, [this](QString topic,QString value)
    {
         emit update_mqtt_response_slots(topic,value);
    });

    bExitThread = false;
    signal(SIGINT, signal_handler);   // Setup signal handlers - exist event
    signal(SIGTERM, signal_handler);  // manager loop on SIGINT and SIGTERM
    qDebug() << "mqtt listen on :" << DEFAULT_LISTEN_ON;
    std::thread th(std::bind(&CMqttBroker::mqtt_thread_function,this));
    th.detach();

}

CMqttBroker::~CMqttBroker()
{
    bExitThread = true;
    while (bExitThread) {
        std::this_thread::sleep_for(std::chrono::milliseconds (1));
    }
}

static size_t mg_mqtt_next_topic(struct mg_mqtt_message *msg,
                                 struct mg_str *topic, uint8_t *qos,
                                 size_t pos) {
  unsigned char *buf = (unsigned char *) msg->dgram.ptr + pos;
  size_t new_pos;
  if (pos >= msg->dgram.len) return 0;

  topic->len = (size_t) (((unsigned) buf[0]) << 8 | buf[1]);
  topic->ptr = (char *) buf + 2;
  new_pos = pos + 2 + topic->len + (qos == NULL ? 0 : 1);
  if ((size_t) new_pos > msg->dgram.len) return 0;
  if (qos != NULL) *qos = buf[2 + topic->len];
  return new_pos;
}
[... 644 lines cut ...]
cec7da 3 mongoose.c:4258:write_conn     59 0x3d4 snd 5/2048 rcv 0/2048 n=5 err=0
cec7da 3 mongoose.c:4387:accept_conn    60 0x3d8 accepted c0a8105b.37480 -> 0.1885
cec7e9 3 mongoose.c:4247:read_conn      60 0x3d8 snd 0/0 rcv 0/2048 n=56 err=0
cec7e9 3 mongoose.c:4258:write_conn     60 0x3d8 snd 4/2048 rcv 0/2048 n=4 err=0
cec7e9 3 mongoose.c:4247:read_conn      60 0x3d8 snd 0/2048 rcv 0/2048 n=26 err=0
cec7e9 3 mongoose.c:4258:write_conn     60 0x3d8 snd 5/2048 rcv 0/2048 n=5 err=0
cec7e9 3 mongoose.c:4387:accept_conn    61 0x3dc accepted c0a8105b.37494 -> 0.1885
cec7e9 3 mongoose.c:4247:read_conn      61 0x3dc snd 0/0 rcv 0/2048 n=56 err=0
cec7e9 3 mongoose.c:4258:write_conn     61 0x3dc snd 4/2048 rcv 0/2048 n=4 err=0
cec7e9 3 mongoose.c:4247:read_conn      61 0x3dc snd 0/2048 rcv 0/2048 n=26 err=0
cec7e9 3 mongoose.c:4258:write_conn     61 0x3dc snd 5/2048 rcv 0/2048 n=5 err=0
cec7f9 3 mongoose.c:4387:accept_conn    62 0x3e0 accepted c0a8105b.37498 -> 0.1885
cec7f9 3 mongoose.c:4247:read_conn      62 0x3e0 snd 0/0 rcv 0/2048 n=56 err=0
cec7f9 3 mongoose.c:4258:wMG_EV_OPEN...  62
mqtt connected ... 
sub topic: "/js/0062/response/*"
MG_EV_OPEN...  63
mqtt connected ... 
MG_EV_OPEN...  64
mqtt connected ... 
sub topic: "/js/0063/response/*"
sub topic: "/js/0064/response/*"
```

```

scaprile, 2023-01-16:
> What is the network model of Mongoose in Windows？ select?

The one you choose with your compile options, that you don't say. It defaults to select() if you don't specify otherwise.

I'm too lazy to dig for something in such a long piece of code. If you think you've found an issue then attach the smallest piece of code to reproduce your issue so we can run it and test. _Preferably run our example_ and attack a network capture (Wireshark)

I'm sorry but I can't keep my attention focused beyond line 10 in such a long log. Can you please explain ?
I think I see MG_EV_OPEN events fired so connections establish. There seems to be no data beyond 62 connections. I can't see what happens beyond 64.

Please answer what has been asked to you:

> Who is not allowing more connections ? Mongoose ? Your OS ? Do the event handler receive the connection request or is your OS rejecting it ? Did you sniff your network ? Does TCP connections establish ?

zhousongsong, 2023-01-19:
Maximum Number of Sockets Supported

[https://learn.microsoft.com/en-us/windows/win32/winsock/maximum-number-of-sockets-supported-2](url)

cpq, 2023-01-19:
@zhousongsong try `-DMG_ENABLE_POLL=1`

cpq, 2023-01-29:
@zhousongsong did you try to use poll , as suggested above?

zhousongsong, 2023-01-30:
@cpq 

mongoose.h
```c
#ifndef MG_ENABLE_POLL
#define MG_ENABLE_POLL 1
#endif
```

Compilation failed on windows

` error: invalid use of undefined type 'struct pollfd'`

cpq, 2023-01-31:
Building unit test with VS2017 and POLL support works:

```
cl.exe mongoose.c test/unit_test.c test/packed_fs.c /nologo /W3 /O2 /MD /I. -DMG_MAX_HTTP_HEADERS=7 -DMG_ENABLE_LINES -DMG_ENABLE_PACKED_FS=1 -DMG_ENABLE_SSI=1 -DMG_ENABLE_POLL=1 ws2_32.lib /Fevc17.exe
```

zhousongsong, 2023-02-01:
@cpq 
thank you

closed 2023-02-01 by zhousongsong