cesanta/mongoose #1963: Multithreaded example never unregisters event handler

TeknoVenus, 2023-01-14:
The multithreaded example here: https://github.com/cesanta/mongoose/tree/master/examples/multi-threaded appears to continue to call the pipe callback function every poll even after the parent HTTP connection has closed. 

**Steps to Reproduce**

We can see this by adding some extra logging:

```diff
diff --git a/examples/multi-threaded/main.c b/examples/multi-threaded/main.c
index 2b7913f7..abdb1401 100644
--- a/examples/multi-threaded/main.c
+++ b/examples/multi-threaded/main.c
@@ -57,14 +57,17 @@ static void pcb(struct mg_connection *c, int ev, void *ev_data, void *fn_data) {
   struct mg_connection *parent = (struct mg_connection *) fn_data;
   MG_INFO(("%lu %p %d %p", c->id, c->fd, ev, parent));
   if (parent == NULL) {  // If parent connection closed, close too
+    MG_INFO(("parent is null"));
     c->is_closing = 1;
   } else if (ev == MG_EV_READ) {  // Got data from the worker thread
     mg_http_reply(parent, 200, "Host: foo.com\r\n", "%.*s\n", c->recv.len,
                   c->recv.buf);  // Respond!
     c->recv.len = 0;             // Tell Mongoose we've consumed data
   } else if (ev == MG_EV_OPEN) {
+    MG_INFO(("socket open"));
     link_conns(c, parent);
   } else if (ev == MG_EV_CLOSE) {
+    MG_INFO(("socket close"));
     unlink_conns(c, parent);
   }
 }
```

After running the example, make a curl request to `http://localhost:8000/test`. 

**Actual Behaviour**

As expected, the pipe callback starts firing every poll. However, once the HTTP connection is complete, neither the the `socket close` or `parent is null` messages are printed, and we can see the entry log message to the `pcb` function is printed forever

```
b424c1 3 mongoose.c:3493:mg_listen      1 0x4 http://localhost:8000
b44914 3 mongoose.c:4377:accept_conn    2 0x5 accepted 127.0.0.1.42014 -> 127.0.0.1.8000
b44914 3 mongoose.c:4233:read_conn      2 0x5 snd 0/0 rcv 0/2048 n=88 err=0
b44914 2 main.c:58:pcb                  3 0x8 1 0x55b2aeb217a0
b44914 2 main.c:67:pcb                  socket open
b44914 3 mongoose.c:4437:mg_mkpipe      3 0x8 pipe 7
b44cfd 2 main.c:58:pcb                  3 0x8 2 0x55b2aeb217a0
b450e4 2 main.c:58:pcb                  3 0x8 2 0x55b2aeb217a0
b450e4 3 mongoose.c:4233:read_conn      3 0x8 snd 0/0 rcv 0/2048 n=2 err=0
b450e4 2 main.c:58:pcb                  3 0x8 7 0x55b2aeb217a0
b450e5 2 main.c:58:pcb                  3 0x8 2 0x55b2aeb217a0
b450e5 3 mongoose.c:4244:write_conn     2 0x5 snd 65/2048 rcv 0/2048 n=65 err=0
b454ce 2 main.c:58:pcb                  3 0x8 2 0x55b2aeb217a0
b458b7 2 main.c:58:pcb                  3 0x8 2 0x55b2aeb217a0
b45ca0 2 main.c:58:pcb                  3 0x8 2 0x55b2aeb217a0
b46089 2 main.c:58:pcb                  3 0x8 2 0x55b2aeb217a0
b46472 2 main.c:58:pcb                  3 0x8 2 0x55b2aeb217a0
b4685b 2 main.c:58:pcb                  3 0x8 2 0x55b2aeb217a0
b46c44 2 main.c:58:pcb                  3 0x8 2 0x55b2aeb217a0
b4702d 2 main.c:58:pcb                  3 0x8 2 0x55b2aeb217a0
```

If we repeatedly make HTTP requests, we end up registering a new pcb callback each time, and never clean up the old ones.

**Expected Behaviour**

Once the HTTP connection is closed, I would expect to see the pipe socket closed and the pcb to stop being called every time the mongoose event loop polls

# Environment

- mongoose version: 7.8
- Tested on Windows 10 with MinGW (gcc 12.2) and Ubuntu 20.04 (gcc 9.4)

scaprile, 2023-01-15:
WRT the socket being closed, it should; I don't see that in your log.
Please note that every Mongoose event handler is _always_ called on every loop, it is _polled_ and you get an MG_EV_POLL event, so what you send is normal behavior.
We'll check your report on sockets not closing.

TeknoVenus, 2023-01-15:
Thanks for taking a look at this, really appreciate it.

I've been doing some more testing, and I think this might be a sign of a wider Windows issue beyond this example. It appears on Windows, open connections to mongoose never close properly (no `FIN` message is seen).

I appreciate Windows environments are not your priority, and understand if you're not able to offer much help. But I'll document for prosperity and in the hope you might have an idea!

To test this and prevent any extra confusion from threading, I've modified the `examples/http-server/main.c` sample with the following change to add a timer that prints out the number of open connections every one second for debug

```diff
diff --git a/examples/http-server/main.c b/examples/http-server/main.c
index f2efbc5f..8a30da57 100644
--- a/examples/http-server/main.c
+++ b/examples/http-server/main.c
@@ -36,6 +36,16 @@ static void cb(struct mg_connection *c, int ev, void *ev_data, void *fn_data) {
   (void) fn_data;
 }
 
+static void timer_fn(void *arg) {
+  struct mg_mgr *mgr = (struct mg_mgr *) arg;
+  int i = 0;
+  for (struct mg_connection *c = mgr->conns; c != NULL; c = c->next) {
+    i++;
+  }
+  MG_INFO(("Currently %d open connections", i));
+}
+
+
 static void usage(const char *prog) {
   fprintf(stderr,
           "Mongoose v.%s\n"
@@ -85,6 +95,7 @@ int main(int argc, char *argv[]) {
   signal(SIGTERM, signal_handler);
   mg_log_set(s_debug_level);
   mg_mgr_init(&mgr);
+  mg_timer_add(&mgr, 1000, MG_TIMER_REPEAT, timer_fn, &mgr);  // Init timer
   if ((c = mg_http_listen(&mgr, s_listening_address, cb, &mgr)) == NULL) {
     MG_ERROR(("Cannot listen on %s. Use http://ADDR:PORT or :PORT",
               s_listening_address));

```

**Happy-day scenario - Linux**

Testing on Ubuntu 20.04 with gcc 9, and all HTTP traffic is contained within this environment. Everything works as it should!

* Start the modified example
* Make multiple curl requests to `localhost:8080`. Note the current connection count remains at 1
* Looking at wireshark, I can see `FIN` messages after the HTTP request

```
./example
2779a4 2 main.c:107:main                Mongoose version : v7.8
2779a4 2 main.c:108:main                Listening on     : http://0.0.0.0:8000
2779a4 2 main.c:109:main                Web root         : [/home/stephen/mongoose/examples/http-server]
278176 2 main.c:45:timer_fn             Currently 1 open connections
27855f 2 main.c:45:timer_fn             Currently 1 open connections
27866d 2 main.c:32:cb                   GET /test 404 0000000010
278a56 2 main.c:45:timer_fn             Currently 1 open connections
278e0e 2 main.c:45:timer_fn             Currently 1 open connections
278e0e 2 main.c:32:cb                   GET /test 404 0000000010
2791f7 2 main.c:45:timer_fn             Currently 1 open connections
^C279476 2 main.c:112:main                Exiting on signal 2
```

```
24	7.312380	127.0.0.1	127.0.0.1	HTTP	144	GET / HTTP/1.1 
25	7.312383	127.0.0.1	127.0.0.1	TCP	66	8000 → 36848 [ACK] Seq=1 Ack=79 Win=65408 Len=0 TSval=2344248001 TSecr=2344248001
26	7.312537	127.0.0.1	127.0.0.1	HTTP	2097	HTTP/1.1 200 OK  (text/html)
27	7.312560	127.0.0.1	127.0.0.1	TCP	66	36848 → 8000 [ACK] Seq=79 Ack=2032 Win=64000 Len=0 TSval=2344248001 TSecr=2344248001
28	7.312668	127.0.0.1	127.0.0.1	TCP	66	36848 → 8000 [FIN, ACK] Seq=79 Ack=2032 Win=65536 Len=0 TSval=2344248001 TSecr=2344248001
[... 31 lines cut ...]

```
32	6.213427	127.0.0.1	127.0.0.1	HTTP	178	GET / HTTP/1.1 
33	6.213452	127.0.0.1	127.0.0.1	TCP	44	8000 → 51792 [ACK] Seq=1 Ack=135 Win=2161152 Len=0
34	6.215525	127.0.0.1	127.0.0.1	HTTP	1966	HTTP/1.1 200 OK  (text/html)
35	6.215544	127.0.0.1	127.0.0.1	TCP	44	51792 → 8000 [ACK] Seq=135 Ack=1923 Win=2159360 Len=0
```

Also tested sending HTTP requests from Chrome and Firefox, and the same behaviour is seen.

**Non-Happy-Day Scenario 2 - Windows (WSL)**

Using WSL, I start an Ubuntu 20.04 session to run Mongoose. I then issue HTTP requests from Windows

* Build and run the example under an Ubuntu WSL session on Windows
* Open a PowerShell window and issue an `Invoke-WebRequest` command
* Observe that for each Invoke-WebRequest command, the open connection count increases

Logs and wireshark capture look identical as in the previous scenario - no `FIN` messages seen

**Extra Test - libmicrohttpd**

For completeness, I tried the libmicrohttpd HTTP server library in the same scenarios as I tested Mongoose.

libmicrohttpd behaves the same in all 3 tests, and I can see `FIN` messages being exchanged correctly in all tests. This seems to imply a potential Mongoose issue.

scaprile, 2023-01-16:
1- Please don't paste traffic captures, I need to see what you send and what you don't. Please attach a zipped file. The client is the one to close the connection, _are you sending the correct content-length_ ? If you don't, the client will wait for more data and don't close the connection.
2- In case you are, what is your compiler on Windows ? MinGW ? What threading model ? If POSIX, did you remove the WIN32 #defines ?

TeknoVenus, 2023-01-16:
Hi,

Apologies, I think I've found the issue. By default, Windows is maintaining a TCP connection pool, and Invoke-WebRequest from PowerShell uses this pool. 

Therefore, even after Mongoose returns the response, Windows is keeping this TCP connection alive (hence not sending FIN).

Can also observe this if making request to Mongoose from Android, and from Firefox/Chrome on both Windows and Linux (which maintain their own connection pools)

E.G

* Build http-server example on Ubuntu 20.04
* Apply patch in previous comment to print the number of open connections
* Make request from Firefox
* Observe connection count remains at 2 even after HTTP response sent from Mongoose
* In Firefox, check `about:networking#sockets` - you'll see the 127.0.0.01 socket remains open
* After a few minutes, Firefox will close the socket and the connection count in Mongoose will drop again

So seems this is behaving as expected.

Thanks again for the help you've given me.