cesanta/mongoose #1965: [MIP stack] Bad TCP ACK handling on HTTP file upload

psykokwak-com, 2023-01-16:
I try to implement a HTTP upload file but I have an error on MIP TCP ACK flag that corrupt the uploaded file.

Following this code sample : 
```
#include "driver_teensy41.h"
#include "mongoose.h"

struct mg_mgr mgr;  // Mongoose event manager
struct mip_if mif;  // MIP network interface

static void handler(struct mg_connection *c, int ev, void *ev_data,
                    void *fn_data) {
  struct mg_http_message *hm = (struct mg_http_message *) ev_data;

  switch (ev) {
    case MG_EV_HTTP_MSG:
      // HANDLE /system/upgrade
      if (mg_http_match_uri(hm, "/")) {
        if (!mg_strcmp(hm->method, mg_str("GET"))) {
          const char *txt =
              "<form method=\"POST\" enctype=\"multipart/form-data\">"
              "<input type=\"file\" name=\"file\" />"
              "<input type=\"submit\" value=\"Upload\" />"
              "</form>";
          mg_http_reply(c, 200, "Content-Type: text/html\r\n", txt);
        }
      }
      break;

    case MG_EV_HTTP_CHUNK:
      // HANDLE /system/upgrade
      if (mg_http_match_uri(hm, "/")) {
        if (!mg_strcmp(hm->method, mg_str("POST"))) {
          MG_INFO(("Got chunk len %lu", (unsigned long) hm->chunk.len));
          // LOGF("Query string: [%.*s]\n", (int)hm->query.len, hm->query.ptr);
          // LOGF("Chunk data:\n%.*s", (int)hm->chunk.len, hm->chunk.ptr);
          // Do something here

          mg_http_delete_chunk(c, hm);
          if (hm->chunk.len == 0) {
            MG_INFO(("Last chunk received, sending response"));
            mg_http_reply(c, 200, "", "Complete.\n");
          }
        }
      }
      break;
  }
}

void setup() {
  Serial.begin(115200);

  // Set logging function to a serial print
  mg_log_set_fn([](char ch, void *) { Serial.print(ch); }, NULL);
  mg_log_set(MG_LL_INFO);
  mg_mgr_init(&mgr);

  MG_INFO(("Starting TCP/IP stack..."));

  mif.driver = &mip_driver_teensy41;

  //IPAddress ip(192, 168, 1, 120);
  //IPAddress mask(255, 255, 255, 0);
  mif.enable_dhcp_client = true;
  mif.enable_dhcp_server = false;

  // mif.ip = ip;
  // mif.mask = mask;

  mip_init(&mgr, &mif);

  mg_http_listen(&mgr, "http://0.0.0.0", handler, NULL);
}

void loop() {
  mg_mgr_poll(&mgr, 1);
}
```

During the upload process, On the console, I have some : 
> 2ef67 1 mongoose.c:403:mg_error         7 0x0 SEQ != ACK: 8d449589 8d449371
> 2ef67 2 teensy41.ino:215:handler        Got chunk len 0
> 2ef67 2 teensy41.ino:221:handler        Last chunk received. totsize=11256, 

That resets the connection (and corrupt the file).

Wireshark shows me some inconsistant packets from the device : 
[image]

scaprile, 2023-01-16:
- Please honor the issue template
- Please run one of the examples, your code is too long and we can't afford the time to dig through it
- Please describe your scenario, who is 229 ? who is 120 ? If I guess correctly the device is 120
- Please don't paste pictures, attach capture files, we need to see what is actually going on from the start. In frame 644 the device is sending FIN and the client keeps sending data. That is consistent with the message indicating the end of the expected chunks. What happened before that ? Where did you take the capture, at 229 ?
- Start a capture with a blank Wireshark buffer, otherwise it may report anomalies based on prior traffic
- Please attach your complete Mongoose log for the very same network capture, so we can actually see the same sequence numbers
- If you have a way to check your free memory, please do
- MIP does not handle selective acknowledgements nor retransmissions

psykokwak-com, 2023-01-16:
I added a "while(42) ;" just before the "mg_error(c, "SEQ != ACK: %x %x", seq, s->ack);" to block the program and see on wireshark.
You can download this wireshark dump from here : https://we.tl/t-jXsAGPBUO8
I'm not sure to understand. The client starts sending TCP retransmission for a Seq number that already have been ACKed.

psykokwak-com, 2023-01-16:
Ok. It seems this is a performance issue of the Teensy41 driver https://github.com/cesanta/mongoose/pull/1954
it misses some rx frames.

I increased the driver RX buffer (something like 32/64) to fixes the issue.
But the overall rx throughput is low (average 300kbps).
On wireshark dump, I see that the device takes up to 200ms to send back an ACK to the client.
Reducing the MIP_TCP_ACK_MS to something like 10(ms) increase the rx throughput but it remains quite low (2Mbps).

scaprile, 2023-01-16:
That timer is there to avoid sending unnecessary ACKs when they can be piggybacked on a response. Currently it is retriggered on every received frame, that could probably be improved. However, given that TCP has a window, the other end will keep sending at its will as long as it does not fill that window.

Frames 172-173: 19426 + 536 - 11922 = 8040. MIP advertises an 8K window and it has been filled by the sender, who apparently did not receive that ACK, that I see on frame 160.
MIP does not do selective ACKs, as said above, 

You are losing frames on the send side of your device. That is really weird, check your driver, it doesn't seem to be a performance issue given that you are almost not sending anything.

closed 2023-01-16 by scaprile