cesanta/mongoose #1973: [MIP] Lot of "SEQ != ACK" error

psykokwak-com, 2023-01-18:
- My goal is: Just using the library
- My actions were: using the library with the MIP stack
- My expectation was: Less connections drop
- The result I saw: See below
- My question is: How to fix it.

# Environment

- mongoose version: latest "master" at the time I write this ticket.
- Compiler/IDE and SDK: Visual micro (or Arduino)
- Target hardware/board: Teensy 4.1 board with its MIP driver
- Connectivity chip/module: Ethernet
- Target RTOS/OS (if applicable): Bare metal (mangoose with MIP driver)

# Log

> 261d40 1 mongoose.c:425:mg_error        40 0x0 SEQ != ACK: d8231ee6 d8231ee7
> 27079a 1 mongoose.c:425:mg_error        41 0x0 SEQ != ACK: 71f72668 71f72669
> 27f1f4 1 mongoose.c:425:mg_error        42 0x0 SEQ != ACK: 27ba5463 27ba5464
> 28dc4e 1 mongoose.c:425:mg_error        43 0x0 SEQ != ACK: d3c2cfe7 d3c2cfe8
> 29c6a8 1 mongoose.c:425:mg_error        44 0x0 SEQ != ACK: 2ffc6c83 2ffc6c84

Lot of SEQ != ACK error with a difference of 1 between the SEQ number and ACK number.
I'm not doing anything special.

scaprile, 2023-01-18:
Yes, you are missing frames and MIP does not accept selective ACKs nor does selective retransmissions

scaprile, 2023-01-19:
OK, that looks weird, I'll try to reproduce it here but can't solve a puzzle when there are missing pieces. I don't see the connection, but based on the seq/ack numbers in Wireshark, what are you doing, just connecting and staying silent ? Who connects to whom ? I see two different source port numbers for your PC, do you have 2 connections ? What is it, a browser ? Which one ?

scaprile, 2023-01-19:
That is expected behavior, we have not yet properly implemented the 4-way closing handshake, your client sends ACK instead of FIN ACK, which is correct, but not expected, so the server closes with a RST instead of normal closure.

psykokwak-com, 2023-01-19:
Ok. Thanks for your answer.
Is it possible in some cases that the side does not  fully close the connections ?
That could explain I receive lot of errors like this one (with seq number difference of 1) even with no active connection ?

scaprile, 2023-01-19:
Your browser opened one extra connection, no traffic. It left both connections open and the device closed them after 3 keepalive exchanges, sending a 4th just before closing. I changed that behaviour a couple of days ago, please upgrade. You seem to have tweaked the keepalive time, too.
Check with latest in the repo