cesanta/mongoose #2018: Sorry, the information provided is insufficient

xiaowulovenana, 2023-02-03:
Sorry, the information provided is insufficient

Program running environment：Linux mogo 4.14.98+ #1 SMP PREEMPT Fri May 27 12:01:38 CST 2022 aarch64 aarch64 aarch64 GNU/Linux

###### Code reference: [mongoose/examples/multi-threaded/](https://github.com/cesanta/mongoose/tree/7.8/examples/multi-threaded)）

```c

void thread_function(void *param)
{
	struct userdata *p = (struct userdata *) param;
	// Respond, wakeup event manager
	std::string url = std::string(p->uri.ptr, p->uri.len);
	std::string body = std::string(p->body.ptr, p->body.len);
	auto it = s_handler_map.find(url);
	if (it != s_handler_map.end()){
		ReqHandler handle_func = it->second;
		handle_func(url, body, p, sendHttpRsp);
	}
	else{
		send(p->sock, RSP_ERR, strlen(RSP_ERR), 0);
	}
	close(p->sock);               // Close the connection
	free((void *) p->body.ptr);   // free body
	free((void *) p->uri.ptr);   // free uri
	free(p);                      // free userdata
}

void onHttpEvent(mg_connection *c, int ev, void *ev_data, void *fn_data)
{
	if (ev == MG_EV_HTTP_MSG) {
		mg_http_message *hm = (struct mg_http_message *) ev_data;
		userdata *data = (userdata *)calloc(1, sizeof(*data));  // worker will free it
		data->body = mg_strdup(hm->body);              // worker will free it
		data->uri = mg_strdup(hm->uri);                 // worker will free it
		data->sock = mg_mkpipe(c->mgr, _pipeEvent, c, true);  // Create pipe
		start_thread(thread_function, data);  // Start thread and pass data
	}
  	else if (ev == MG_EV_CLOSE) {
    	if (c->fn_data != NULL){
			_unlink_conns(c, (mg_connection *)c->fn_data);
		}	
  	}
}

```
 

> In a nutshell, multiple url paths are registered, different processing results are returned based on different paths, and multiple clients perform http GET and post

- Ignore my code style, I just want to use multithreading to solve my concurrency scenario

- When the program runs for a few days, the CPU is high and the http get request will be blocked without return

- The above is my problem in version 7.8

1. Today, I changed the version to 7.7, and there was an error print message

```bash
4419:mg_iotest      select: -1 9
Feb 02 12:33:07 mogo Start.sh[3508]: 26b77a 1 mongoose.c:4419:mg_iotest      select: -1 9
Feb 02 12:33:07 mogo Start.sh[3508]: 26b77b 1 mongoose.c:4419:mg_iotest      select: -1 9
Feb 02 12:33:07 mogo Start.sh[3508]: 26b77b 1 mongoose.c:4302:accept_conn    1 accept failed, errno 22
Feb 02 12:33:07 mogo Start.sh[3508]: 26b77b 1 mongoose.c:4419:mg_iotest      select: -1 9
Feb 02 12:33:07 mogo Start.sh[3508]: 26b77b 1 mongoose.c:4419:mg_iotest      select: -1 9
Feb 02 12:33:07 mogo Start.sh[3508]: 26b77c 1 mongoose.c:4419:mg_iotest      select: -1 9
Feb 02 12:33:07 mogo Start.sh[3508]: 26bb65 1 mongoose.c:4419:mg_iotest      select: -1 9
Feb 02 12:33:07 mogo Start.sh[3508]: 26bb66 1 mongoose.c:4419:mg_iotest      select: -1 9
Feb 02 12:33:07 mogo Start.sh[3508]: 26bb66 1 mongoose.c:4419:mg_iotest      select: -1 9
Feb 02 12:33:07 mogo Start.sh[3508]: 26bb66 1 mongoose.c:4302:accept_conn    1 accept failed, errno 22
Feb 02 12:33:07 mogo Start.sh[3508]: 26bb66 1 mongoose.c:4419:mg_iotest      select: -1 9
Feb 02 12:33:07 mogo Start.sh[3508]: 26bb67 1 mongoose.c:4419:mg_iotest      select: -1 9
Feb 02 12:33:07 mogo Start.sh[3508]: 26bb67 1 mongoose.c:4419:mg_iotest      select: -1 9
Feb 02 12:33:07 mogo Start.sh[3508]: 26bb67 1 mongoose.c:4302:accept_conn    1 accept failed, errno 22
Feb 02 12:33:07 mogo Start.sh[3508]: 26bb67 1 mongoose.c:4419:mg_iotest      select: -1 9
Feb 02 12:33:07 mogo Start.sh[3508]: 26bb67 1 mongoose.c:4419:mg_iotest      select: -1 9
Feb 02 12:33:07 mogo Start.sh[3508]: 26bb68 1 mongoose.c:4419:mg_iotest      select: -1 9
Feb 02 12:33:07 mogo Start.sh[3508]: 26bb68 1 mongoose.c:4302:accept_conn    1 accept failed, errno 22
Feb 02 12:33:07 mogo Start.sh[3508]: 26bb68 1 mongoose.c:4419:mg_iotest      select: -1 9
Feb 02 12:33:07 mogo Start.sh[3508]: 26bb68 1 mongoose.c:4419:mg_iotest      select: -1 9
```

- Please help me

_Originally posted by @xiaowulovenana in https://github.com/cesanta/mongoose/discussions/2013#discussioncomment-4851389_

scaprile, 2023-02-03:
Please don't crosspost, and don't open several issues for the same problem of yours. I'll link your question and keep this issue open, please post here. If you are a paying customer contact support@cesanta.com, otherwise please wait and you'll be served.

xiaowulovenana, 2023-02-06:
Sorry, this is the first time I submit a question. I don't understand the rules very well

Back to the problem, in the `handle_ In func (url, body, p, sendHttpRsp);` I only did time-consuming processing and sent data through
```
void sendHttpRsp(userdata *p_userdata, std::string rsp)
{
	send(p_userdata->sock, rsp.c_str(), rsp.length(), 0);
}
```
 How to keep the link open. Finally, `close(p->sock);`The connection should be closed.

HTTP client I use: [mongoose](https://github.com/cesanta/mongoose)/[examples](https://github.com/cesanta/mongoose/tree/master/examples)/[http-client](https://github.com/cesanta/mongoose/tree/master/examples/http-client)/main.c

scaprile, 2023-02-06:
Sniff your network
Check your OS resources

closed 2023-02-06 by scaprile