cesanta/mongoose #2049: As http client download large file error

lifw555, 2023-02-10:
As http client download  large file（> 1GB）

when i download small file, it is success （<3MB）

when as file > 3MB, error:
```
mongoose.c:404:mg_error      1 0x3ac max_recv_buf_size reached
```
i know can change `max_recv_buf_size ` mcro value to download file (>3MB)
but that's not a good way, i can not change `max_recv_buf_size `  to 2GB.

so my question is how to do download large file （> 1GB） ?
I can not find any examples。

thanks

# Environment
- mongoose version: 7.8

scaprile, 2023-02-10:
Use `MG_EV_HTTP_CHUNK`
https://mongoose.ws/tutorials/file-uploads/#binary-upload

closed 2023-02-10 by scaprile

lifw555, 2023-02-10:
> Use `MG_EV_HTTP_CHUNK` https://mongoose.ws/tutorials/file-uploads/#binary-upload

i use mongoose as http client . download from other website . not use mongoose as http server .

so how to do ?

scaprile, 2023-02-10:
The example may be for a server, but if you take the time to read the link, pay attention, read the docs, and try, you'll soon discover that an event is an event is an event and that applies to your question.
Do client and server have different events for HTTP msgs ? No
https://mongoose.ws/documentation/#events

lifw555, 2023-02-10:
> The example may be for a server, but if you take the time to read the link, pay attention, read the docs, and try, you'll soon discover that an event is an event is an event and that applies to your [...]

yes, i got it .

thanks !!!