cesanta/mongoose #2091: Compling as a Library to port other language

yesilcimenahmet, 2023-02-23:
Hi, 
I want to compile Mongoose as library to port for other library such as Delphi, Can I compile Mongoose lib as DLL? but I could not do that for real. Is it any info about that ?

scaprile, 2023-02-23:
Will you please be so kind to honor the issue template and actually fill it ?

yesilcimenahmet, 2023-02-23:
> Will you please be so kind to honor the issue template and actually fill it ?

Fixed, Thanks.

yesilcimenahmet, 2023-02-24:
Any info any help? I could not find documents about compilation enough.

scaprile, 2023-02-26:
Mongoose is just a single C source file and a .h header file. What compilation information do you need to compile a single C source ? Do you actually have a problem ? What exactly is it ?
Do you just want to know how to create a DLL ? Did you google that ? This is one of a dozen of possible answers: https://interviewsansar.com/dll-dynamic-link-library-in-c/

yesilcimenahmet, 2023-02-26:
Hi, I mean compiling Mongoose as Shared Library (DLL) and thus I can use it in other programming language thanks by Mongoose.h I am working on it to compile as DLL

closed 2023-02-26 by scaprile

karashiiro, 2024-02-03:
For anyone looking at this in the future, it's easy to do this with CMake on Windows ([example CMakeLists.txt](https://github.com/karashiiro/MongooseSharp/blob/6d8daeba541cace44d316e20d4125f85c18bbf5c/CMakeLists.txt))