cesanta/mongoose #2104: mg_http_printf_chunk format does not follow printf semantic

shunkica, 2023-03-05:
- My goal is:
- output a double as string with fixed decimal points

- My actions were:

```
double num = 0.14;
mg_http_printf_chunk(c, "%1.*f", 1, num);
mg_http_printf_chunk(c, "%1.*f", 2, num);
mg_http_printf_chunk(c, "%1.*f", 3, num);
```

- My expectation was: 

> 0.1
> 0.14
> 0.140

- The result I saw: 

> 1e-01
> 0.14
> 0.14

- My question is:
Why does this not produce the same result as printf fmt?

# Environment

- mongoose version: 7.9
- Compiler/IDE and SDK: gcc version 11.3.1
- Target hardware/board: x86_64
- Connectivity chip/module: _?_
- Target RTOS/OS (if applicable): _?_

scaprile, 2023-03-05:
Yes, that looks like a bug affecting all `*printf()` derivatives https://mongoose.ws/documentation/#mg_snprintf-mg_vsnprintf

Wrt your side question:
30 secs / 300000 = 100us per iteration, you are sending chunks of 10 bytes where 40% is overhead
Yes, Mongoose will resize its TCP buffer everytime you keep adding more and more data but it should also be sending. Please sniff your network and open a question telling us your use case.

shunkica, 2023-03-06:
The problem is not related to calling mg_http_printf_chunk many times. It is a general problem with sending large responses. The same thing happened when I buffered the entire thing and sent it with mg_http_reply.

I have opened issue #2105

cpq, 2023-03-20:
@shunkica thanks for reporting this. We need to fix floating point print, and add a respective test (to the large list of existing tests).

W.r.t. second question. Set `MG_IO_SIZE` to 5 megabytes and it will be fast.

commit 6b29a2a7 (2023-04-07): Fix #2104 - %.*f formatting
commit 2b325102 (2023-04-07): Merge pull request #2143 from cesanta/dbl

Fix #2104 - %.*f formatting
closed 2023-04-07 by scaprile