cesanta/mongoose #2123: How to contribute to documentation?

strager, 2023-03-17:
- My goal is: Fix some issues in the Mongoose documentation (https://mongoose.ws/documentation/), and also add new documentation for new features I want to add.
- My actions were: Clone https://github.com/cesanta/mongoose and look in the `docs` folder, as I did a year ago.
- My expectation was: `docs/README.md` has the documentation.
- The result I saw: Found Git commit b9f225027890e761a39ae2b6f122dc822f1f28e8 with no pointer to where the new documentation is. Found https://github.com/cesanta/mongoose-os-docs but it doesn't seem to contain docs for the HTTP library.
- My question is: In which repo is Mongoose documentation stored? How do I submit patches to the documentation?

# Environment

- mongoose version: N/A
- Compiler/IDE and SDK: N/A
- Target hardware/board: N/A
- Connectivity chip/module: N/A
- Target RTOS/OS (if applicable): N/A

scaprile, 2023-03-17:
> * new documentation for new features I want to add

Submit a PR with a README.md and if that is merged, the README will find its way to the docs, https://github.com/cesanta/mongoose#contributions

Wrt to issues in the docs, we are in a process of integrating user guide and tutorials; that is the reason why docs/README.md was deleted. If you feel like, you can take release 7.9 README.md or rewind your fork to the commit before it was deleted, and mark them there. 

Mongoose-OS is a different product, it is based on Mongoose but on an earlier release of it.

closed 2023-03-17 by scaprile

strager, 2023-03-17:
> Submit a PR with a README.md

Where should I place this README.md file?

scaprile, 2023-03-17:
That is a good question if you are not writing an example exercising your features. If that is the case, I would suggest you just leave the documentation snippets in the PR itself, it will be easier to pick them up. AFAIK there is no official procedure for that, at least until we finish integrating the documentation.