cesanta/mongoose #2149: OPTIONS request with openssl connection more times and client can not recieve any results

zhanghuaifeng, 2023-04-10:
- My goal is: Run example/http-restful-server on Ubuntu 20.04
- My actions were:webbrower send  ajax request https to server
- My expectation was: The browser can interact with the server over https
- The result I saw: When the browser sends an ajax post request, it sends the options request for the first time, but receives no reply
-
[image]

- My question is: OPTIONS request with openssl connection more times and client can not recieve any results

# Environment

- mongoose version:7.9
- Compiler/IDE and SDK: GCC 9.3.0
- Target hardware/board: _?_
- Connectivity chip/module: _?_
- Target RTOS/OS (if applicable): _?_

scaprile, 2023-04-10:
40222 is not the port we use in the example
That long path I see is not to our certificate and key as provided in the example.
Our logs have timestamps and use English, so obviously you modified the example code.
Please run the example as is, read the tutorials and if the example does not work, then provide all pertaining information
https://mongoose.ws/documentation/tutorials/http-server/
https://mongoose.ws/documentation/tutorials/tls/
OpenSSL version ? Browser ? Version ?
A network capture file

If the example works and your code does not, then it is your code, your mods, your certs.
What are you doing ? Does it work without TLS ? Etc.

cpq, 2023-04-10:
@zhanghuaifeng

1. Don't paste IMAGES of the logs. Paste logs.
2. If you're using a modified example, share a code.

It should be very easy for us to reproduce your issue. Please make it so. Otherwise, we simply close it as a useless noise.

closed 2023-04-17 by cpq