cesanta/mongoose #2153: How to port Mongoose library to a different platform

niseva-openwrt, 2023-04-12:
- My goal is: Port mongoose library to a new platform
- My question is: Do you have any guide to steps to get started

# BG95 module with ThreadX operating system

I want to port mongoose to ThreadX OS running on BG95 NBIoT module. Is there any step by step guide you have?

scaprile, 2023-04-12:
We support AzureRTOS, and it seems to be ThreadX-based. Maybe you can start setting up for it and work your way from there:
https://github.com/cesanta/mongoose/blob/d9f05532cd4f6e139e09cc86f11107185224460c/src/arch.h#L32-L33

Docs:
https://github.com/cesanta/mongoose/blob/master/README.md --> https://mongoose.ws/documentation/ --> 
https://mongoose.ws/documentation/#build-options
https://mongoose.ws/documentation/#custom-build

Going down the ladder one more step, check the headers for other known platforms
https://github.com/cesanta/mongoose/blob/master/src/arch_freertos.h
https://github.com/cesanta/mongoose/blob/master/src/arch_azurertos.h
and TCP/IP stacks
https://github.com/cesanta/mongoose/blob/master/src/net_ft.h
https://github.com/cesanta/mongoose/blob/master/src/net_lwip.h

Finally, if source code has to be modified, add a special case where needed:
https://github.com/cesanta/mongoose/blob/d9f05532cd4f6e139e09cc86f11107185224460c/src/sock.c#L311-L326

closed 2023-04-12 by scaprile

niseva-openwrt, 2023-04-12:
Ok. Thanks for your prompt response.