cesanta/mongoose #2155: Support multiple sockets on W5500

sagarreddypatil, 2023-04-13:
- My goal is: to listen on multiple ports simultaneously
- My actions were: nothing, library doesn't support this functionality
- My expectation was: ability to setup multiple mg_listen's
- The result I saw: N/A
- My question is: N/A

# Environment

- mongoose version: 7.9 (though functionality seems to be missing on master)
- Compiler/IDE and SDK: Pico SDK version 1.4.0
- Target hardware/board: RP2040 + W5500
- Connectivity chip/module: W5500
- Target RTOS/OS (if applicable): N/A, Pico SDK only

scaprile, 2023-04-13:
> * My actions were: nothing, library doesn't support this functionality

I beg to differ. the example is doing exactly that
https://github.com/cesanta/mongoose/blob/52ed68ed248363e780f66bba261cd5368193d2a4/examples/rp2040/pico-w5500/main.c#L92-L93

closed 2023-04-13 by scaprile

sagarreddypatil, 2023-04-16:
@scaprile  Your example demonstrates the handling of multiple HTTP servers, not multiple TCP sockets. In the W5500 driver program, only one socket on the W5500 is used whereas 6 are available.

scaprile, 2023-04-16:
You opened this issue saying
> My goal is: to listen on multiple ports simultaneously

We showed you how you actually already can. Each port, of course, will use at least one socket (one for listening and one for any established connection).
Add to that we are connected to an MQTT server (TCP) and we resolve names (DNS requests, UDP) and get the time via SNTP (UDP), all at the same time; you see, we don't have a limit on the amount of simultaneous connections (beyond available RAM, of course). Have you actually tried the example ?

**_We are not using any of the W5500 sockets in TCP/UDP mode, we use the chip at the Ethernet level, we send IP and receive IP_** (MACRAW mode)

https://github.com/cesanta/mongoose/blob/cacd9c7f9526441f66bd8a45bf8b26beb866ab26/src/tcpip/driver_w5500.c#L71-L79

scaprile, 2023-04-16:
You opened this issue saying
> My goal is: to listen on multiple ports simultaneously

We showed you how you actually already can. Each port, of course, would use at least one socket (one for listening and one for each established connection), though we actually do not use sockets internally.
The server not only serves pages but maintains an active WebSocket connection where it sends data that is graphed,
Add to that we are connected to an MQTT server (TCP) and we resolve names (DNS requests, UDP) and get the time via SNTP (UDP), all at the same time; does this look like it runs in only one socket ?
You see, we don't have a limit on the amount of simultaneous connections (beyond available RAM, of course). Have you actually tried the example ? Seen the driver code ? Read the tutorial ? https://mongoose.ws/documentation/tutorials/rp2040/pico-w5500/

**_We are not using any of the W5500 sockets in TCP/UDP mode, we use the chip at the Ethernet level, we send IP and receive IP_** (MACRAW mode)

https://github.com/cesanta/mongoose/blob/cacd9c7f9526441f66bd8a45bf8b26beb866ab26/src/tcpip/driver_w5500.c#L71-L79