cesanta/mongoose #2188: mg_iobuf_resize ERROR

OmarMorceli, 2023-05-12:
- My goal is: Sending payloads via MQTT

- My actions were:  formatting JSON messages and publishing them to an MQTT broker

- My question is: After a minute, I receive the following error message:

...
26bc5 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26bc6 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26bc7 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26bc8 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26bcd 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26bd1 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26bd6 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26bda 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26bdf 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26be3 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26be8 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26bec 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26bf1 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26bf5 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26bfa 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26bfe 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26c03 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26c07 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26c0c 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26c10 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26c15 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26c19 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26c1e 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26c22 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26c27 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26c2b 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26c30 1 iobuf.c:39:mg_iobuf_resize      3072->3328
26c34 1 iobuf.c:39:mg_iobuf_resize      3072->3328
...

The ESP32 keeps doing this until I restart it.

# Environment

- mongoose version: 7.8

- Compiler/IDE and SDK: ESP-IDF / VS Code

- Target hardware/board: ESP32 DevKitc v4

scaprile, 2023-05-12:
I would say that some function of yours is leaking memory and causing the resize to fail or maybe your stack is overflowing. Check your heap size and stack size.
Please use the latest release and if your issue persists post the _minimum_ piece of code reproducing it.

scaprile, 2023-05-18:
No follow up, closing

closed 2023-05-18 by scaprile