cesanta/mongoose #2192: Binding to a Link-Local/zone-indexed IPv6 address fails

comburent, 2023-05-12:
Addresses of the form  `fe80::1%lo` (using the interface name) or `fe80::1%1` (standard) can't be used with `mg_http_listen`.
I've tried:
```
// note: would require a if_nametoindex before passing to getaddrinfo
http://[fe80::1%lo]:8000

// note: can be passed directly to getaddrinfo
http://[fe80::1%1]:8000

// attempt 2: percent-encoded since these are URIs
http://[fe80::1%25lo]:8000
http://[fe80::1%251]:8000
```

# Environment

- mongoose version: 7.9
- Compiler/IDE and SDK: GCC
- Target RTOS/OS (if applicable): Linux

scaprile, 2023-05-13:
Proper syntax would be with percent encoding, however, AFAIK, we currently do not support zone indexed addresses; though that might change. Can you please describe your use case (beyond "I want to bind to an interface") and why you can't do it in a different way ? Thanks.

comburent, 2023-05-17:
> Proper syntax would be with percent encoding, however, AFAIK, we currently do not support zone indexed addresses; though that might change. Can you please describe your use case (beyond "I want to bind to an interface") and why you can't do it in a different way ? Thanks.

I'd like to use it for the following cases:
- To put it behind a reverse proxy in a LAN without depending on a GUA/ULA: GUA/ULA can be relatively unstable (and depends on a router) while a Link Local address is much more dependable and can be used straight-away.
- To avoid fiddling around with the _port_: When both the reverse proxy and the application are hosted in the same machine, using additional addresses would allow for a “lateral” expansion of places that can be used to bind the application, instead of randomly pulling a port from the magical hat and hope that in the future no app decides to use that same port. (a UNIX socket would be a better choice here, if mongoose supported it)

Another observation is that IIUC under IPv6 the closest approximation to the `127.0.0.0/8` block would be to add additional link-local addresses to the loopback device to go beyond that single [::1] address.

cpq, 2023-06-01:
@comburent we don't currently support scope ID in our IPv6 parsing function

https://github.com/cesanta/mongoose/blob/1dde75472424191b7ba8e6e1d844cceb3680ca44/src/net.c#L74-L109

Neither we have scope id in the address struct:
https://github.com/cesanta/mongoose/blob/1dde75472424191b7ba8e6e1d844cceb3680ca44/src/net.h#L15-L20

So adding those would require some effort.

@comburent  can you elaborate on your use case please? What application you're working on?

cpq, 2023-07-19:
No response, closing this

comburent, 2023-09-20:
@scaprile @cpq Could we reopen this issue?

scaprile, 2023-09-20:
Please elaborate on your use case

comburent, 2023-09-20:
> Please elaborate on your use case

I think I did so in the bullet points at <https://github.com/cesanta/mongoose/issues/2192#issuecomment-1551688805>. Is there something in particular that you would like me to expand further?

scaprile, 2023-09-20:
Those are very nice technical explanations. Do you have a selling point so you can convince us to invest our time in doing that ? What is your actual use case in real life, why can't you do it in a different way ? Is this a show stopper ? Is it a convenience ? Etc.

comburent, 2023-09-20:
> Those are very nice technical explanations. Do you have a selling point so you can convince us to invest our time in doing that ? What is your actual use case in real life, why can't you do it in a different way ? Is this a show stopper ? Is it a convenience ? Etc.

I'm mostly after the first point, to put it behind a reverse proxy and to have a stable access to a mongoose application when I'm connected to the same IPv6 only LAN. I don't have a stable GUA prefix so I can't use those addresses within configuration files as they would find themselves unexpectedly outdated. On the other hand, a LLA has infinite lifetime which I can safely hardcode into the reverse proxy configuration.

I could setup an ULA but that on its own is inconvenient and would require more reasoning on the firewall as ULAs are routable when a simple point-to-point link with LLAs would be much easier to reason about.

From a end-user perspective it can be a bit surprising to find out that this kind of URI isn't supported (hence my initial attempts listed in the report) but I wouldn't classify it as an app-breaking showstopping issue but rather an inconvenience. Can be worked around but annoying and complexity adding.

For background I'm not developing a mongoose app, I'm just a user of an app that is powered with mongoose.

scaprile, 2023-09-20:
OK, let's see what CPQ says.
For background, Mongoose is oriented towards embedded systems and 1 interface is mostly the case.

cpq, 2023-09-27:
It shows once again that ipv6 is an overengineered &^%.
16 bytes address, and still not enough to implement routing? Perhaps one of the reasons why ipv4 still dominates.

@comburent thanks for the detailed background.
What's that mongoose-powered app?

comburent, 2023-09-29:
> @comburent thanks for the detailed background. What's that mongoose-powered app?

It's <https://github.com/jcorporation/myMPD>.

IMO, I'd say its unfair to lay the blame squarely on IPv6 because I think there's actually 2 issues underneath here:
1. URI handling.
All that mongoose is missing here is accounting that a zone-index may occur within a IPv6 address. This URI is just like any other URI so it's not being treated in any special way.
2.  Actual support for the zone-index underneath.
This is the IPv6 part and pertains to the socket logic. I don't think there's anything special to do here save for passing the zone-index to the socket logic.

FTR, I was mistaken when I wrote the comments in the original report. It looks like `if_nametoindex` isn't really necessary as I found out when investigating a related issue at <https://sourceware.org/bugzilla/show_bug.cgi?id=30604>.
I think mongoose here can decide what to do with _non-numeric_ zone indices, whether to do a `if_nametoindex` first, relay the zone-index string as is or force zone-indices to be _numeric_. (since numeric zone indices must be universally supported)

cpq, 2023-09-29:
@comburent thank you.

I think we'll go with the numeric index. Less lookups, and easier to parse.

commit 1cd55e91 (2023-09-29): Fix #2192 - honor addr%scopeid ipv6 notation
commit 46ecb07f (2023-09-29): Fix #2192 - honor addr%scopeid ipv6 notation
commit 435654ec (2023-10-10): Fix #2192 - honor addr%scopeid ipv6 notation
commit 194a7638 (2023-10-10): Fix #2192 - honor addr%scopeid ipv6 notation
closed 2023-09-29 by cpq