cesanta/mongoose #2203: `make install` should be able to recognize `lib64`

mkoncek, 2023-05-18:
While Makefile respects `$(DESTDIR)` and `$(PREFIX)` in the current state it hardcodes `/lib/` in the installation path.
A better solution would be to add a variable `$(LIBDIR)` that combines `$(PREFIX)` and `/lib` by default but can be overridden.

scaprile, 2023-05-18:
The Makefile is mostly for testing purposes, please honor the issue template and fill all the information, we can't guess your system and wouldn't have to.
Does it work for you ? Is this a suggestion ? We wouldn't have to guess that either.

mkoncek, 2023-05-18:
- My goal is: Package this library in Fedora.
- My actions were: Standard build procedure using make. This includes running
```
/usr/bin/make install DESTDIR=.../rpmbuild/BUILDROOT/%{NAME}-%{VERSION}-%{RELEASE}.x86_64 INSTALL="/usr/bin/install -p"
```
- My expectation was: The installation target would be fully configurable with no harcoded entries in the middle of the path.
- The result I saw: I am unable to specify target of installation as `$(DESTDIR)/$(PREFIX)/lib64/...`
- My question is: Can you make `install` step be able to use `lib64` instead of `lib` somehow?

# Environment

- mongoose version: 7.9
- Compiler/IDE and SDK: gcc (GCC) 13.1.1
- Target hardware/board: N/A
- Connectivity chip/module: N/A
- Target RTOS/OS (if applicable): Fedora Linux 39

scaprile, 2023-05-18:
I would favor something like
```
LIB ?= lib
INC ?= include
LIBDIR ?= $(PREFIX)/$(LIB)
INCDIR ?= $(PREFIX)/$(INC)
...
install: linux-libs
	install -Dm644 libmongoose.a libmongoose.so.$(VERSION) $(DESTDIR)$(LIBDIR)
	ln -s libmongoose.so.$(VERSION) $(DESTDIR)$(LIBDIR)/libmongoose.so
	install -Dm644 mongoose.h $(DESTDIR)$(INCDIR)/mongoose.h

uninstall:
	rm -rf $(DESTDIR)$(LIBDIR)/libmongoose.a $(DESTDIR)$(LIBDIR)/libmongoose.so.$(VERSION) $(DESTDIR)$(INCDIR)/mongoose.h $(DESTDIR)$(LIBDIR)/libmongoose.so
```
Something like that wouldn't break current state of things (defaults to install in /usr/local/lib and we haven't had any similar comments from other platforms so far)
If that works for you, you are welcome to do it and submit a PR.

BTW, 7.10 is coming out soon.

mkoncek, 2023-05-19:
1. Yes, that looks good.
2. I cannot easily sign a CLA for pull requests.
3. There are more issues in the Makefile and some of them are Linux-specific:
* https://github.com/cesanta/mongoose/blob/master/Makefile#L38
`LDFLAGS` of an extra feature (SSL support) gets dropped when environment variable `LDFLAGS` is defined due to `?=`, distribution builds can specify teir own link flags for example for hardening.
  * Suggestion: use `LDFLAGS +=` for *extra features* like SSL support.
* According to [GNU Make documentation](https://www.gnu.org/software/make/manual/html_node/Implicit-Variables.html#index-LDFLAGS) `LDFLAGS` and `LDLIBS` should be separated, but many developers do not do it and it is not necessary.
* (Linux-specific): There is no link flag `-Wl,-soname,<NAME>`. The `<NAME>` string indicates binary compatibility. This is important in Linux world. For example when an author of a library releases a new patch version fixing a security vulnerability, they would keep the same soname and keep the library binary compatible. On the other hand, releasing a new version where signatures / layouts were changed, symbols removed and so, would not be binary compatible and s would require changing the soname. [Some more details](https://stackoverflow.com/questions/12637841/what-is-the-soname-option-for-building-shared-libraries-for).
  * Libraries are usually binary compatible per major version.
  * If you don't want to concern yourself with this, you can just put `-Wl,-soname,libmongoose.so.<MAJOR>.<MINOR>`

mkoncek, 2023-05-19:
It turned out that I didn't need to update the package in Fedora, in any case I had a patch like this prepared.

```diff
From 2c1713c8035976dcb4f04f2853dda1b409e3511c Mon Sep 17 00:00:00 2001
From: Marian Koncek <mkoncek@redhat.com>
Date: Fri, 19 May 2023 11:00:50 +0200
Subject: [PATCH] Fix Makefile

---
 Makefile | 15 +++++++++------
 1 file changed, 9 insertions(+), 6 deletions(-)

diff --git a/Makefile b/Makefile
index d1f1189..e3cae2a 100644
--- a/Makefile
+++ b/Makefile
@@ -14,6 +14,8 @@ ASAN ?= -fsanitize=address,undefined,alignment -fno-sanitize-recover=all -fno-om
 ASAN_OPTIONS ?= detect_leaks=1
 EXAMPLES := $(dir $(wildcard examples/*/Makefile))
 PREFIX ?= /usr/local
+LIBDIR ?= $(PREFIX)/lib
+INCLUDEDIR ?= $(PREFIX)/include
 VERSION ?= $(shell cut -d'"' -f2 src/version.h)
 COMMON_CFLAGS ?= $(C_WARN) $(WARN) $(INCS) $(DEFS) -DMG_ENABLE_IPV6=$(IPV6) $(TFLAGS)
 CFLAGS ?= $(OPTS) $(ASAN) $(COMMON_CFLAGS)
@@ -32,13 +34,13 @@ endif
 ifeq "$(SSL)" "MBEDTLS"
 MBEDTLS ?= /usr/local
 CFLAGS  += -DMG_ENABLE_MBEDTLS=1 -I$(MBEDTLS)/include -I/usr/include
-LDFLAGS ?= -L$(MBEDTLS)/lib -lmbedtls -lmbedcrypto -lmbedx509
+LDFLAGS += -L$(MBEDTLS)/lib -lmbedtls -lmbedcrypto -lmbedx509
 endif
 
 ifeq "$(SSL)" "OPENSSL"
 OPENSSL ?= /usr/local
 CFLAGS  += -DMG_ENABLE_OPENSSL=1 -I$(OPENSSL)/include
-LDFLAGS ?= -L$(OPENSSL)/lib -lssl -lcrypto
+LDFLAGS += -L$(OPENSSL)/lib -lssl -lcrypto
 endif
 
 all:
@@ -163,17 +165,18 @@ mingw++: Makefile mongoose.h $(SRCS)
 	$(DOCKER) mdashnet/mingw x86_64-w64-mingw32-g++ $(SRCS) -W -Wall -Werror -I. $(DEFS) -lwsock32 -o $@.exe
 
 linux-libs: CFLAGS += -fPIC
+linux-libs: LDFLAGS += -Wl,-soname,libmongoose.so.$(VERSION)
 linux-libs: mongoose.o
 	$(CC) mongoose.o $(LDFLAGS) -shared -o libmongoose.so.$(VERSION)
 	$(AR) rcs libmongoose.a mongoose.o
 
 install: linux-libs
-	install -Dm644 libmongoose.a libmongoose.so.$(VERSION) $(DESTDIR)$(PREFIX)/lib
-	ln -s libmongoose.so.$(VERSION) $(DESTDIR)$(PREFIX)/lib/libmongoose.so
-	install -Dm644 mongoose.h $(DESTDIR)$(PREFIX)/include/mongoose.h
+	install -Dm644 libmongoose.a libmongoose.so.$(VERSION) $(DESTDIR)$(LIBDIR)
+	ln -s libmongoose.so.$(VERSION) $(DESTDIR)$(LIBDIR)/libmongoose.so
+	install -Dm644 mongoose.h $(DESTDIR)$(INCLUDEDIR)/mongoose.h
 
 uninstall:
-	rm -rf $(DESTDIR)$(PREFIX)/lib/libmongoose.a $(DESTDIR)$(PREFIX)/lib/libmongoose.so.$(VERSION) $(DESTDIR)$(PREFIX)/include/mongoose.h $(DESTDIR)$(PREFIX)/lib/libmongoose.so
+	rm -rf $(DESTDIR)$(LIBDIR)/libmongoose.a $(DESTDIR)$(LIBDIR)/libmongoose.so.$(VERSION) $(DESTDIR)$(INCLUDEDIR)/mongoose.h $(DESTDIR)$(LIBDIR)/libmongoose.so
 
 mongoose.c: Makefile $(wildcard src/*) $(wildcard mip/*.c)
 	(cat src/license.h; echo; echo '#include "mongoose.h"' ; (for F in src/*.c mip/*.c ; do echo; echo '#ifdef MG_ENABLE_LINES'; echo "#line 1 \"$$F\""; echo '#endif'; cat $$F | sed -e 's,#include ".*,,'; done))> $@
-- 
2.40.1
```

scaprile, 2023-05-19:
OK, thank you, we'll discuss these changes.

PR #2214 (2023-05-24): make more distro-friendly
Make the Makefile more Linux distro-package friendly as per #2203 :

- Remove hardcoded /lib and /include and let them be specified
- Add to a possibly specified LDFLAGS 

( closes #2203 )
closed 2023-05-24 by cpq