cesanta/mongoose #2204: 7.10 fails to bind on Windows

jasonoverland, 2023-05-19:
We upgraded from 7.9 to 7.10 and now mongoose is not binding on Windows. I reverted mg_open_listener to 7.9 and it's working again.

https://github.com/cesanta/mongoose/commit/48333041f45da73f242cf605e62c59fef6499b6f

In 7.9 only SO_REUSEADDR was being set.

In 7.10 SO_EXCLUSIVEADDRUSE is being set and then SO_REUSEADDR is being set. The second call sets WSAGetLastError to EINVAL. If I'm understanding this correctly, these are mutually exclusive options and this should not happen.

scaprile, 2023-05-19:
#2124

scaprile, 2023-05-22:
Somehow this is passing our unit-tests in W

joverland-extensis, 2023-05-22:
I'm using Windows Server 2022 and MSVC (2015 and 2022). Maybe a difference between that and mingw?

https://github.com/search?q=repo%3Acesanta%2Fmongoose%20wsock32&type=code

Should link with ws2_32 rather than wsock32?

scaprile, 2023-05-22:
That is for our testing purposes, it is usually tied to a Docker image and usually runs pver _wine_ on Linux
For regular use just follow the example Makefile, they all use `ws2_32` and have something like this:
```
ifeq ($(OS),Windows_NT)   # Windows settings. Assume MinGW compiler. To use VC: make CC=cl CFLAGS=/MD OUT=/Feprog.exe
  PROG ?= example.exe           # Use .exe suffix for the binary
  CC = gcc                      # Use MinGW gcc compiler
  CFLAGS += -lws2_32            # Link against Winsock library
  DELETE = cmd /C del /Q /F /S  # Command prompt command to delete files
  OUT ?= -o $(PROG)             # Build output
  MAKE += WINDOWS=1 CC=$(CC)
endif
```

scaprile, 2023-05-22:
Let me know if that fixes your issue.

joverland-extensis, 2023-05-23:
Yes that fixes our issue.

closed 2023-05-23 by scaprile