cesanta/mongoose #2211: Problem when sending data using MQTT

OmarM17, 2023-05-23:
- My goal is: 
Sending periodic data using MQTT.

- The result I saw: 
I got messages in the cloud, but the data stopped sending while this log messages kept coming
```
cloud_mqtt_forwarder.url=mqtts://a2n53blisc65if-ats.iot.us-east-1.amazonaws.com:8883

24bd 3 net.c:159:mg_connect             8 0xffffffff mqtts://a2n53blisc65if-ats.iot.us-east-1.amazonaws.com:8883
24c7 3 net.c:126:cloud_mqtt_forwarder_h ^^ cloud_mqtt_forwarder_handler ^^ MG_EV_OPEN

24d4 3 sock.c:146:mg_send               3 0x37 0:0 64 err 88
24d6 2 net.c:165:mqtt_forwarders_timer_ => Connect to : cloud_mqtt_forwarder.c=3ffe36b8

3018 1 event.c:21:mg_error              7 0xffffffff DNS timeout
302c 3 net.c:117:consol_mqtt_forwarder_ ^^ consol_mqtt_forwarder_handler ^^ MG_EV_CLOSE

302d 3 net.c:136:mg_close_conn          7 0xffffffff closed
3040 3 net.c:159:mg_connect             9 0xffffffff mqtts://a2n53blisc65if-ats.iot.us-east-1.amazonaws.com:8883
3042 3 net.c:90:consol_mqtt_forwarder_h ^^ consol_mqtt_forwarder_handler ^^ MG_EV_OPEN

304d 3 sock.c:146:mg_send               3 0x37 0:0 64 err 88
304f 2 net.c:157:mqtt_forwarders_timer_ => Connect to : consol_mqtt_forwarder.c=3ffe2c9c

3090 1 event.c:21:mg_error              8 0xffffffff DNS timeout
30a4 3 net.c:141:cloud_mqtt_forwarder_h ^^ cloud_mqtt_forwarder_handler ^^ MG_EV_CLOSE

30a6 3 net.c:136:mg_close_conn          8 0xffffffff closed
```
 I increase the DNS timeout like this :
```
    struct mg_mgr mgr;
    mg_mgr_init(&mgr);
    mgr.dnstimeout = 10000;
```

I am currently receiving these log messages.

```
3e95d 3 net.c:126:cloud_mqtt_forwarder_ ^^ cloud_mqtt_forwarder_handler ^^ MG_EV_OPEN

3e965 3 sock.c:146:mg_send              2574 0x37 0:0 -1 err 118
3e96a 1 event.c:21:mg_error             2574 0x37 DNS send
3e970 2 net.c:165:mqtt_forwarders_timer => Connect to : cloud_mqtt_forwarder.c=3fff5890

3e978 1 event.c:21:mg_error             2575 0xffffffff DNS error
3e97d 1 event.c:21:mg_error             2573 0xffffffff DNS error
3e983 3 net.c:136:mg_close_conn         2574 0x37 closed
3e988 3 net.c:117:consol_mqtt_forwarder ^^ consol_mqtt_forwarder_handler ^^ MG_EV_CLOSE

3e990 3 net.c:136:mg_close_conn         2573 0xffffffff closed
3e9a6 3 net.c:159:mg_connect            2576 0xffffffff mqtts://a2n53blisc65if-ats.iot.us-east-1.amazonaws.com:8883
3e9a8 3 net.c:90:consol_mqtt_forwarder_ ^^ consol_mqtt_forwarder_handler ^^ MG_EV_OPEN

3e9af 3 net.c:159:mg_connect            2577 0xffffffff udp://8.8.8.8:53
3e9b6 3 sock.c:146:mg_send              2577 0x37 0:0 -1 err 118
3e9bb 1 event.c:21:mg_error             2577 0x37 DNS send
3e9c0 2 net.c:157:mqtt_forwarders_timer => Connect to : consol_mqtt_forwarder.c=3fff4e98

3e9c8 1 event.c:21:mg_error             2576 0xffffffff DNS error
3e9ce 3 net.c:136:mg_close_conn         2577 0x37 closed
3e9d3 3 net.c:117:consol_mqtt_forwarder ^^ consol_mqtt_forwarder_handler ^^ MG_EV_CLOSE

3e9db 3 net.c:136:mg_close_conn         2576 0xffffffff closed
3e9e1 3 net.c:141:cloud_mqtt_forwarder_ ^^ cloud_mqtt_forwarder_handler ^^ MG_EV_CLOSE

3e9e8 3 net.c:136:mg_close_conn         2575 0xffffffff closed
3ea00 3 net.c:159:mg_connect            2578 0xffffffff mqtts://a2n53blisc65if-ats.iot.us-east-1.amazonaws.com:8883
3ea02 3 net.c:90:consol_mqtt_forwarder_ ^^ consol_mqtt_forwarder_handler ^^ MG_EV_OPEN

3ea09 3 net.c:159:mg_connect            2579 0xffffffff udp://8.8.8.8:53
3ea10 3 sock.c:146:mg_send              2579 0x37 0:0 -1 err 118
3ea15 1 event.c:21:mg_error             2579 0x37 DNS send
3ea1a 2 net.c:157:mqtt_forwarders_timer => Connect to : consol_mqtt_forwarder.c=3fff4ed8

3ea22 2 net.c:162:mqtt_forwarders_timer ## cloud_mqtt_forwarder.url=mqtts://a2n53blisc65if-ats.iot.us-east-1.amazonaws.com:8883

3ea2e 3 net.c:159:mg_connect            2580 0xffffffff mqtts://a2n53blisc65if-ats.iot.us-east-1.amazonaws.com:8883
3ea39 3 net.c:126:cloud_mqtt_forwarder_ ^^ cloud_mqtt_forwarder_handler ^^ MG_EV_OPEN
```

- My question is:  how to solve this problem ?

# Environment

- mongoose version: _7.8_
- Compiler/IDE and SDK:  VS CODE / ESP_IDF 
- Target hardware/board: ESP32
- Connectivity chip/module: WIFI

scaprile, 2023-05-23:
Looks like you either exhausted the stack, or the heap, or both.
It is hard (impossible) to tell without knowing what you are actually doing.
I also suggest you move to a more recent release.

OmarM17, 2023-05-24:
Here's a brief description of my project:
I'm using mg_timer to send periodic data to an AWS MQTT broker, mg_tls_opts to use a certificate for MQTT, and mg_snprintf to format my JSON messages for MQTT. I'm not using dynamic allocation.

I've been watching the heap size, and it continues to decrease every time an MG_EV_CLOSE event occurs, plus there are  aother error: (iobuf.c:39:mg_iobuf_resize 0->2048 )
```
942ca 3 net.c:136:mg_close_conn         3166 0xffffffff closed
942de 3 net.c:159:mg_connect            3169 0xffffffff mqtts://a2n53blisc65if-ats.iot.us-east-1.amazonaws.com:8883
942e0 3 net.c:90:consol_mqtt_forwarder_ ^^ consol_mqtt_forwarder_handler ^^ MG_EV_OPEN

942e7 3 net.c:159:mg_connect            3170 0xffffffff udp://8.8.8.8:53
942ee 3 sock.c:146:mg_send              3170 0x37 0:0 -1 err 118
942f3 1 event.c:21:mg_error             3170 0x37 DNS send
942f8 1 iobuf.c:39:mg_iobuf_resize      0->2048
942fc 1 iobuf.c:39:mg_iobuf_resize      0->2048
94300 1 iobuf.c:39:mg_iobuf_resize      0->2048
94305 1 iobuf.c:39:mg_iobuf_resize      0->2048
94309 1 iobuf.c:39:mg_iobuf_resize      0->2048
9430d 2 net.c:161:mqtt_forwarders_timer => Connect to : consol_mqtt_forwarder.c=3ffff1ec

94316 1 event.c:21:mg_error             3169 0xffffffff DNS error
9431b 3 net.c:136:mg_close_conn         3170 0x37 closed
94320 3 net.c:117:consol_mqtt_forwarder ^^ consol_mqtt_forwarder_handler ^^ MG_EV_CLOSE
```

Does that make sense or do you need more details?

scaprile, 2023-05-24:
> I'm not using dynamic allocation.

You are, you are using Mongoose

> I've been watching the heap size, and it continues to decrease every time an MG_EV_CLOSEDoes that make sense or do you need more details? event occurs

> Those msgs indicate you lost your connectivity but you may also have exhausted your heap or will do it as your code keeps pushing data and Mongoose can only pile it in the heap.

` MG_EV_CLOSE` means "Your connection has been closed, you no longer have an active connection, you are not able to send data"

> Does that make sense or do you need more details?

No, pardon my weakness, I'm not able to do what you are doing unless you show me what you are doing. Please reduce your code to the _**bare minimum possible shortest**_ piece of code that reproduces your issue, then attach it here.

OmarM17, 2023-05-25:
My project is based on mongoose [ESP32: UART Bridge](https://mongoose.ws/documentation/tutorials/esp32/uart-bridge/) example
I Modify it this way

main.c 
```
...
...
void app_main(void) {
  // Mount filesystem
  ...
  // WIFI connection
  ...

  // Wifi is Connected to the configured.. , now start HTTP server
    struct mg_mgr mgr;
    mg_mgr_init(&mgr);
    mgr.dnstimeout = 10000;
    mg_log_set(MG_LL_DEBUG);  // Set log level
    MG_INFO(("Mongoose v%s on %s", MG_VERSION, s_listening_url));
    mg_http_listen(&mgr, s_listening_url, gw_host_fn, &mgr);
    for (;;) mg_mgr_poll(&mgr, 10);  // Infinite event loop
    
}
```

net.c
```

...
/* * Global Variables * */
uint8_t NET_CONFIG_VERSION=0 ;
uint32_t GW_HOST_PERIODIC_UPDATE_INTERVAL = 30000 ;  // default : 30 sec 
struct mg_timer *periodic_update_timer=NULL ;
char *NET_ID_str;
char GW_WIFI_MAC[] = "XX:XX:XX:XX:XX:XX";
// MQTT
char CONSOL_MQTT_URL[]= "mqtts://a2n53blisc65if-ats.iot.us-east-1.amazonaws.com:8883";
char *consol_mqtt_forwarder_cert = "-----BEGIN CERTIFICATE-----\n"
"MIIDWTC...\n"
"-----END CERTIFICATE-----" ;//"console_cert.pem";

char *consol_mqtt_forwarder_key = "-----BEGIN RSA PRIVATE KEY-----\n"
"MIIEpA..\n"
...
"-----END RSA PRIVATE KEY-----";//"consol_key.pem";

char *consol_mqtt_forwarder_ca = "-----BEGIN CERTIFICATE-----\n"
"MIIDQ....\n"
...
"-----END CERTIFICATE-----";//"console_ca.pem";
...

// JSON messages buffer
char HOST_MSG_JSON_MSG[5000];

...
// HTTP request handler function
void gw_host_fn(struct mg_connection *c, int ev, void *ev_data,
                    void *fn_data) {
  if (ev == MG_EV_OPEN && c->is_listening) { // one time in the beginning, init function should be called here

    ...
    //- - - GET the WIFI MAC- - - - - - -
    unsigned char mac_base[6] = {0};
    esp_efuse_mac_get_default(mac_base);
    mg_snprintf(GW_WIFI_MAC, sizeof(GW_WIFI_MAC), "%02x:%02x:%02x:%02x:%02x:%02x", mac_base[0], mac_base[1], mac_base[2], mac_base[3], mac_base[4], mac_base[5]);
    MG_INFO(("wifi mac == %s \n\n",GW_WIFI_MAC));

    // init function
    GW_HOST_init(c->mgr); // mgr (mongoose manager) 
[... 233 lines cut ...]
                        " \"ADMIN_STATUS\":\"DISABLED\", \"STATUS\":\"DOWN\",  "
                        " \"UP_TIME\":\"UP_TIME\", \"PRIORITY\":\"3\", "
                        " \"UART_ID\":\"UART_ID\", \"PACKETS_SENT\":\"PACKETS_SENT\",  "
                        " \"BYTES_SENT\":\"BYTES_SENT\", \"PACKETS_RECEIVED\":\"PACKETS_RECEIVED\",  "
                        " \"BYTES_RECEIVED\":\"BYTES_RECEIVED\" } "
                    " } }",
                    json_gw_header,
                    GW_HOST_FIRMWARE_VER_str, GW_HOST_UP_TIME_str, 
                    CURRENT_NET_CONFIG_VERSION_str,GW_HOST_RAM_USAGE_str,
                    GW_HOST_STORAGE_USAGE_str,GW_HOST_CPU_USAGE_str, 
                    GW_HOST_TEMPERATURE_str,GW_SOC_MAC, GW_WIFI_MAC );
    // Publish to MQTT 
    if ( consol_mqtt_forwarder.c !=NULL)
    {
        mg_mqtt_pub(consol_mqtt_forwarder.c, mg_str(MQTT_CONTROL_TOPIC_str),mg_str(HOST_MSG_JSON_MSG) , 1, false);
        MG_INFO(("### consol_mqtt_forwarder:[GW_HOST_PERIODIC_UPDATE] Msg Pub to [%s] == %s \n",MQTT_CONTROL_TOPIC_str,HOST_MSG_JSON_MSG));
    }else{
        MG_INFO(("==X [GW_HOST_PERIODIC_UPDATE] not published .c=%d \n\n",consol_mqtt_forwarder.c));
    }

}
```

This part of code contain periodic message publication only, there is another part of forwarding UART data, Previously, both functionalities were functioning properly. However, to address issues related to repeated allocation and deallocation, I performed memory optimization by replacing dynamic arrays with static arrays, eliminating the need for unnecessary memory operations.
Unfortunately, after making these changes, neither of the them are sending.

scaprile, 2023-05-25:
TL:DR
Sorry, that is not
> the bare minimum possible shortest piece of code that reproduces your issue

I don't have the time to debug your code

OmarM17, 2023-05-25:
ok thanks, 

I believe that removing some details may have caused the bug in my code. As a beginner in using Mongoose, I feel that an expert with more experience could easily identify my mistakes by simply reviewing the code. I am hopeful that with their assistance, we can identify and resolve the issues I am facing.
Since this is the only community site available to me, I am relying on the help and expertise of its members.

closed 2023-05-25 by scaprile

dzobbe, 2023-05-26:
I have a similar situation. A workaround I found was to add a sleep(1) inside the infinite loop. And that make it work but at the cost of performance. I would really like to find the root cause of this. 
Data starts to arrive and at a certain point there is this DNS error which I don't think it is actually related to the DNS but more to some event loop / memory issue.  I will try to make a small test code to reproduce.

scaprile, 2023-05-26:
@dzobbe Well, Giovanni, the OP is (for some reason) looping every 10ms
`for (;;) mg_mgr_poll(&mgr, 10);`
You can certainly replace that 10 for a 1000 in your code, instead of adding a sleep() call, though I would rather like you to open your own issue and properly explain what you do and what you don't, what fails, why, and provide enough information for us to reproduce it and fix it.
While you do it, pay attention to the fact that, if you use MG_EV_POLL, it is not called every time the manager times out but every time the manager wakes up, that is, whenever it is ready to send or receive, or when it times out. Timers are OK.