cesanta/mongoose #2218: small remark in the documentation for `io_buffers`

llange, 2023-05-26:
Hello,

I couldn't find the repo for the docs so I'm posting here instead.
There are some possible improvements in the documentation for `io_buffers`, more specifically [mg_iobuf_del](https://mongoose.ws/documentation/#mg_iobuf_del) :
* The "Usage example" for `mg_iobuf_del` looks wrong (last line - `mg_iobuf_del` - has 4 parameters, while the function signature has 3)
* The [image](https://mongoose.ws/documentation/images/mg_iobuf_del.svg) associated with the usage example also seems incorrect : it doesn't seem to correspond with the proposed example ("hello" / "ho" vs "hi" / "").

Thanks in advance.

scaprile, 2023-05-26:
Thx, we'll take a look at it
Docs have been integrated into the website and no longer publicly accessible.

llange, 2023-05-26:
Thank you.

While I'm at it:
In the docs for https://mongoose.ws/documentation/#mg_print_ip and https://mongoose.ws/documentation/#mg_print_ip_port , the examples refer to a `struct mg_address` - I believe it should be instead a `struct mg_addr`.

Out of curiosity, what is the reason for having the docs outside of the sources ? Is there a will to have them not publicly accessible ? and if so, why ? Thanks for the insight.

llange, 2023-05-29:
Another one:
[`struct mg_mqtt_opts`](https://mongoose.ws/documentation/#struct-mg_mqtt_opts) in the documentation is not in sync with [the definition](https://github.com/cesanta/mongoose/blob/efd2db129f8d56875bec8913919649ed96f992d0/src/mqtt.h#L69-L84).
It looks like it has been updated between 7.9 and 7.10.

scaprile, 2023-05-29:
Yeah, sorry about that, 7.10 includes support for some MQTT5 fields

llange, 2023-05-29:
more fields yes, but also some have changed name. "will_" prefix removed.

llange, 2023-05-29:
Another one:
signature for functions [`mg_mqtt_pub()`](https://mongoose.ws/documentation/#mg_mqtt_pub), [`mg_mqtt_sub()`](https://mongoose.ws/documentation/#mg_mqtt_sub) have also changed and are not updated.
`mg_mqtt_disconnect()` and `mg_mqtt_pong()` are not documented.

scaprile, 2023-05-30:
> more fields yes, but also some have changed name. "will_" prefix removed.

Not exactly, the `will_` prefix corresponds to the last will messages and those were grouped; looks like the old comment stayed where it shouldn't.
We'll check this and fix it

llange, 2023-05-30:
> > more fields yes, but also some have changed name. "will_" prefix removed.
> 
> Not exactly, the `will_` prefix corresponds to the last will messages and those were grouped; looks like the old comment stayed where it shouldn't. We'll check this and fix it

Hmm, sorry, I did not get it. I understood that [`will_topic`, `will_message`, `will_qos` and `will_retain` in 7.9](https://github.com/cesanta/mongoose/blob/4236405b90e051310aadda818e21c811e404b4d8/src/mqtt.h#L28-L30) have been renamed to [`topic`, `message`, `qos` and `retain` in 7.10](https://github.com/cesanta/mongoose/blob/b379816178abdcd59135aa32f990a4b3bbbfb54b/src/mqtt.h#L73-L75), and do indeed relate to "last will messages".
At least it looks that it's [still handled that way in the code](https://github.com/cesanta/mongoose/blob/b379816178abdcd59135aa32f990a4b3bbbfb54b/src/mqtt.c#L273-L276), even if the name of those attributes does not imply "last will" any more.
Is it the case, or not ?

If not, so how should we specify the "last will" when connecting ?

scaprile, 2023-05-30:
Let me check that

scaprile, 2023-05-30:
You are right, don't know the reason why that prefix was removed; the login code uses them for last will.

scaprile, 2023-06-05:
Docs fixed and update, thank you for reporting !

closed 2023-06-05 by scaprile