cesanta/mongoose #2220: mqtt: handling of a `PUBLISH` packet

llange, 2023-05-29:
- My goal is: _to port a source code from mongoose 6.11 to 7.10 ; this source code is using MQTT and I'm trying to figure out if my interpretation of the source code is correct, and if mongoose source code is line with the spec._
- My actions were: _read our codebase source code, red mongoose 7.x source code, read spec,_
- My expectation was: _According to the spec ([v5](https://docs.oasis-open.org/mqtt/mqtt/v5.0/os/mqtt-v5.0-os.html#_Toc3901120) or [v3.1.1](http://docs.oasis-open.org/mqtt/mqtt/v3.1.1/os/mqtt-v3.1.1-os.html#_Toc398718041)) a `PUBLISH` packet is responded with a `PUBACK` packet if `qos == 1`, or a `PUBREC` packet if `qos == 2`_
- The result I saw: _In [the source code](https://github.com/cesanta/mongoose/blob/efd2db129f8d56875bec8913919649ed96f992d0/src/mqtt.c#L447-L459) it seems that the response is always responded with a `PUBACK` _
- My question is: _shouldn't we distinguish between the 2 different qos and chose the correct response packet accordingly?_

# Environment

- mongoose version: _6.11, 7.10_
- Compiler/IDE and SDK: ESP-IDF version _3.3.4, 5.0.1, ..._
- Target hardware/board: _ESP-32_
- Connectivity chip/module: _?_
- Target RTOS/OS (if applicable): _FreeRTOS_

# Details

Hello !

To set the record straight, I'm no expert in Mongoose, nor am I in MQTT neither.

When porting the codebase of https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3 from Mongoose 6.11 to 7.10, I've found an [instance of our code](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/646049a09e1b30cdff1923e6c83e9e88227f4970/vehicle/OVMS.V3/components/ovms_server_v3/src/ovms_server_v3.cpp#L134-L151) (MQTT client) trying to handle a `PUBLISH` message with the old 6.x API:
```
    case MG_EV_MQTT_PUBLISH:
      ESP_LOGV(TAG, "OvmsServerV3MongooseCallback(MG_EV_MQTT_PUBLISH)");
      ESP_LOGI(TAG, "Incoming message %.*s: %.*s", (int) msg->topic.len,
             msg->topic.p, (int) msg->payload.len, msg->payload.p);
      if (MyOvmsServerV3)
        {
        MyOvmsServerV3->IncomingMsg(std::string(msg->topic.p,msg->topic.len),
                                    std::string(msg->payload.p,msg->payload.len));
        }
      if (msg->qos == 1)
        {
        mg_mqtt_puback(nc, msg->message_id);
        }
      else if (msg->qos == 2)
        {
        mg_mqtt_pubrec(nc, msg->message_id);
        }
      break;
```

while trying to port this block to 7.10, I found that while `PUBLISH` commands seemed to be automatically handled in `mqtt_cb
()`, I could only find the handling corresponding to the `qos == 1` case in [the source code](https://github.com/cesanta/mongoose/blob/efd2db129f8d56875bec8913919649ed96f992d0/src/mqtt.c#L447-L459):
```
          case MQTT_CMD_PUBLISH: {
            MG_DEBUG(("%lu [%.*s] -> [%.*s]", c->id, (int) mm.topic.len,
                      mm.topic.ptr, (int) mm.data.len, mm.data.ptr));
            if (mm.qos > 0) {
              uint16_t id = mg_htons(mm.id);
              uint32_t remaining_len = sizeof(id);
              if (c->is_mqtt5) remaining_len += 1;

              mg_mqtt_send_header(c, MQTT_CMD_PUBACK, 0, remaining_len);
              mg_send(c, &id, sizeof(id));

              if (c->is_mqtt5) {
                uint16_t zero = 0;
                mg_send(c, &zero, sizeof(zero));
              }
            }
            mg_call(c, MG_EV_MQTT_MSG, &mm);
            break;
          }
```

missing is the `qos == 2` case.

Not being familiar with mongoose nor MQTT, I thought I could ask the question here : is this a real issue ? Or did I misread/misunderstood the spec ?

Thanks for any insight.

scaprile, 2023-05-29:
Can you please explain exactly what you are doing ? I get lost with client and server features, please detail your scenario, are you a client or a broker ? Do you want to publish with QoS=2 to a broker or receive with QoS=2 from a broker (if you are a client), or both ?
That piece of code would affect reception of PUB msgs, that is, broker -> client if you are a client or client -> broker if you are a broker.
I'm not familiar with 6.11, but 7.x can work as a client or a broker

llange, 2023-05-30:
Yes, sorry for partial context.

So our embedded system connects to a central server. We implement this in `ovms_server_v2` (proprietary protocol) or in `ovms_server_v3` (connecting to a MQTT broker).

So we have a 'client' role here, and connecting to an (unspecified) MQTT broker.

To this broker we publish a few things:
* ["metrics" with a QOS of 1 (retained values)](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/646049a09e1b30cdff1923e6c83e9e88227f4970/vehicle/OVMS.V3/components/ovms_server_v3/src/ovms_server_v3.cpp#L318-L321)
* [Notifications info (e.g.) with a QOS of 1](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/646049a09e1b30cdff1923e6c83e9e88227f4970/vehicle/OVMS.V3/components/ovms_server_v3/src/ovms_server_v3.cpp#L332-L334)
* [Notification data with a QOS of 2](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/646049a09e1b30cdff1923e6c83e9e88227f4970/vehicle/OVMS.V3/components/ovms_server_v3/src/ovms_server_v3.cpp#L389-L390)
* [Events with a QOS of 0](https://github.com/openvehicles/Open-Vehicle-Monitoring-System-3/blob/646049a09e1b30cdff1923e6c83e9e88227f4970/vehicle/OVMS.V3/components/ovms_server_v3/src/ovms_server_v3.cpp#L543-L544)
* etc...

So we, as a client, publish with all values of QOS (0, 1, 2) to the broker (client->broker).

With 6.11 it seems that most of the protocol handling had to be done by the user of mongoose (replying to the commands with the appropriate confirmation etc..., like in the case of the `PUBLISH` command).

With 7.10, it seems that - at least for the `PUBLISH` command - some of these acknowledgment is now done automatically by Mongoose.
(However it's not 100% of the protocol as, for example, we have to manage `PUBREL` or `PUBREC`, etc...)
(btw: I'd suggest that the documentation states what is, and what is not, managed by Mongoose)

So my question, relatively to this implementation of the `PUBLIC` handling in `mqtt_cb()`, is related to the different response that -per the spec- should be sent after a `PUBLIC` command, depending on the QOS.

scaprile, 2023-05-30:
Please see that I updated my response (Github sends you a mail with the first response and then you don't get the updates...)
I'll make sure the docs and tutorials reflect reality, though it will take some time; thx a lot for letting us know.
I'll see if we can ease some of the process, our current focus is elsewhere but we do thrive for user experience.

llange, 2023-05-30:
> Currently there is no support for handling QoS=2 messages. [The server example](https://github.com/cesanta/mongoose/tree/master/examples/mqtt-server) shows how you can run your own handler, so you can add it in your application. Mongoose is event-driven and QoS2 requires keeping state and storing the message, and we don't even do retransmissions in QoS1, the app has to handle them.
> 
> QoS0 has no ACK QoS1 requires an ACK. If the ACK gets lost, the publisher will resend the message and it may be published twice (or more) to subscribers QoS2 has a 4-way handshake to ensure a msg is only published once, so it uses different ACK msgs

Thanks for the information.
Handling QoS2 manually is fine (and that's what we were already doing with 6.11).

By "run your own handler", you're talking about handling `MQTT_CMD_*` in the event handler `MG_EV_MQTT_CMD` case (which is what I'm already doing) ?

If so, how one can "bypass" the default behaviour of [`mqtt_cb()`](https://github.com/cesanta/mongoose/blob/efd2db129f8d56875bec8913919649ed96f992d0/src/mqtt.c#L447-L459) - for the handling of `MQTT_CMD_PUBLISH` ? (The test `if (mm.qos > 0) {` will make this handler send a `MQTT_CMD_PUBACK` packet, which is NOT what I want when `qos == 2`.)

`mqtt_cb()` is registered as a protocol function (`pfn`) in the connection, but the rest of the function is OK and I wouldn't want to duplicate it if possible.
Besides this `mqtt_cb()` function is really needed as it is the one that fires `mg_call(c, MG_EV_MQTT_CMD, &mm);` which is the mechanism by which I'll be able to handle those commands...

Could you please elaborate how could I properly handle `MQTT_CMD_PUBLISH` without having `mqtt_cb()` in the way ; but still benefit from the rest of the function ?

scaprile, 2023-05-30:
> By "run your own handler", you're talking about handling `MQTT_CMD_*` in the event handler `MG_EV_MQTT_CMD` case (which is what I'm already doing) ?

Yes, you translated me correctly

"Bypass" is a confusing term, sorry about that, in your event handler function you should handle `MG_EV_MQTT_CMD` and parse the MQTT packet as shown in the server example linked above
It should work OK on sending as `mqtt_cb()` won't handle the incoming PUBREC. However, when the broker would send with QoS=2 it will get in the middle ("in the tail", actually, adding a PUBACK after your PUBREC).
I already started internal discussion, for now I think changing `qos > 0` to `qos == 1` will do to keep it out of the way.

llange, 2023-05-30:
Thanks for the perfectly clear answer.
I'm looking forward to the condition change in master so that I can test it.
Thanks again.

scaprile, 2023-05-31:
Let me know if that fixes your issue. Some cues on QoS1 and QoS2 implementation on the commit comments

PR #2229 (2023-06-02): MQTT QoS + v5
- Fix a bug when sending PUBACK/PUBREC in MQTT5: one extra byte sent and no props.
  - Note:   According to MQTT5 3.4.2-1: _The Client or Server sending the PUBACK packet MUST use one of the PUBACK Reason Codes [MQTT-3.4.2-1]. The Reason Code and Property Length can be omitted if the Reason Code is 0x00 (Success) and there are no Properties. In this case the PUBACK has a Remaining Length of 2._ (3.5.2-1: idem for PUBREC).
  - We don't actually need to do as we are doing now, sending empty props and reason code = 0, I assume that is for future expansion
- Add bare minimum (MQTT5-c [... cut]
closed 2023-06-02 by cpq