cesanta/mongoose #2257: WSS MQTT Broker on ESP32

JonRobo, 2023-06-15:
I want to add websocket secure features to my mqtt broker on the esp32. I am using V7.9 of Mongoose and esp idf v5.0.2
I now have an unsecured websocket MQTT broker working, I have followed all the tutorials and examples that I can find about any sort of secure server, but I keep receiving runtime errors.

When I try to setup the tls opts as in the examples

case MG_EV_ACCEPT:
struct mg_tls_opts opts = {
        .cert = "server.pem",     // Certificate PEM file
        .certkey = "server.pem",  // This pem contains both cert and key
    };
I receive an error from mbedTLS of 0x2800, which seems to mean MBEDTLS_ERR_X509_BAD_INPUT_DATA.

I then try to pass this parameter in just like I do on my esp idf mqtt client where I'd define global variables
extern const uint8_t cert_pem_start[] asm("_binary_cert_pem_start");
extern const uint8_t key_pem_start[] asm("_binary_key_pem_start");

Then pass them in as 
struct mg_tls_opts opts = {
                    .cert = (const char *)cert_pem_start,
                    .certkey = (const char *)key_pem_start,
                };
I no longer receive error 0x2800 but now 0x7F00 MBEDTLS_ERR_SSL_ALLOC_FAILED (from mbedtls ssl.h)

In order to get this far I cloned the mbedtls repository into the src folder of mongoose.
I then added the following lines to the mongoose folder's CMakeLists file

target_compile_definitions(${COMPONENT_LIB} PRIVATE MG_ENABLE_MBEDTLS=1)
target_compile_definitions(${COMPONENT_LIB} PRIVATE MG_ARCH=MG_ARCH_ESP32)
target_compile_definitions(${COMPONENT_LIB} PRIVATE MG_ENABLE_FILE=0)
target_compile_definitions(${COMPONENT_LIB} PRIVATE MBEDTLS_PEM_PARSE_C=1)
target_compile_definitions(${COMPONENT_LIB} PRIVATE MBEDTLS_RSA_C=1)

I created the certificate and key using openssl using he command openssl req  -nodes -new -x509  -keyout key.pem -out cert.pem
from the TLS tutorial https://mongoose.ws/documentation/tutorials/tls/

I am running an MQTT client that attempts to connect to this WSS MQTT broker, it is using the esp idf mqtt client library and was tested on the wss://mqtt.eclipseprojects.io:443/mqtt broker and worked just as expected.

I feel like I am getting close but there is clearly some confusion on how these certificates can be setup and passed in to initialize and run TLS.

Any help would be greatly appreciated

scaprile, 2023-06-15:
Honestly, I don't get what you do and you don't do, what is your actual issue and how are we related to that.
If you have MbedTLS errors, look for their code meaning and configure it accordingly; the version used is in the ESP-IDF. You have to configure THAT version in the IDF.
The architecture you don't have to define, it is detected at compile time unless you are not using the SDK, in which case we can't help at all.
If you are a server and your certificate is in PEM format (instead of DER), yes, you have to define PEM_PARSE and probably also enable BASE64; but that is MbedTLS, not Mongoose.
Instead of debugging both a server and a client at the same time, I would use a known-good client like mosquitto_pub and _sub
If your certificate is self-signed, you need to provide the CA cert to the client, otherwise it may reject it, but that is also not Mongoose.

JonRobo, 2023-06-15:
Thank you for your prompt response.

1 thing that you can confirm for me is the exact syntax needed to pass in the PEM files to the mg_tls_opts struct using the ESP32

struct mg_tls_opts opts = {
.cert = (const char *)cert_pem_start,
.certkey = (const char *)key_pem_start,
};

or

struct mg_tls_opts opts = {
.cert = "cert.pem",
.certkey = "key.pem", 
};

I have not found an example of using TLS with the esp32.
I was looking at the 2 examples found in the mongoose repo here https://github.com/cesanta/mongoose/tree/master/examples/esp32

Perhaps more secure client/server examples would help. Having to install mbedtls as an extra step as well as trying to figure out the configurations in the CMake took up a lot of time

closed 2023-06-15 by scaprile