cesanta/mongoose #2263: Small memory-leak in 'unit_test.exe'

gvanem, 2023-06-20:
Running `unit_test.exe` on Win-10 with the excellent [VisualLeakDetector](https://github.com/KindDragon/vld) built in, 
shows a small 256 byte leak from `mg_iobuf_resize()`:
```
WARNING: Visual Leak Detector detected memory leaks!
---------- Block 236 at 0x000000004A2580B0: 256 bytes ----------
  Leak Hash: 0xB794F29A, Count: 1, Total 256 bytes
  Call Stack (TID 16476):
    ucrtbase.dll!_calloc_base()
    F:\MinGW32\src\inet\Web\Mongoose\src\iobuf.c (28): unit_test.exe!mg_iobuf_resize() + 0xE bytes
    F:\MinGW32\src\inet\Web\Mongoose\src\printf.c (42): unit_test.exe!mg_pfn_iobuf() + 0x16 bytes
    F:\MinGW32\src\inet\Web\Mongoose\src\fmt.c (206): unit_test.exe!mg_vxprintf() + 0x15 bytes
    F:\MinGW32\src\inet\Web\Mongoose\src\printf.c (73): unit_test.exe!mg_mprintf()
    F:\MinGW32\src\inet\Web\Mongoose\test\unit_test.c (1974): unit_test.exe!fn1() + 0x12 bytes
    F:\MinGW32\src\inet\Web\Mongoose\src\event.c (22): unit_test.exe!mg_error() + 0x1F bytes
    F:\MinGW32\src\inet\Web\Mongoose\src\dns.c (142): unit_test.exe!dns_cb()
    F:\MinGW32\src\inet\Web\Mongoose\src\event.c (12): unit_test.exe!mg_call()
    F:\MinGW32\src\inet\Web\Mongoose\src\sock.c (631): unit_test.exe!mg_mgr_poll() + 0x1B bytes
    F:\MinGW32\src\inet\Web\Mongoose\test\unit_test.c (1989): unit_test.exe!test_dns_error() + 0x1A bytes
    F:\MinGW32\src\inet\Web\Mongoose\test\unit_test.c (2044): unit_test.exe!test_dns() + 0x13 bytes
    F:\MinGW32\src\inet\Web\Mongoose\test\unit_test.c (3099): unit_test.exe!main()
    D:\a\_work\1\s\src\vctools\crt\vcstartup\src\startup\exe_common.inl (288): unit_test.exe!__scrt_common_main_seh() + 0x22 bytes
  Data:
    44 4E 53 20    74 69 6D 65    6F 75 74 00    00 00 00 00     DNS.time out.....
    00 00 00 00    00 00 00 00    00 00 00 00    00 00 00 00     ........ ........
    00 00 00 00    00 00 00 00    00 00 00 00    00 00 00 00     ........ ........
    00 00 00 00    00 00 00 00    00 00 00 00    00 00 00 00     ........ ........
    00 00 00 00    00 00 00 00    00 00 00 00    00 00 00 00     ........ ........
    00 00 00 00    00 00 00 00    00 00 00 00    00 00 00 00     ........ ........
    00 00 00 00    00 00 00 00    00 00 00 00    00 00 00 00     ........ ........
    00 00 00 00    00 00 00 00    00 00 00 00    00 00 00 00     ........ ........
    00 00 00 00    00 00 00 00    00 00 00 00    00 00 00 00     ........ ........
    00 00 00 00    00 00 00 00    00 00 00 00    00 00 00 00     ........ ........
    00 00 00 00    00 00 00 00    00 00 00 00    00 00 00 00     ........ ........
    00 00 00 00    00 00 00 00    00 00 00 00    00 00 00 00     ........ ........
    00 00 00 00    00 00 00 00    00 00 00 00    00 00 00 00     ........ ........
    00 00 00 00    00 00 00 00    00 00 00 00    00 00 00 00     ........ ........
    00 00 00 00    00 00 00 00    00 00 00 00    00 00 00 00     ........ ........
    00 00 00 00    00 00 00 00    00 00 00 00    00 00 00 00     ........ ........

Visual Leak Detector detected 1 memory leak (256 bytes).
Largest number used: 12583344 bytes.
Total allocations: 23675468 bytes.
Visual Leak Detector is now exiting.
```

(disregard the line-numbers in `unit-test.c` above; I've done some changes unrelated to this leak).

I fail to see how the code:
```c
size_t mg_vsnprintf(char *buf, size_t len, const char *fmt, va_list *ap) {
  struct mg_iobuf io = {(uint8_t *) buf, len, 0, 0};  << !! here
  size_t n = mg_vxprintf(mg_putchar_iobuf_static, &io, fmt, ap);
  if (n < len) buf[n] = '\0';
  return n;
}
```

should plug this leak with a local `struct mg_iobuf`.

## Environment

- mongoose version: 7.1 (`git master`)
- Compiler: MSVC 19.37.32705 for x64
- Target hardware/board: Windows-10.

cpq, 2023-06-30:
There is `mg_mprintf ` call in the backtrace, which uses malloc.
I believe this is https://github.com/cesanta/mongoose/blob/0c7e6b548f4207cc272715f9c55eb56cbe7a0539/test/unit_test.c#L1924

DNS error string gets allocated in the handler and never released

gvanem, 2023-06-30:
> - maybe because the test fails?

I'm getting `SUCCESS. Total tests: 1163`. Isn't that a success?

gvanem, 2023-06-30:
> If a test succeeds, buf should be release and there should be no leak

That's not where it leaks according to VLD. It's an internal buffer in `mg_vsnprintf()`:
```
 Call Stack (TID 16476):
    ucrtbase.dll!_calloc_base()
    F:\MinGW32\src\inet\Web\Mongoose\src\iobuf.c (28): unit_test.exe!mg_iobuf_resize() + 0xE bytes
    F:\MinGW32\src\inet\Web\Mongoose\src\printf.c (42): unit_test.exe!mg_pfn_iobuf() + 0x16 bytes
    F:\MinGW32\src\inet\Web\Mongoose\src\fmt.c (206): unit_test.exe!mg_vxprintf() + 0x15 bytes
    F:\MinGW32\src\inet\Web\Mongoose\src\printf.c (73): unit_test.exe!mg_mprintf()
    F:\MinGW32\src\inet\Web\Mongoose\test\unit_test.c (1974): unit_test.exe!fn1() + 0x12 bytes
    F:\MinGW32\src\inet\Web\Mongoose\src\event.c (22): unit_test.exe!mg_error() + 0x1F bytes
    F:\MinGW32\src\inet\Web\Mongoose\src\dns.c (142): unit_test.exe!dns_cb()
```

See above.

cpq, 2023-07-01:
No it is not vnsprintf.
1. dns.c calls mg_error: https://github.com/cesanta/mongoose/blob/0c7e6b548f4207cc272715f9c55eb56cbe7a0539/src/dns.c#L146
2. mg_error calls mg_vsnprintf() (innocent, and not part of this backtrace, line 18). Then it calls MG_EV_ERROR user handler, line 22: https://github.com/cesanta/mongoose/blob/0c7e6b548f4207cc272715f9c55eb56cbe7a0539/src/event.c#L22
3. User handler leaks by calling mprintf: https://github.com/cesanta/mongoose/blob/0c7e6b548f4207cc272715f9c55eb56cbe7a0539/test/unit_test.c#L1924

cpq, 2023-07-10:
@gvanem any progress on this?

gvanem, 2023-07-10:
@cpq What do you want **me** to do about this?

gvanem, 2023-07-18:
@cpq I finally had some time for this issue. But I'm amazed you didn't see the same leak with *valgrind* or whatever.

But trying your patches above, gave more problems in Windows since `test_dns_error()` now fails with:
```
1859c9f9 3 net.c:187:mg_listen          1 0x1e4 tcp://0.0.0.0:0
1859c9f9 3 net.c:164:mg_connect         2 0xffffffff http://google.com
1859c9f9 3 net.c:164:mg_connect         3 0xffffffff tcp://0.0.0.0:0
1859ca19 1 event.c:20:mg_error          2 0xffffffff DNS error
FAILURE test/unit_test.c:2034: *(void **) fn_data == NULL
```

since the port is already in "*use*" from a previous use the same port (?).
Winsock2 needs some time for it to be re-used I think; it's *lingering*. And the leak is still the same.

cpq, 2023-07-19:
So I guess this would be a fix:

```c
static void fn1(struct mg_connection *c, int ev, void *ev_data, void *fn_data) {
  if (ev == MG_EV_ERROR) {
    free(*(char **) fn_data);
    *(char **) fn_data = mg_mprintf("%s", (char *) ev_data);
    MG_INFO(("fn_data: '%s'", *(char **)fn_data));
  }
  (void) c;
}
```

commit 17f37381 (2023-08-17): Fix #2263 - allow multiple errors in DNS handler
commit 88e3017e (2023-08-17): Merge pull request #2348 from cesanta/dns

Fix #2263 - allow multiple errors in DNS handler
closed 2023-08-17 by cpq