cesanta/mongoose #2269: Provide Security Issue Reporting Functionality

todb, 2023-06-22:
How do I disclose security vulnerabilities to this project? The [README](https://github.com/cesanta/mongoose#security) states this project is familiar with vulnerability reporting, but doesn't actually provide a mechanism. There's no [security policy](https://github.com/cesanta/mongoose/security), there's nothing at https://mongoose.ws/.well-known/security.txt, and I don't see anything appropriate on the contact page.

Feel free to reply here, or better, populate mongoose/security or .well-known/security.txt, or worse, write to me directly at todb@hugesuccess.org and I can get the details over. Thanks!

scaprile, 2023-06-22:
- https://mongoose.ws/ -> Contact -> https://mongoose.ws/contact/
  - Fill the form
  - Send email to support@cesanta.com
- https://cesanta.com/ -> Contact -> https://cesanta.com/contact.html

closed 2023-06-22 by scaprile

todb, 2023-06-22:
Seems odd to direct vuln reports to a regular support queue, but you're the boss! Thanks!

scaprile, 2023-06-22:
> Seems odd to direct vuln reports to a regular support queue, but you're the boss! Thanks!

I'm not the boss, but I know that email account is read by trusted personnel, including the boss...