cesanta/mongoose #2280: Mulithreading Clarification

kujukuju, 2023-06-29:
- My goal is: get clarification about the multithreading support
- My actions were: read through a bunch of issues and previous commits
- My expectation was: N/A
- The result I saw: N/A
- My question is: 

Hi, I saw a lot of changes in the history of multithreading in mongoose. I was looking to implement a multithreaded http server and I enjoy the mongoose API.

I'm not very well versed on how low level http servers operate, but I was surprised to see that the mongoose multithreading support is basically offloading the actual data processing and memory allocation into separate threads.

Then at the end of the process, the data gets pushed back onto a stack and the actual socket data gets sent out __back on the main thread__.

My expectation is that this isn't how most web services would work? For example, if you have a server that you just need to scale up linearly with cores, and you're just sending really basic HTML data back to each user, this multithreading implementation might actually significantly slow down your server, right?

Could you please let me know if my understanding here is wrong?

Otherwise, is there any way to hack multithreading support into mongoose myself? I see that some people did this in previous issues in the past by messing with the source code and also using mg_broadcast (which no longer exists), as well as just straight up writing to sockets. I think processing all incoming events on a single main thread makes sense, but I was very surprised to see that the data is __also__ sent back out on that same main thread.

I think if there's some way to send my data out per socket on whatever arbitrary thread that would likely solve all my concerns.

Please let me know if I misunderstood anything!

scaprile, 2023-06-29:
Next time, please start from the current documentation, and follow our tutorials, it will save you a lot of time

From [the docs](https://mongoose.ws/documentation/#2-minute-integration-guide)
> NOTE: Since Mongoose's core is not protected against concurrent accesses, make sure that all mg_* API functions are called from the same thread or RTOS task.

If you absolutely need multi-threading, take the time to go through the user guide (linked above) and [tutorials](https://mongoose.ws/tutorials/), there is one that shows you how to do [multi-threading](https://mongoose.ws/tutorials/multi-threaded/).

Mongoose is an embedded web server

closed 2023-06-29 by scaprile

kujukuju, 2023-06-29:
Hey @scaprile thanks for your reply. I did see that in the docs. That multi threading example is actually why I have this question and what I was referring to.

The way it's "multi threaded" is that it's off loading the request work to separate threads, then the data gets pushed back onto the main thread to be sent by the OS.

My questions are:
1. If you're just sending static data that's loaded in memory, wouldn't this be slower and more expensive than a single threaded version since there's no work to do? It's scaling up the potential to do more work, not to receive more requests, right?
2. I don't know enough about how web servers usually work at a low level, but am I just wrong and it's normal for multi threaded web servers to receive and send data from a single thread? Or is this not true multi threading when referring to web servers?

scaprile, 2023-06-29:
Mongoose is not multi-threaded, and it always receives requests from the network, the OS takes care of that, Mongoose just listens on a socket and every new connection opens a new socket. Every socket, though, will be on the same thread.
Usually you don't need more threads to handle more connections, the highest workload is in processing the requests themselves, this is an embedded web server for embedded systems.

kujukuju, 2023-06-29:
That answers my questions. Thanks!