cesanta/mongoose #2287: MQTT Issue using STM32F767

BrunoSilvaUM, 2023-07-04:
Hi,
My goal is to be able to connect through ethernet to a mosquitto broker running on a Windows machine using an STM32F767 Nucleo Board.
 I've configured my board with STM32CubeMX activating Ethernet, FreeRTOS_CMSIS_V2 and RNG. 
Then I integrated  mongoose .c and .h files and added the mongoose_custom.h. 
I'm running a mosquitto broker in the host machine

**Here's the code I'm using:**
```
#ifndef __MONGOOSE_CUSTOM_H__
#define __MONGOOSE_CUSTOM_H__

#if defined(MG_ARCH)
#define MG_STMPACK_ARCH MG_ARCH
#undef MG_ARCH
#elif defined(__GNUC__)
#define MG_STMPACK_ARCH MG_ARCH_NEWLIB
#endif

#define MG_ENABLE_PACKED_FS  1

#define MG_ENABLE_CUSTOM_MILLIS  0

#define MG_ENABLE_MBEDTLS  0

#define MG_ARCH MG_ARCH_CMSIS_RTOS2

#define MG_STMPACK_NET 0

#define MG_ENABLE_CUSTOM_RANDOM  1

// Translate to Mongoose macros
#if MG_STMPACK_NET == 0
#define MG_ENABLE_TCPIP 1
#elif MG_STMPACK_NET == 1
#define MG_ENABLE_LWIP 1
#elif MG_STMPACK_NET == 2
#define MG_ENABLE_FREERTOS_TCP 1
#elif MG_STMPACK_NET == 3
#define MG_ENABLE_RL 1
#endif

#if MG_ENABLE_PACKED_FS
#define MG_ENABLE_FILE 0
#endif

#endif
```

**Flags defined on CubeMX**

[image]

[image]

**Main task created with stack size of 2048** - Defines a staticIP  on my target: 10.56.32.103
```
void StartTask02(void *argument)
{
  /* USER CODE BEGIN StartTask02 */
	  struct mg_mgr mgr;

	  mg_mgr_init(&mgr);

	  struct mg_tcpip_driver_stm32_data driver_data = {.mdc_cr = 4};
	  	struct mg_tcpip_if mif = {.mac = GENERATE_LOCALLY_ADMINISTERED_MAC(),
	  	                          .driver = &mg_tcpip_driver_stm32,
	  	                          .driver_data = &driver_data,
								  .ip = 0x6720380a, //10.56.32.102
								  .mask = 0x000000FF, // 255.0.0.0 
	  							  };
	  	mg_tcpip_init(&mgr, &mif);

	  	mg_timer_add(&mgr, 3000, MG_TIMER_REPEAT | MG_TIMER_RUN_NOW, timer_fn, &mgr);

  /* Infinite loop */
	for(;;)
	{
		mg_mgr_poll(&mgr, 1000);
		HAL_GPIO_TogglePin(GPIOB, GPIO_PIN_14);
		osDelay(200);
	}
	mg_mgr_free(&mgr);
  /* USER CODE END StartTask02 */
}
```
**Other functions:**

```
void mg_random(void *buf, size_t len) {  // Use on-board RNG
  extern RNG_HandleTypeDef hrng;
  for (size_t n = 0; n < len; n += sizeof(uint32_t)) {
    uint32_t r;
    HAL_RNG_GenerateRandomNumber(&hrng, &r);
    memcpy((char *) buf + n, &r, n + sizeof(r) > len ? len - n : sizeof(r));
  }
}
```

```
static void fn(struct mg_connection *c, int ev, void *ev_data, void *fn_data) {
  if (ev == MG_EV_OPEN) {
    MG_INFO(("%lu CREATED", c->id));
    // c->is_hexdumping = 1;
  } else if (ev == MG_EV_ERROR) {
    // On error, log error message
    MG_ERROR(("%lu ERROR %s", c->id, (char *) ev_data));
  } else if (ev == MG_EV_CONNECT) {
    // If target URL is SSL/TLS, command client connection to use TLS
//    if (mg_url_is_ssl(s_url)) {
//      struct mg_tls_opts opts = {.ca = "ca.pem"};
//      mg_tls_init(c, &opts);
//    }
  } else if (ev == MG_EV_MQTT_OPEN) {
    // MQTT connect is successful
    struct mg_str subt = mg_str(s_sub_topic);
    struct mg_str pubt = mg_str(s_pub_topic), data = mg_str("hello");
    MG_INFO(("%lu CONNECTED to %s", c->id, s_url));
    struct mg_mqtt_opts sub_opts;
    memset(&sub_opts, 0, sizeof(sub_opts));
    sub_opts.topic = subt;
    sub_opts.qos = s_qos;
    mg_mqtt_sub(c, &sub_opts);
    MG_INFO(("%lu SUBSCRIBED to %.*s", c->id, (int) subt.len, subt.ptr));
    struct mg_mqtt_opts pub_opts;
    memset(&pub_opts, 0, sizeof(pub_opts));
    pub_opts.topic = pubt;
    pub_opts.message = data;
    pub_opts.qos = s_qos, pub_opts.retain = false;
    mg_mqtt_pub(c, &pub_opts);
    MG_INFO(("%lu PUBLISHED %.*s -> %.*s", c->id, (int) data.len, data.ptr,
             (int) pubt.len, pubt.ptr));
  } else if (ev == MG_EV_MQTT_MSG) {
    // When we get echo response, print it
    struct mg_mqtt_message *mm = (struct mg_mqtt_message *) ev_data;
    MG_INFO(("%lu RECEIVED %.*s <- %.*s", c->id, (int) mm->data.len,
             mm->data.ptr, (int) mm->topic.len, mm->topic.ptr));
  } else if (ev == MG_EV_CLOSE) {
    MG_INFO(("%lu CLOSED", c->id));
    s_conn = NULL;  // Mark that we're closed
  }
  (void) fn_data;
}

static void timer_fn(void *arg) {
  struct mg_mgr *mgr = (struct mg_mgr *) arg;
  struct mg_mqtt_opts opts = {.clean = true,
                              .qos = s_qos,
                              .topic = mg_str(s_pub_topic),
                              .version = 4,
                              .message = mg_str("bye")};
  if (s_conn == NULL) s_conn = mg_mqtt_connect(mgr, s_url, &opts, fn, NULL);
}
```

**Global Variables:**
```
static const char *s_url = "10.56.32.102:1883";  //Broker IP
static const char *s_sub_topic = "MQTT/Sub/Test";     // Publish topic
static const char *s_pub_topic = "MQTT/123/Pub";  // Subscribe topic
static int s_qos = 0;                             // MQTT QoS
static struct mg_connection *s_conn;              // Client connection
```

**Issue**
I can't get a connection with the broker. The output on console is:

```
83f    2 tcpip.c:178:onstatechange      READY, IP: 10.56.32.103

845    2 tcpip.c:179:onstatechange             GW: 0.0.0.0

84a    2 tcpip.c:180:onstatechange            MAC: 02:24:19:5e:34:30

c39    2 main.c:124:fn                  2 CREATED

21eb5  1 event.c:21:mg_error            2 0xffffffff keepalive

21eba  1 main.c:128:fn                  2 ERROR keepalive

21ec0  2 main.c:160:fn                  2 CLOSED
```

I've disabled Windows Firewall, tried activating ICS on Windows network sharing center and it still doesn't work.
I've tested this on linux and it works fine.
Is there any compiler flag that I've missed or any windows setting needed? 

** Environment**

- mongoose version: Latest available on Github
-  IDE: STM32CubeIDE  compiler: GNU11 ISO C11 + gnu extensions
- Target hardware/board: STM32F767ZI NUCLEO
- Connectivity chip/module: LAN8742
- Target RTOS/OS (if applicable): FREERTOS_CMSISV2

Thank you!
Best Regards

scaprile, 2023-07-04:
We don't have a driver for the F767, I remember I told you that on your question, we don't claim that works as we did not test it. We have drivers for F746 and H743
0.0.0.0 is not a valid gateway address; if you want to make a stab at it, properly configure IP addresses, run your debugger, sniff your network. The most likely reason is that the MAC controller is different and our driver does not work, but you may also not have connected the board properly. If you don't even know if you have the TCP/IP stack up and running (ping) there is no point in trying to debug MQTT
Did you notice you are delaying Mongoose operation by 200ms each loop ? Please check your code against working examples
>  mg_mgr_poll(&mgr, 1000);
>  HAL_GPIO_TogglePin(GPIOB, GPIO_PIN_14);
>  **osDelay(200);** <-- ???

We don't know how to configure your Windows machine nor your Mosquitto broker

BrunoSilvaUM, 2023-07-04:
The code worked in a linux environment. Wouldn't that mean that the drivers work for this board as well? And that the problem is just at the tcp communication?

Enviado de Outlook para Android<https://aka.ms/AAb9ysg>

scaprile, 2023-07-04:
> The code worked in a linux environment. Wouldn't that mean that the drivers work for this board as well? And that the problem is just at the tcp communication?

I don't remember reading that in this issue.
Anyway, if that is the case, we are happy to know that the driver works.
Check you have an Ethernet link (Do you have a switch ? How do you connect ? Etc)
Properly configure IP addressing
Remove that unnecessary delay instruction, and google for how to configure your Windows machine and Mosquitto on that, we can't help you with that.

closed 2023-07-04 by scaprile

BrunoSilvaUM, 2023-07-06:
Hello again,

I can now get an IP and ping it properly. When trying to connect to a public mqtt broker (broker.hivemq.com) I get this error:

[image]

What could it be about?

br,

scaprile, 2023-07-06:
That log is telling you that the DNS can't be reached.
You have what seems to be a routing problem, a misconfiguration, a misconnection.
You should know how to connect devices to your network. You should know basic routing. You should know how to use a network sniffer. Hence, you should know how to solve this; I don't know your scenario, you don't describe it, hence I don't know what you connect with what, who is who, what is your network, whether you have a firewall or not, etc. You do.