cesanta/mongoose #2288: mg_check_ip_acl signature no longer fits to struct mg_addr

jcorporation, 2023-07-04:
I get following compiler error with mongoose 7.11, with mongoose 7.10 it works:

```
myMPD/src/web_server/web_server.c: In function ‘check_acl’:
myMPD/src/web_server/web_server.c:397:58: error: passing argument 2 of ‘mg_check_ip_acl’ makes integer from pointer without a cast [-Werror=int-conversion]
  397 |     int acl_result = mg_check_ip_acl(mg_str(acl), nc->rem.ip);
      |                                                   ~~~~~~~^~~
      |                                                          |
      |                                                          uint8_t * {aka unsigned char *}
In file included from myMPD/src/web_server/utility.h:10,
                 from myMPD/src/web_server/web_server.h:11,
                 from myMPD/src/web_server/web_server.c:9:
myMPD/dist/mongoose/mongoose.h:861:49: note: expected ‘uint32_t’ {aka ‘unsigned int’} but argument is of type ‘uint8_t *’ {aka ‘unsigned char *’}
  861 | int mg_check_ip_acl(struct mg_str acl, uint32_t remote_ip);
```

It seems commit dd32deb2ad88133d48bf8c837dd6ee7d6ffc7e51 changes the structure of mg_addr and the function `mg_check_ip_acl` was not changed accordingly.

Code:
```
struct mg_addr {
  uint8_t ip[16];  // Holds IPv4 or IPv6 address, in network byte order
  uint16_t port;   // TCP or UDP port in network byte order
  bool is_ip6;     // True when address is IPv6 address
};

int mg_check_ip_acl(struct mg_str acl, uint32_t remote_ip);
```

scaprile, 2023-07-04:
> It seems commit [dd32deb](https://github.com/cesanta/mongoose/commit/dd32deb2ad88133d48bf8c837dd6ee7d6ffc7e51) changes the structure of mg_addr and the function `mg_check_ip_acl` was not changed accordingly.

Yes, we compacted `struct mg_addr` but did not think changing `mg_check_ip_acl` was required.
We'll discuss that

cpq, 2023-07-05:
I think the signature should be changed to this:
```c
bool mg_check_ip_acl(struct mg_str acl, struct mg_addr);
```

This way it may work for v6 addresses, too.

jcorporation, 2023-07-05:
> This way it may work for v6 addresses, too.

This would be great!

PR #2289 (2023-07-05): update `mg_check_ip_acl`
closes #2288
closed 2023-07-05 by scaprile