cesanta/mongoose #2295: Publisher and Subscriber in the same MQTT client

nitol-saha, 2023-07-07:
I want to implement 2 clients with feature of publisher and subscriber in the same client.

Client 1: Publisher will send data in every 1 second and subscriber will listen for any incoming message in another topic.

Client 2: Subscriber will receive any data and as soon as it receives the data it will publish the data to another topic.

Can you provide me with any examples? I am new to this library.

scaprile, 2023-07-07:
You have plenty of examples and tutorials; what you describe is no different from what you can learn from them.
Read a tutorial, see the example, learn, do what you want to do.
https://mongoose.ws/documentation/#mqtt
https://mongoose.ws/documentation/tutorials/timers/

closed 2023-07-07 by scaprile

nitol-saha, 2023-07-20:
Thank you for the previous links. I got the project working. However, I have some general questions about the library to find compatibility for my project. If you can answer the following questions that would be really helpful:

- Is the library compatible or can be made compatible with ACE (Adaptive Communication Environment) (http://www.dre.vanderbilt.edu/~schmidt/ACE-overview.html)?

- Is there any dynamic memory allocation (new, delete)?

- Does the library have threading model or single thread execution?

- How are the exception handled? (Exception handling should not throw and end execution)

scaprile, 2023-07-20:
ACE: I'm sorry, I'm terribly lazy, if you read the doc and ask for specifics, you'll get answers
Dynamic memory allocation: This is C. Mongoose uses it, the source code is open, you can see it
Threading: [The docs](https://mongoose.ws/documentation/#2-minute-integration-guide) are clear, did you read them ? 
> NOTE: Since Mongoose's core is not protected against concurrent accesses, make sure that all mg_* API functions are called from the same thread or RTOS task.
Exceptions: This is not Python, nor JS, nor C++. Again, this is C, the source code is open, you can see it