cesanta/mongoose #2299: it will be packet loss,when use websocket send bin type data for a long time use esp32.

xu18838022837, 2023-07-13:
- My goal is: It will be packet loss so that data can't be send,when use websocket send bin type data for a long time.I have a question here,Did you run a long delivery test use esp32 board？
- My actions were: use ws client mg_ws_send op is WEBSOCKET_OP_BINARY,per 30ms send 1024KB bin type data
- My expectation was: I think you need fix bug in esp32.
- The result I saw: 
- 
- My question is: it will be packet loss,when use websocket send bin type data for a long time.

# Environment

- mongoose version: 7.8/7.11
- Compiler/IDE and SDK: ESP-IDF release/v5.1
- Target hardware/board: esp32
- Connectivity chip/module: _?_
- Target RTOS/OS (if applicable):esp-idf/freertos

xu18838022837, 2023-07-13:
[image]

scaprile, 2023-07-13:
I'm sorry, I don't understand what you try to mean. Please fill the template properly, and describe what you do in a clear way.
Check your code, you may be running out of memory and causing the OS to fail.
If you need to open the issue again, please do it with all proper data so we can reproduce what you are doing, attach the minimum possible piece of code that reproduces your issue, our code of course works and it does it well.

closed 2023-07-13 by scaprile