cesanta/mongoose #2300: SO_REUSEADDR no longer working on newer Linux kernels

joeyparrish, 2023-07-13:
- My goal is: to start and stop an embedded web server in unit tests for another project
- My actions were: run Mongoose according to docs
- My expectation was: second and subsequent server instances could bind to the same port
- The result I saw: errno 98 (EADDRINUSE - Address already in use)

I found an explanation and a solution to this.  Since Linux kernel 3.9, both SO_REUSEADDR and SO_REUSEPORT have to be set as socket options.  I'm going to assume this will not be controversial and send a PR to address it.

# Environment

- mongoose version: 818f99b286df9668803085e9d86fcf7f140aa02e (`master` as of today)
- Compiler/IDE and SDK: gcc 11.3.0 on Ubuntu 22.04
- Target hardware/board: N/A
- Connectivity chip/module: N/A
- Target RTOS/OS (if applicable): N/A

joeyparrish, 2023-07-13:
See also https://stackoverflow.com/a/25193462/2673491

scaprile, 2023-07-13:
We close and start again servers several times per day when developing. Can you please describe under what conditions you experience that ? Of course c->fd holds your listening socket and you are welcome to change its settings as you wish.
I think you are experiencing the fact that a socket is usually kept open for a while after closure, in case more data arrives.

joeyparrish, 2023-07-14:
You know, I think somewhere between filing this issue and finishing/merging the embedded server in my project, I must have fixed a bug in my own code to do with cleanup.  If I revert my changes to mongoose, everything is fine now.

FWIW, I really struggled with cleanup for a while today.  For each case in a unit test suite, we're starting an embedded server, then stopping it, then again on the next test.  Even after "fixing" errno 98 with my PR, I had many other issues, and fixing all of those also seems to fix this first problem.  I think at some point, I was doing something stupid that resulted in a leak of the listening socket from each instance.

Thanks for the feedback!

closed 2023-07-14 by joeyparrish