cesanta/mongoose #2302: Download speed drops by 0k/bps(mg_send)

Kraftww, 2023-07-14:
- My goal is: Improving the download speed of HTTP files.
- My actions were: Only MG_ MAX_ HTTP_ SEND_ IOBUF has been modified size, default : 4 * 1024 -> det 256 * 1024;
- My expectation was: Improving the download speed of HTTP files.
- The result I saw: The download speed of files has improved.
- My question is: However, there is a probability of a download speed of 0k/bps and reconnection cannot be restored. We are verifying based on WiFi 5G;

# Environment

- mongoose version: 6.0
- Compiler/IDE and SDK: aarch64-linux
- Target hardware/board: Embedded devices
- Connectivity chip/module: WiFi Module
- Target RTOS/OS (if applicable): Linux

scaprile, 2023-07-14:
Mongoose 6.0 has been deprecated eons ago, current version is 7.11
Mongoose 6 does not have a TCP/IP stack, ergo it is not responsible of how your TCP/IP stack or your network behaves. You don't provide your event handler code, we are not able to guess what you do and what you don't.
Please upgrade to a supported release.

closed 2023-07-14 by scaprile

Kraftww, 2023-07-23:
I'm sorry for not being able to explain the process clearly；Due to development reasons, we still hope to view issues based on version 6.0；So I really hope to receive your help；

Kraftww, 2023-07-23:
`#include "mongoose.h"
#ifdef NS_MODULE_LINES
#line 1 "src/internal.h"
/**/
#endif
/*
 * Copyright (c) 2014 Cesanta Software Limited
 * All rights reserved
 */

#ifndef MG_INTERNAL_HEADER_INCLUDED
#define MG_INTERNAL_HEADER_INCLUDED

#ifndef MG_MALLOC
#define MG_MALLOC malloc
#endif

#ifndef MG_CALLOC
#define MG_CALLOC calloc
#endif

#ifndef MG_REALLOC
#define MG_REALLOC realloc
#endif

#ifndef MG_FREE
#define MG_FREE free
#endif

#ifndef MBUF_REALLOC
#define MBUF_REALLOC MG_REALLOC
#endif

#ifndef MBUF_FREE
#define MBUF_FREE MG_FREE
#endif

#define MG_SET_PTRPTR(_ptr, _v) \
  do {                          \
    if (_ptr) *(_ptr) = _v;     \
  } while (0)

#ifndef MG_INTERNAL
#define MG_INTERNAL static
#endif

#if !defined(MG_MGR_EV_MGR)
/*
 * Switches between different methods of handling sockets. Supported values:
 * 0 - select()
 * 1 - epoll() (Linux only)
 */
#define MG_MGR_EV_MGR 0 /* select() */
#endif

#ifdef PICOTCP
#define NO_LIBC
#define MG_DISABLE_FILESYSTEM
#define MG_DISABLE_POPEN
#define MG_DISABLE_CGI
#define MG_DISABLE_DIRECTORY_LISTING
#define MG_DISABLE_SOCKETPAIR
#define MG_DISABLE_PFS
#endif

/* Amalgamated: #include "../mongoose.h" */

/* internals that need to be accessible in unit tests */
MG_INTERNAL struct mg_connection *mg_finish_connect(struct mg_connection *nc,
                                                    int proto,
[... 6182 lines cut ...]
#endif
/* Copyright (c) 2014 Cesanta Software Limited */
/* All rights reserved */

#ifndef MG_DISABLE_JSON_RPC

/* Amalgamated: #include "internal.h" */

int mg_rpc_create_reply(char *buf, int len, const struct mg_rpc_request *req,
                        const char *result_fmt, ...) {
  static const struct json_token null_tok = {"null", 4, 0, JSON_TYPE_NULL};
  const struct json_token *id = req->id == NULL ? &null_tok : req->id;
  va_list ap;
  int n = 0;

  n += json_emit(buf + n, len - n, "{s:s,s:", "jsonrpc", "2.0", "id");
  if (id->type == JSON_TYPE_STRING) {
    n += json_emit_quoted_str(buf + n, len - n, id->ptr, id->len);
  } else {
    n += json_emit_unquoted_str(buf + n, len - n, id->ptr, id->len);
  }
  n += json_emit(buf + n, len - n, ",s:", "result");

  va_start(ap, result_fmt);
  n

Kraftww, 2023-07-23:
`#ifdef __AVR__
#include "avrsupport.h"
#endif
/*
 * Copyright (c) 2004-2013 Sergey Lyubka
 * Copyright (c) 2013-2015 Cesanta Software Limited
 * All rights reserved
 *
 * This software is dual-licensed: you can redistribute it and/or modify
 * it under the terms of the GNU General Public License version 2 as
 * published by the Free Software Foundation. For the terms of this
 * license, see <http://www.gnu.org/licenses/>.
 *
 * You are free to use this software under the terms of the GNU General
 * Public License, but WITHOUT ANY WARRANTY; without even the implied
 * warranty of MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
 * See the GNU General Public License for more details.
 *
 * Alternatively, you can license this software under a commercial
 * license, as set out in <https://www.cesanta.com/license>.
 */

#define MG_VERSION "6.0"
/*
 * Copyright (c) 2015 Cesanta Software Limited
 * All rights reserved
 */

#ifndef OSDEP_HEADER_INCLUDED
#define OSDEP_HEADER_INCLUDED

#if !defined(MG_DISABLE_FILESYSTEM) && defined(AVR_NOFS)
#define MG_DISABLE_FILESYSTEM
#endif

#undef UNICODE                /* Use ANSI WinAPI functions */
#undef _UNICODE               /* Use multibyte encoding on Windows */
#define _MBCS                 /* Use multibyte encoding on Windows */
#define _INTEGRAL_MAX_BITS 64 /* Enable _stati64() on Windows */
#ifndef _CRT_SECURE_NO_WARNINGS
#define _CRT_SECURE_NO_WARNINGS /* Disable deprecation warning in VS2005+ */
#endif
#undef WIN32_LEAN_AND_MEAN /* Let windows.h always include winsock2.h */
#undef _XOPEN_SOURCE
#define _XOPEN_SOURCE 600    /* For flockfile() on Linux */
#define __STDC_FORMAT_MACROS /* <inttypes.h> wants this for C++ */
#define __STDC_LIMIT_MACROS  /* C++ wants that for INT64_MAX */
#ifndef _LARGEFILE_SOURCE
#define _LARGEFILE_SOURCE /* Enable fseeko() and ftello() functions */
#endif
#define _FILE_OFFSET_BITS 64 /* Enable 64-bit file offsets */

#if !(defined(AVR_LIBC) || defined(PICOTCP))
#include <sys/types.h>
#include <sys/stat.h>
#include <fcntl.h>
#include <time.h>
#include <signal.h>
#endif

#ifndef BYTE_ORDER
#define LITTLE_ENDIAN 0x41424344
#define BIG_ENDIAN 0x44434241
#define PDP_ENDIAN 0x42414443
/* TODO(lsm): fix for big-endian machines. 'ABCD' is not portable */
/*#define BYTE_ORDER 'ABCD'*/
#define BYTE_ORDER LITTLE_ENDIAN
#endif

/*
[... 2455 lines cut ...]
 * The caller have to check results and treat COAP_NOT_ENOUGH_DATA according to
 * underlying protocol:
 *
 * - in case of UDP COAP_NOT_ENOUGH_DATA means COAP_FORMAT_ERROR,
 * - in case of TCP client can try to receive more data
 *
 * Return value: see `mg_coap_send_message()`
 */
uint32_t mg_coap_parse(struct mbuf *io, struct mg_coap_message *cm);

/*
 * Composes CoAP message from mg_coap_message structure.
 * This is a helper function.
 * Return value: see `mg_coap_send_message()`
 */
uint32_t mg_coap_compose(struct mg_coap_message *cm, struct mbuf *io);

#ifdef __cplusplus
}
#endif /* __cplusplus */

#endif /* MG_ENABLE_COAP */

#endif /* MG_COAP_HEADER_INCLUDED */
`

Kraftww, 2023-07-23:
@scaprile I apologize again and hope to receive your help