cesanta/mongoose #2316: mg_tcpip_rx call mg_hexdump when log is disabled

lestofante, 2023-07-26:
- My goal is: 
disable output from the library
- My actions were: 
disabled logging with mg_log_set(MG_LL_NONE);  // Set log level
- My expectation was:
no output
- The result I saw:
A lot of "Unknown eth type"
- My question is: 
the line causing this are the following:
```
  } else {
    MG_DEBUG(("  Unknown eth type %x", mg_htons(pkt.eth->type)));
    mg_hexdump(buf, len >= 16 ? 16 : len);
  }
```
seems like an oversight the mg_hexdump() ignore the log level? especially here that is a DEBUG message.
I saw this happening i another couple place in the code, but is with MG_ERROR or disabled by #if 0.
I would expect disabling log, the library would never call mg_log_set_fn callback

# Environment

- mongoose version: 7.11
- Compiler/IDE and SDK: gcc version 13.1.0
- Target hardware/board: nucleo stm32f429zi

scaprile, 2023-07-26:
Define MG_ENABLE_LOG=0 to completely disable logging
https://github.com/cesanta/mongoose/blob/ed54b60ea35b564840ed4fd089bcbd44fcbc404a/mongoose.h#L920C2-L920C2
or Use https://mongoose.ws/documentation/#mg_log_set_fn as you wish

closed 2023-07-26 by scaprile

lestofante, 2023-07-26:
Side effect:
just found myself disabling the full output to avoid the messages from mg_hexdump(), and that caused me to miss some ERROR message:
the mg_hexdump() from those INFO section are a consequence from the mg_tcpip_init() failing to allocate the RX buffer, but the actual error was lost as the hexdump was flooding the console, or the logs where disabled.  
In this specific case, making the mg_hexdump() respect the same log level as the MG_DEBUG command it accompany, it would have been very helpful