cesanta/mongoose #2326: TCP Limits in 7.11

cliffowen, 2023-08-01:
Hey everyone,

I'm looking to use Mongoose in a small server application. I've used Mongoose in the past to do some REST API stuff and I thought I'd start here. The server is intended to create a TCP socket to communicate with clients with a reliable protocol, and then creates a back channel communication for UDP for non-reliable transmission.

To start, I wanted to simply create a basic networking layer to toss messages around between the client and the server, both TCP and UDP. It's all built together using localhost.

Once I got it started, I added a couple more test clients.. debugged, added a couple more. All good and fine until I got over 20 connections. At this point the TCP, not the UDP, connections started dropping.

With some investigation into the code, I see that select() is starting to return error sockets. I modified mg_iotest() a little bit so that  when it walked the connection list it will actually keep count of how many real connections there are before select() starts to fail. It's 64.

This is where I started Googling. There is a lot of information about 20 connections, 64 connections.. etc and suggestions on what to do about it. So far I've tried identifying why the individual sockets fail, tried bumping the FD_SETSIZE to 1000. Neither of these has worked.

Before I got too far along, I should say that this is a Windows 10 PRO machine. The more I read, the more it appeared to be a Windows limitation. As the final target is Linux, I decided to try a Mac because I have one right here that I do other dev work on and I have Linux at another desk. The Mac also hit 64 and then started dropping connections. I can find nothing about Mac having any limits.

I will try Linux but haven't yet. It's a little harder to move the code over there, but now that I have 2 different OS's showing the exact same behavior I cannot help feel this points at a library limitation. A limitation I cannot find anywhere.

My expectation is that I should be able to setup a testbed to prove and harden the core network functionality with a minimum of 1,000 simultaneous connections.

The question is: Is there a limit from the library? Can I change the limit? Is this documented? I can find no documentation and no sample that changes the limits.

Mongoose 7.11
Microsoft Windows 10 PRO with Visual Studio 2022
MacBookPro Ventura 13.1 M1 Max with Jetbrains CLion 2023.1.3

Linux is a little moot at the moment. Most of the dev environment will be Windows for the next few weeks.

Side question: In another, older, issue someone reported, it was stated that performance networking is not a priority for mongoose. I believe this was around Mongoose 5. We're now on Mongoose 7. I'm actually surprised how slow it is with 20 simultaneous connections. It wouldn't expect to see this kind of speed unless I had more than 200 simultaneous connections. Any thoughts on performance changes? I'm a little worried this might all be moot because unless I can get this moving substantially faster than it does, I might not be able to use it anyway.

scaprile, 2023-08-01:
https://github.com/cesanta/mongoose/issues/1962#issuecomment-1396732667

(perhaps you'll find the whole thread useful)

closed 2023-08-01 by scaprile

cliffowen, 2023-08-02:
Thank you for the reference. I have switched to polling and this does help. Unfortunately it's not reliable either. I've only now gotten to about 80 connections. Sometimes it's a fully clean run, sometimes it'll drop 20 sockets. Clearly there is something wrong with the sockets, and the polling code in mg_iotest() triggers here:

      if (fds[n].revents & POLLERR) {
        mg_error(c, "socket error");
      } else {

Same place triggers on a Mac. It's no more reliable than Windows.

I saw a reference to wireshark, and logging. I should say I had used the logging levels but they're not informative for this type of error. Best you see is "socket error" without an explanation as to why.

BUT, I will try out wireshark. I'm not a networking guy. If I was, I probably wouldn't be looking for a 3rd party solution. But I'll try to do a little more debugging and ask a more intelligent question.