cesanta/mongoose #2369: epoll_fd set CLOEXEC?

kylemilz, 2023-08-31:
Hi,

We are using Mongoose 7.11 on ARM using GCC, and epoll.

We have an application that does many `exec(3)`. I noticed that after a while, the program runs out of file descriptors. I have tracked the descriptor leak down to the mongoose `mgr->epoll_fd`. It looks like it's being created without the `EPOLL_CLOEXEC` flag.

I don't think this is desired behavior, and indeed some of the other file descriptors created by mongoose are setting CLOEXEC, notably in `mg_set_non_blocking_mode()`. What do you think of this diff?

```
diff --git a/src/net.c b/src/net.c
index 45b60904..879f22ac 100644
--- a/src/net.c
+++ b/src/net.c
@@ -238,7 +238,7 @@ void mg_mgr_free(struct mg_mgr *mgr) {
 void mg_mgr_init(struct mg_mgr *mgr) {
   memset(mgr, 0, sizeof(*mgr));
 #if MG_ENABLE_EPOLL
-  if ((mgr->epoll_fd = epoll_create1(0)) < 0) MG_ERROR(("epoll: %d", errno));
+  if ((mgr->epoll_fd = epoll_create1(EPOLL_CLOEXEC)) < 0) MG_ERROR(("epoll: %d", errno));
 #else
   mgr->epoll_fd = -1;
 #endif
```

kylemilz, 2023-08-31:
oops, yeah this is on Linux 5.15.

scaprile, 2023-08-31:
We don't set `O_CLOEXEC` when we `open()`, we set `FD_CLOEXEC` here:
https://github.com/cesanta/mongoose/blob/d5b5cec7970473d0526262ac42c93ea8a98f0699/mongoose.c#L5742-L5771

What you post is executed only once when calling `mg_mgr_init()`:
https://github.com/cesanta/mongoose/blob/d5b5cec7970473d0526262ac42c93ea8a98f0699/mongoose.c#L3806-L3812

and then for EPOLL, we do this:
https://github.com/cesanta/mongoose/blob/d5b5cec7970473d0526262ac42c93ea8a98f0699/mongoose.c#L5880-L5891

You know, the issue template exists for a reason, to save yours and our time, so please tell us _exactly_ what are you doing.

WRT
> an application that does many `exec(3)`

my best guess would be:
From [the docs](https://mongoose.ws/documentation/#2-minute-integration-guide)
> NOTE: Since Mongoose's core is not protected against concurrent accesses, make sure that all mg_* API functions are called from the same thread or RTOS task.

If you absolutely need multi-threading, take the time to go through the user guide (linked above) and [tutorials](https://mongoose.ws/tutorials/), there is one that shows you how to do [multi-threading](https://mongoose.ws/tutorials/multi-threaded/).

scaprile, 2023-09-02:
I fail to grasp why you might want to do something like that. I don't see a usage model like that in any of our examples nor the user guide, I won't call yours "a simple modification".
In my limited view, that of a microcontroller-oriented old dog, I can only see that every time you replace yourself by a clone of yourself, you run the initialization sequence again without freeing any of the allocated resources. You will not only run out of file descriptors, but you will also run out of memory.
I've assigned this issue to our master mind, let's see what he thinks.

cpq, 2023-09-04:
Thanks @scaprile 

@kylemilz , the change looks reasonable. We missed the fact that the epoll descriptors are also inherited.

By the way, we also don't set CLOSEXEC on the static UI files opened by the HTTP server - a standard `fopen` without `e`
 flag is used. So I guess you can see static UI files could be inherited by a child process.

Quick question. Are your child processes not getting closed?
My understanding is that if child exits, all descriptors are released by the OS. Thus if mongoose (the parent) closes epoll fd in the normal cause of action, and if children get exited, then no resource exhaustion should be observed. Basically, I don't anticipate a parent (mongoose) leaking descriptors, even in the current scenario. 
What is you observation?

cpq, 2023-09-05:
@kylemilz appreciate your BSD style in the test program, btw.

#2371 should fix the issue

closed 2023-09-05 by scaprile

gvanem, 2023-09-09:
This has broken the Windows version. Dr. Watson kicks in. I fixed it by:
```diff
--- a/src/fs_posix.c 2023-09-08 07:15:48
+++ b/src/fs_posix.c 2023-09-09 09:06:12
@@ -166,13 +166,14 @@
 }

 static void *p_open(const char *path, int flags) {
-  const char *mode = flags == MG_FS_READ ? "rbe" : "a+be";  // e for CLOEXEC
 #if MG_ARCH == MG_ARCH_WIN32
+  const char *mode = flags == MG_FS_READ ? "rb" : "a+b";
   wchar_t b1[MG_PATH_MAX], b2[10];
   MultiByteToWideChar(CP_UTF8, 0, path, -1, b1, sizeof(b1) / sizeof(b1[0]));
   MultiByteToWideChar(CP_UTF8, 0, mode, -1, b2, sizeof(b2) / sizeof(b2[0]));
   return (void *) _wfopen(b1, b2);
 #else
+  const char *mode = flags == MG_FS_READ ? "rbe" : "a+be";  // e for CLOEXEC
   return (void *) fopen(path, mode);
 #endif
 }
```

cpq, 2023-09-09:
Oops. Our CI did not complain about windows tests.
I was under impression that the C lib `fopen` should ignore unknown flags.

@gvanem could you elaborate a bit more what exactly do you observe? An antivirus pops up?

scaprile, 2023-09-09:
We only test examples for build, and that one is not in the main list so I'm not sure we actually test it. Plus, we are not building with every combination, we either build with POLL or EPOLL. However, we should catch that in the unit tests. I'll check on Monday and update.

kylemilz, 2023-09-11:
> I fail to grasp why you might want to do something like that. I don't see a usage model like that in any of our examples nor the user guide, I wouldn't call yours "a simple modification".

Our use case is quite valid. What we have is a configuration update coming from http, ie we have a user dashboard where they can set some values, then these are sent to mongoose. Upon validation of the configuration, it gets written to a database and then we use `exec()` to reinitialize the whole process, which reads the new configuration from the database. This is a very simple way to handle configuration changes, and makes sure you are always using the latest configuration. It's a common pattern in Unix.

What happened is we have tests that write many permutations of configuration, and we noticed after a while the tests would fail. The tests failed because the configuration database could not be opened, because the process had too many open file descriptors. That lead us to discover this leak.

> In my limited view, that of a microcontroller-oriented old dog, I can only see that every time you replace yourself by a clone of yourself, you run the initialization sequence again without freeing [...]

Unix process management is kind of weird. IIRC all other resources except file descriptors get free'd across an `exec()`. I don't really know why as there's no way to retain the file descriptor number.

kylemilz, 2023-09-11:
> https://github.com/cesanta/mongoose/pull/2371 should fix the issue

Thanks, presuming this fix is in 7.12 i'll give it a try when it's released.

scaprile, 2023-09-11:
Yes, it will be included in 7.12

commit 8140ea16 (2023-09-05): Merge pull request #2371 from cesanta/cloexec

Fix #2369 - set CLOEXEC on epoll and regular file fds