cesanta/mongoose #2385: How to correctly close client connection?

HiperDoo, 2023-09-17:
Hello, I have recently discovered this library and I loved it, while I was doing TCP communication tests, I came across a strange behavior.

I use `netcat` to do simple tests, simulating clients that can connect and send text messages.

When the server closes the connection with the user `mg_close_conn(conn)` (in this example, the client sends the text `"exit"`, the server closes its connection); everything is going well until here. The client doesn't realize that (I think) and if I try to send any other message or close the client process (Ctrl+C), the server seems to ignore the timeout assigned within `mg_mgr_poll(&mgr, 1000)`, so such a command It runs as fast as possible, making the process reach 100% CPU consumption (in a clear thread).

Curiously, even if this error occurs, the server continues to function, allowing the entry and messaging of new users.

I am somewhat new to this world of networking, and I know that doing this type of testing is not the best thing in the world, and although I don't know exactly what 'netcat' actually does, it is an easily replicable bug that could make my server suffer.

I am sure that the error is mine and that I am not handling correctly the way to close a connection; I would be grateful if you could guide me on how to handle this kind of scenario, what is the correct way to close a connection (or why `mg_mgr_poll(&mgr, 1000)` seems to ignore the timeout in this scenario).

# Code
```cpp
// I simplified my program to only show the scenario to replicate.

// main.cpp
#include "mongoose.h"

void func(mg_connection* conn, int event, void*, void*) {
    if (event == MG_EV_READ) {
        if (!strncmp((const char*)conn->recv.buf, "exit", 4))
            mg_close_conn(conn);

        static const char* header = "[SERVER]: ";
        mg_send(conn, header, 10);
        mg_send(conn, conn->recv.buf, conn->recv.len);

        mg_iobuf_del(&conn->recv, 0, conn->recv.len);
    } else if (event == MG_EV_ACCEPT) {
        printf("NEW CONNECTION\n");
    } else if (event == MG_EV_CLOSE) {
        printf("CLOSE CONNECTION\n");
    }
}

int main(void) {
    mg_log_set(MG_LL_INFO);

    mg_mgr mgr;
    mg_mgr_init(&mgr);

    mg_listen(&mgr, "tcp://localhost:1234", func, nullptr);

    while (true) {
        mg_mgr_poll(&mgr, 1000);

        static uint32_t counter{0};
        printf("%u\n", ++counter);
    }

    mg_mgr_free(&mgr);

    return EXIT_SUCCESS;
}
```

# Details
- Command to simulate client: `nc localhost 1234` *install -> sudo pacman -S netcat*
- When I close the client (Ctrl+C), I get an error/warning message (does not stop the program from running): `8ea7cd 1 mongoose.cpp:403:mg_error 2 0x5 socket error`

Also note that every time an event is detected (be it a `MG_EV_READ`, `MG_EV_ACCEPT` or `MG_EV_CLOSE`), the timeout cycle is immediately terminated. So if 3 events are executed in one second, `mg_mgr_poll(&mgr, 1000)` will have been executed 3 times.

With this in mind, my guess is that perhaps the server is trying to access a client resource (which was already freed with `mg_close_conn(conn)`), which is why these events are repeatedly called so close together.

# Environment

- Mongoose Version: `7.11`
- Compiler: `gcc (GCC) 13.2.1 20230801`
- Operative System: `Arch Linux x86_64` `6.5.3-arch1-1`
- Hardware: `Intel i5-10400F (12) @ 4.300GHz` `RAM 15876MiB`

cpq, 2023-09-17:
@HiperDoo user must never call `mg_close_conn()`, that is a private function, not documented.

The right way is to either
- `c->is_draining = 1;` - tell mongoose to send the remaining data to the client, and close when everything is sent
- `c->is_closing = 1;` - tell mongoose to close straight away, on the next poll cycle.

cpq, 2023-09-17:
Added appropriate section in the user guide
https://mongoose.ws/documentation/#opening-and-closing-connections

closed 2023-09-17 by cpq

scaprile, 2023-09-17:
We already had something in the docs: https://mongoose.ws/documentation/#best-practices , updated to latest built-in stack compatible behavior. Quote:

- If you need to close the connection, set `c->is_draining = 1;` in your event handler function. That tells the event manager to send all remaining data in its send buffer ("drain" the connection), then properly close the connection.
  - If you need to remove the connection immediately, regardless of any buffered data or a clean TCP closure, then use c->is_closing = 1;`.

There happens to be an example that shows how to do a generic server and client, and how to close connections: 
https://github.com/cesanta/mongoose/blob/a04c94444e6e3e803c6f0b539aad5872f7a8dd19/examples/tcp/main.c#L80
and a tutorial: https://mongoose.ws/documentation/tutorials/tcp/

HiperDoo, 2023-09-17:
Thank you very much for all the advice and help, this was easier than I thought.

I had seen this about `is_draning`, I thought it was an informational flag and not another means to perform actions (sorry for not reading more carefully). It also seemed strange to me that the `mg_close_conn()` function was not documented, I see why.

I also appreciate you showing me these documentations of examples of good practices, this will be very helpful.