cesanta/mongoose #2396: UDP Broadcast with RP2040 and W5500

rickbassham, 2023-09-29:
- My goal is: to send a UDP broadcast message
- My actions were: See gist [here](https://gist.github.com/rickbassham/fba0763f00a14e464d474486964875de)
- My expectation was: To see the UDP packet in Wireshark or with netcat
- The result I saw: 

```
fa2    2 main.cpp:74:setup              MAC: 02:f7:0b:83:90:36 Waiting for IP...
1772   1 mongoose.c:9378:onstatechange  Link up
1793   2 mongoose.c:9619:rx_dhcp_client Lease: 43200 sec (43206)
1795   2 mongoose.c:9371:onstatechange  READY, IP: 192.168.10.167
1797   2 mongoose.c:9372:onstatechange         GW: 192.168.10.1
1798   2 mongoose.c:9373:onstatechange        MAC: 02:f7:0b:83:90:36
179b   2 main.cpp:82:setup              common_setup complete
2b24   2 main.cpp:21:sfn                sfn 1
2b25   2 main.cpp:21:sfn                sfn 3
2b29   2 main.cpp:21:sfn                sfn 8
2b2a   2 main.cpp:21:sfn                sfn 9
2b2b   2 main.cpp:36:sfn                Connection closed
3eac   2 main.cpp:21:sfn                sfn 1
3ead   2 main.cpp:21:sfn                sfn 3
3eb1   2 main.cpp:21:sfn                sfn 8
3eb2   2 main.cpp:21:sfn                sfn 9
```

- My question is: Am I configuring this incorrectly? I don't see any packets in wireshark or with netcat.

# Environment

- mongoose version: 7.11
- Compiler/IDE and SDK: VSCode
- Target hardware/board: RP2040
- Connectivity chip/module: W5500
- Target RTOS/OS (if applicable): Arduino

rickbassham, 2023-09-29:
I tried using both `mg_send` and `mg_printf` to send the packet. When using `mg_send`, I only get the events `MG_EV_OPEN` and `MG_EV_RESOLVE`.

scaprile, 2023-09-29:
There are UDP examples sending multicast.
If that doesn't work for you, please post your minimum smallest code here showing what does not work and why
You should have already seen at least one broadcast packet: DHCP request.
Don't "try", r
Please follow the guidelines in the [documentation](https://mongoose.ws/documentation)

If there is something that you can't do, please describe it with the smallest piece of code possible.

If you think an example can be useful, please describe what you want to achieve that is not covered in any of the [examples](https://github.com/cesanta/mongoose/tree/master/examples) and [tutorials](https://mongoose.ws/tutorials/) available.

rickbassham, 2023-09-29:
I have included a gist of the smallest piece of code to reproduce the problem. I have read the documentation, but still having issues, which is why I reached out for help.

scaprile, 2023-09-29:
Unfortunately I'm a bit busy to see such a long gist. Please provide the minimum smallest piece of code

rickbassham, 2023-09-29:
Here is the basics. I start a timer to run every 5 seconds.

```cpp
    mg_timer_add(&mgr, 5000, MG_TIMER_REPEAT, &timer_fn, &mgr);
```

When the timer fires, connect to the broadcast address

```cpp
static void timer_fn(void *arg)
{
    struct mg_mgr *mgr = (struct mg_mgr *) arg;
    mg_connect(mgr, "udp://255.255.255.255:23433", sfn, NULL);
}
```

The callback for the connection:

```cpp
static void sfn(struct mg_connection *c, int ev, void *ev_data, void *fn_data)
{
    if (ev != MG_EV_POLL)
    {
        MG_INFO(("sfn %d", ev));
    }

    if (ev == MG_EV_OPEN)
    {
        mg_send(c, "test", 4);
    }

    if (ev == MG_EV_RESOLVE)
    {
        mg_send(c, "test", 4);
    }

    if (ev == MG_EV_CONNECT)
    {
        mg_send(c, "test", 4);
    }

    if (ev == MG_EV_WRITE)
    {
        mg_close_conn(c);
    }

    if (ev == MG_EV_CLOSE)
    {
        MG_INFO(("Connection closed"));
    }
}
```

scaprile, 2023-09-29:
> ```c++
> mg_connect(mgr, "udp://255.255.255.255:23433", sfn, NULL);
> ```

That should work
Do you see the DHCP request broadcast ?

https://github.com/cesanta/mongoose/blob/5ebede509699fbe2fb8168fe5323e5fb5447c167/examples/udp-ssdp-search/main.c#L4-L64

Can you test if the above multicast code works for you ? (The udp-ssdp-search example)

rickbassham, 2023-09-29:
Here is the output from that: 

```
fa2    2 main.cpp:93:setup              MAC: 02:f7:0b:83:90:36 Waiting for IP...
1772   1 mongoose.c:9378:onstatechange  Link up
1794   2 mongoose.c:9619:rx_dhcp_client Lease: 43200 sec (43206)
1795   2 mongoose.c:9371:onstatechange  READY, IP: 192.168.10.167
1797   2 mongoose.c:9372:onstatechange         GW: 192.168.10.1
1798   2 mongoose.c:9373:onstatechange        MAC: 02:f7:0b:83:90:36
179b   2 main.cpp:101:setup             common_setup complete
2b24   2 main.cpp:56:tfn                Sending M-SEARCH
2b25   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b26   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b27   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b28   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b2a   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b2b   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b2c   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b2d   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b2e   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b2f   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b30   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b32   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b33   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b34   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b35   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b36   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b37   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b38   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b3a   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b3b   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b3c   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b3d   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b3e   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b3f   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b41   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b42   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b43   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b44   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b45   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b47   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b48   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b49   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b4a   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b4b   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b4d   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b4e   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b4f   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b50   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b51   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b52   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b54   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b55   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b56   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b57   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b58   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b59   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b5a   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b5c   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b5d   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b5e   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b5f   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b60   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b61   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b63   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b64   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b65   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b66   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b67   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2b68   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
[... 50 lines cut ...]
2ba3   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2ba4   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2ba5   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2ba7   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2ba8   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2ba9   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2baa   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2bab   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2bac   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2bad   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2baf   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2bb0   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2bb1   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2bb2   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2bb3   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2bb4   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2bb5   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2bb7   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2bb8   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
2bb9   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
3eac   2 main.cpp:56:tfn                Sending M-SEARCH
3ead   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
3eae   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
3eaf   1 mongoose.c:2953:mg_iobuf_resiz 0->536875032
```

rickbassham, 2023-09-29:
Here is the output when running that: 

```
fa2    2 main.cpp:93:setup              MAC: 02:f7:0b:83:90:36 Waiting for IP...
1772   1 mongoose.c:9378:onstatechange  Link up
1793   2 mongoose.c:9619:rx_dhcp_client Lease: 43200 sec (43206)
1795   2 mongoose.c:9371:onstatechange  READY, IP: 192.168.10.167
1796   2 mongoose.c:9372:onstatechange         GW: 192.168.10.1
1798   2 mongoose.c:9373:onstatechange        MAC: 02:f7:0b:83:90:36
179b   2 main.cpp:105:setup             common_setup complete
179e   2 main.cpp:56:tfn                Sending M-SEARCH
1f6c   2 main.cpp:56:tfn                Sending M-SEARCH
273c   2 main.cpp:56:tfn                Sending M-SEARCH
```

rickbassham, 2023-09-29:
I do not see the packets in wireshark

scaprile, 2023-09-29:
Then you do have another problem... that is why I asked twice if you see the DHCP broadcast packet

rickbassham, 2023-09-29:
I do see the DHCP packet in wireshark

rickbassham, 2023-09-29:
But not the SSDP or the manual UDP broadcast I was doing.

scaprile, 2023-09-29:
That is quite weird.
I'll see if I can prepare a small example.

scaprile, 2023-09-30:
> ```c++
> mg_connect(mgr, "udp://255.255.255.255:23433", sfn, NULL);
> ```

That won't work. I corrected the former response, I overlooked that before. You can't send a broadcast to everybody in the world. Use a local broadcast, send to "all hosts in your network". E,g.: 192.168.10.255
It should send a broadcast frame (broadcast destination MAC) with a local broadcast destination IP.

PS: Multicast should work, maybe you are filtering to see a broadcast frame, filter by IP, multicast frames use a special multicast MAC. We've tested for Alpaca.

scaprile, 2023-09-30:
I happen to have written that part of our code, that is why I tell you, our stack does not allow the user to send full IP broadcasts, send local broadcasts instead. We haven't had the need to add full broadcasts and those are not recommended.
https://github.com/cesanta/mongoose/blob/5ebede509699fbe2fb8168fe5323e5fb5447c167/src/net_builtin.c#L1008-L1024
Now that I see the code again, since we added ARP lookup that needs to be polished a bit.
Multicast was working fine last time I checked, I'll test it again. We don't test in your hardware, though, the W5500 could be filtering something. It shouldn't but it is a possibility.

scaprile, 2023-09-30:
Fixed and allowed global broadcasts:
Change lines 1008-1018 to
```
  if (c->is_udp && (rem_ip == 0xffffffff || rem_ip == (ifp->ip | ~ifp->mask))) {
    struct connstate *s = (struct connstate *) (c + 1);
    memset(s->mac, 0xFF, sizeof(s->mac));  // global or local broadcast
  } else if (((rem_ip & ifp->mask) == (ifp->ip & ifp->mask))) {
    // If we're in the same LAN, fire an ARP lookup.
    MG_DEBUG(("%lu ARP lookup...", c->id));
    arp_ask(ifp, rem_ip);
    settmout(c, MIP_TTYPE_ARP);
    c->is_arplooking = 1;
    c->is_connecting = 1;
  } else if ((*((uint8_t *) &rem_ip) & 0xE0) == 0xE0) {
```
We'll push a fix in the following weeks.

rickbassham, 2023-10-01:
Which file is this in?

rickbassham, 2023-10-01:
This still isn't working for me. This may help: I added a log statement to `tx_udp` and the `mac_dst` is `00:00:00:00:00:00`. I would expect a broadcast mac of `ff:ff:ff:ff:ff:ff`, which I do see when a DHCP request is sent.

scaprile, 2023-10-01:
Well, it does work for me, and I'm able to execute the code line by line and explain why it works. Maybe you are not patching those lines correctly.

rickbassham, 2023-10-01:
Here is my patched function:
```cpp
void mg_connect_resolved(struct mg_connection *c) {
  struct mg_tcpip_if *ifp = (struct mg_tcpip_if *) c->mgr->priv;
  uint32_t rem_ip;
  memcpy(&rem_ip, c->rem.ip, sizeof(uint32_t));
  c->is_resolving = 0;
  if (ifp->eport < MG_EPHEMERAL_PORT_BASE) ifp->eport = MG_EPHEMERAL_PORT_BASE;
  memcpy(c->loc.ip, &ifp->ip, sizeof(uint32_t));
  c->loc.port = mg_htons(ifp->eport++);
  MG_DEBUG(("%lu %M -> %M", c->id, mg_print_ip_port, &c->loc, mg_print_ip_port,
            &c->rem));
  mg_call(c, MG_EV_RESOLVE, NULL);
  if (c->is_udp && (rem_ip == 0xffffffff || rem_ip == (ifp->ip | ~ifp->mask))) {
    struct connstate *s = (struct connstate *) (c + 1);
    memset(s->mac, 0xFF, sizeof(s->mac));  // global or local broadcast
  } else if (((rem_ip & ifp->mask) == (ifp->ip & ifp->mask))) {
    // If we're in the same LAN, fire an ARP lookup.
    MG_DEBUG(("%lu ARP lookup...", c->id));
    arp_ask(ifp, rem_ip);
    settmout(c, MIP_TTYPE_ARP);
    c->is_arplooking = 1;
    c->is_connecting = 1;
  } else if ((*((uint8_t *) &rem_ip) & 0xE0) == 0xE0) {
    struct connstate *s = (struct connstate *) (c + 1);  // 224 to 239, E0 to EF
    uint8_t mcastp[3] = {0x01, 0x00, 0x5E};              // multicast group
    memcpy(s->mac, mcastp, 3);
    memcpy(s->mac + 3, ((uint8_t *) &rem_ip) + 1, 3);  // 23 LSb
    s->mac[3] &= 0x7F;
  } else {
    struct connstate *s = (struct connstate *) (c + 1);
    memcpy(s->mac, ifp->gwmac, sizeof(ifp->gwmac));
    if (c->is_udp) {
      mg_call(c, MG_EV_CONNECT, NULL);
    } else {
      send_syn(c);
      settmout(c, MIP_TTYPE_SYN);
      c->is_connecting = 1;
    }
  }
}
```

rickbassham, 2023-10-01:
Ok, if I comment out some code in `mg_send` I can get it to work with your patch above, which means it's not a hardware issue, there is some logic that is not happening correctly in `mg_send` when sending UDP. Likely because it is bypassing the io buf altogether. With this modified `mg_send`, it works for both local broadcast and global broadcast.

```cpp
bool mg_send(struct mg_connection *c, const void *buf, size_t len) {
  struct mg_tcpip_if *ifp = (struct mg_tcpip_if *) c->mgr->priv;
  bool res = false;
  uint32_t rem_ip;
  memcpy(&rem_ip, c->rem.ip, sizeof(uint32_t));
  if (ifp->ip == 0 || ifp->state != MG_TCPIP_STATE_READY) {
    mg_error(c, "net down");
  // } else if (c->is_udp) {
  //   struct connstate *s = (struct connstate *) (c + 1);
  //   len = trim_len(c, len);   // Trimming length if necessary
  //   tx_udp(ifp, s->mac, ifp->ip, c->loc.port, rem_ip, c->rem.port, buf, len);
  //   res = true;
  } else {
    res = mg_iobuf_add(&c->send, c->send.len, buf, len);
  }
  return res;
}
```

scaprile, 2023-10-02:
How come it works for me and I don't need to comment out anything ?
Maybe you are not using Mongoose properly ? Did you see that all embedded examples loop waiting for the interface to be ready before trying to do anything else ? Please see the examples

rickbassham, 2023-10-02:
Yes, I've been using mongoose for a while for many other things, and feel I'm pretty well versed in it. As you can see in the gist I posted in my first comment, I am waiting until the mgr is ready before starting the timer: 

```cpp
    mg_tcpip_init(&mgr, &mif);

    MG_INFO(("MAC: %M Waiting for IP...", mg_print_mac, mif.mac));
    while (mif.state != MG_TCPIP_STATE_READY)
    {
        mg_mgr_poll(&mgr, 1);
    }

    mg_timer_add(&mgr, 5000, MG_TIMER_REPEAT, &timer_fn, &mgr);
```

rickbassham, 2023-10-02:
And of course, after doing a full clean build it is working with just your patch. It is working with just your patch.

PR #2402 (2023-10-03): send broadcasts in MIP
Adding ARP lookup exposed shadowed local broadcast handling (broadcast address was taken as local address and fired an ARP lookup). So, check is done now _before_ checking for local addresses.
Added also global broadcast MAC resolution.
Should we disallow users from sending global broadcasts ?

Fixes #2396
closed 2023-10-02 by scaprile