cesanta/mongoose #2398: TLS / Mismatch Doc - Example

dev31337, 2023-10-01:
Hi,

There is an inconsistency between the documentation and the source code for the TLS.

The documentation [here](https://mongoose.ws/documentation/#mg_tls_ctx_init) says that we should use mg_tls_ctx_init() to set up HTTPS (with certificates in options) and not mg_tls_init().

The example [here](https://mongoose.ws/documentation/tutorials/tls/) use mg_tls_ctx_init().

But the mg_tls_ctx_init() does not have an mg_tls_opts second parameter anymore.

Are we supposed to use mg_tls_init() on MG_EV_ACCEPT instead as shown in the restful server example source code [here](https://github.com/cesanta/mongoose/blob/master/examples/http-restful-server/main.c) ?

Regards.

cpq, 2023-10-01:
@dev31337 example source code is correct, we did not fix our docs yet.

dev31337, 2023-10-01:
Thanks for the reply.

When I try to do it (using mg_tls_init() on MG_EV_ACCEPT), using master, I get the following error: "1b508a464 1 mongoose.c:905:mg_error     2 5 HTTP parse".

Here is a small event trace:
MG_EV_OPEN
MG_EV_ACCEPT
MG_EV_POLL
MG_EV_READ
1b508a464 1 mongoose.c:905:mg_error     2 5 HTTP parse
MG_EV_ERROR
MG_EV_CLOSE

I double checked everything twice, I tried 7.11 (where mg_tls_init takes certificates by files instead of mg_str) and I still get the same error.

Do you have any idea ?

scaprile, 2023-10-01:
Our examples do work, so my best bet is that the problem is in your code. Did you try the example as is ?
What are you doing, what are you trying to do, how do you build, what is your environment ? Please fill the issue template

We are changing our TLS API, docs could not keep up with the changes so far. HOWEVER, the instructions to build are correct, so follow the build instructions, test with an example, and then follow that example as a guide for your code.
If you want to use 7.11 release, see the examples there. If you will be using git HEAD, see the examples there. Unfortunately the details in the docs right now are not correct for any of them, there have been very recent changes in our API.

dev31337, 2023-10-01:
Let's close this issue and make a new one with my example code.

closed 2023-10-01 by dev31337