cesanta/mongoose #2399: HTTPS Issue

dev31337, 2023-10-01:
- My goal is: simply to launch a HTTPS server and server a text file
- My actions were: build my own code
- My expectation was: that the request would go through
- The result I saw: I get the error "mongoose.c:905:mg_error     2 5 HTTP parse"
- My question is: Why do I get this error ?

# Environment

- mongoose version: Master
- Compiler/IDE and SDK: Linux Ubuntu 20.04 LTS (and also 22.04 LTS)
- Target hardware/board: PC
- Connectivity chip/module: None
- Target RTOS/OS (if applicable): Linux Ubuntu 20.04 LTS (and also 22.04 LTS)

Hi,

When I compile the following code using the master version of mongoose and I make a HTTPS request to port 8443 then mongoose display the error "mongoose.c:905:mg_error     2 5 HTTP parse".

I double checked that the cert and key are valid and that they are successfully loaded by mg_file_read().

I tested on Ubuntu 20.04 LTS (OpenSSL v1) and Ubuntu 22.04 LTS (OpenSSL v2).

I compile with "-DMG_TLS=MG_TLS_OPENSSL" and link with "-lssl -lcrypto".

Here is the full code:

#include "mongoose.h"

char *cert = NULL, *key = NULL;
size_t csize = 0, ckey = 0; 

static void fn(struct mg_connection *c, int ev, void *ev_data, void *fn_data)
{
    if (ev == MG_EV_ACCEPT && fn_data != NULL)
    {
        struct mg_tls_opts opts;
        opts.cert = mg_str(cert);
        opts.key = mg_str(key);
        mg_tls_init(c, &opts);
    }
     if (ev == MG_EV_HTTP_MSG) 
    {
        struct mg_http_message *hm = (struct mg_http_message *) ev_data;
        struct mg_http_serve_opts opts = {
            .mime_types = "txt=text/plain"
            };
            mg_http_serve_file(c, hm, "test.txt", &opts);
    }
}

int main(void)
{
    cert = mg_file_read(&mg_fs_posix, "ssl.pem", &csize);
    key = mg_file_read(&mg_fs_posix, "ssl.key", &ckey);
    struct mg_mgr mgr;
    mg_mgr_init(&mgr);
    mg_http_listen(&mgr, "https://0.0.0.0:8443", fn, NULL); 
    while(1) mg_mgr_poll(&mgr, 1000);
    mg_mgr_free(&mgr);
    }

scaprile, 2023-10-01:
Are you able to run the example as is ?
(Follows from #2398)

dev31337, 2023-10-01:
Found the issue by reducing the example to the bare minimum, closing the ticket.

closed 2023-10-01 by dev31337