cesanta/mongoose #2413: websocket + ssl

MarlikAlmighty, 2023-10-06:
Hi, thanks for Mongoose.

My question is, is there a code sample somewhere that creates a websocket server with SSL?

I started learning C a little, and I still can’t read the source code. I think you should update the documentation https://mongoose.ws/documentation/ I liked it, but it’s not relevant everywhere. Thanks again.

scaprile, 2023-10-06:
This is not a Mongoose issue
I beg to differ on your impressions on the documentation and its relevancy. Thank you anyway.
As you probably know, SSL is TLS and WebSocket upgrades an HTTPS connection, so you already have what you need where all the examples are.
https://github.com/cesanta/mongoose/tree/master/examples/http-restful-server
https://github.com/cesanta/mongoose/tree/master/examples/websocket-server

closed 2023-10-06 by scaprile

MarlikAlmighty, 2023-10-06:
Thank you, this is a quality product.

MarlikAlmighty, 2023-10-10:
I have a problem, the upgrade to websocket doesn’t work for me, could you please kick me in the right direction, otherwise I’ve already broken my whole brain. Here's the code:

```
#include "mongoose.h"

// static const char *s_http_addr = "http://0.0.0.0:8000";   // http
static const char *s_https_addr = "https://0.0.0.0:8443"; // https
static const char *s_root_dir = ".";

const char *cert =
    "-----BEGIN CERTIFICATE-----\n"
    "MIICUDCCAfegAwIBAgIUZN7fqIgS2IOs6gXA6TS/5ItXbOAwCgYIKoZIzj0EAwIw\n"
    "fjELMAkGA1UEBhMCQVUxDTALBgNVBAgMBFNPTUUxDTALBgNVBAcMBFNJVFkxDDAK\n"
    "BgNVBAoMA0xURDENMAsGA1UECwwEVU5JVDESMBAGA1UEAwwJbG9jYWxob3N0MSAw\n"
    "HgYJKoZIhvcNAQkBFhFtYWlsQG1vbmdvb3NlLmNvbTAeFw0yMzEwMDkxNjI4Mzha\n"
    "Fw0yMzExMDgxNjI4MzhaMH4xCzAJBgNVBAYTAkFVMQ0wCwYDVQQIDARTT01FMQ0w\n"
    "CwYDVQQHDARTSVRZMQwwCgYDVQQKDANMVEQxDTALBgNVBAsMBFVOSVQxEjAQBgNV\n"
    "BAMMCWxvY2FsaG9zdDEgMB4GCSqGSIb3DQEJARYRbWFpbEBtb25nb29zZS5jb20w\n"
    "WTATBgcqhkjOPQIBBggqhkjOPQMBBwNCAAQiesU2udwWi8+diO0yAabXUtDbTQ+I\n"
    "U81xA/rAgz7ODmkHmHNT8Tseqaq3jvMH2tQV7TRTM8Rv6puZxs7efZuco1MwUTAd\n"
    "BgNVHQ4EFgQUdDb6dmJMNhm7cfspr3OKQsLN5fEwHwYDVR0jBBgwFoAUdDb6dmJM\n"
    "Nhm7cfspr3OKQsLN5fEwDwYDVR0TAQH/BAUwAwEB/zAKBggqhkjOPQQDAgNHADBE\n"
    "AiARQ3rHoDPHf41WW35gik6F7T0s/Xk2+ryPuy9bt/7IYgIgI4m8xc7eYY26HZwN\n"
    "FtQHXyeY65zA8YLeyLF+rCqA/Y8=\n"
    "-----END CERTIFICATE-----\n";

const char *key =
    "-----BEGIN EC PRIVATE KEY-----\n"
    "MHcCAQEEIOddcyY5cq0beuS3R0PxOweubkClrXRCG07RE972epMvoAoGCCqGSM49\n"
    "AwEHoUQDQgAEInrFNrncFovPnYjtMgGm11LQ200PiFPNcQP6wIM+zg5pB5hzU/E7\n"
    "Hqmqt47zB9rUFe00UzPEb+qbmcbO3n2bnA==\n"
    "-----END EC PRIVATE KEY-----\n";

// We use the same event handler function for HTTP and HTTPS connections
// fn_data is NULL for plain HTTP, and non-NULL for HTTPS
static void fn(struct mg_connection *c, int ev, void *ev_data, void *fn_data)
{

  if (ev == MG_EV_ACCEPT && fn_data != NULL)
  {
    struct mg_tls_opts opts =
        {
            .cert = mg_str(cert),
            .key = mg_str(key)};
    mg_tls_init(c, &opts);
  }

  if (ev == MG_EV_HTTP_MSG)
  {
    struct mg_http_message *hm = (struct mg_http_message *)ev_data;
    if (mg_http_match_uri(hm, "/websocket"))
    {
      mg_ws_upgrade(c, hm, NULL);
    }
    else
    {
      struct mg_http_serve_opts opts = {.root_dir = s_root_dir};
      mg_http_serve_dir(c, ev_data, &opts);
    }
  }

  if (ev == MG_EV_WS_MSG)
  {
    struct mg_ws_message *wm = (struct mg_ws_message *)ev_data;
    mg_ws_send(c, wm->data.ptr, wm->data.len, WEBSOCKET_OP_TEXT);
  }

  (void)fn_data;
}

int main(void)
{
  struct mg_mgr mgr;
  mg_log_set(MG_LL_DEBUG);
  mg_mgr_init(&mgr);
  // mg_http_listen(&mgr, s_http_addr, fn, NULL);     // http
  mg_http_listen(&mgr, s_https_addr, fn, (void *)1); // https
  for (;;)
    mg_mgr_poll(&mgr, 1000);
  mg_mgr_free(&mgr);
  return 0;
}
```

The assembly is like this:

```
make clean all CFLAGS_EXTRA="-DMG_TLS=MG_TLS_OPENSSL -lssl -lcrypto"
```

scaprile, 2023-10-10:
> the upgrade to websocket doesn’t work for me

You'll have to be a bit more specific... may I ask how are you testing it and what does not work ? 
LGTM

MarlikAlmighty, 2023-10-10:
Thanks for answering. I took these two examples:

https://github.com/cesanta/mongoose/tree/master/examples/http-restful-server
https://github.com/cesanta/mongoose/tree/master/examples/websocket-server

And combined them, I have an https server and I'm trying to access websockets through it.

I took your test.html file, opened it via https, but can't get the "upgrade to websockets" to work.

Here's the error I get in the console:

```
./example
331c4d6 3 net.c:195:mg_listen           1 4 https://0.0.0.0:8443
331f16e 3 sock.c:417:accept_conn        2 5 accepted 127.0.0.1:56674 -> 0.0.0.0:8443
331f170 3 tls_openssl.c:74:mg_tls_init  2 Setting TLS
331f17c 3 tls_openssl.c:143:mg_tls_init 2 SSL accept OK
331f1a9 3 tls_openssl.c:155:mg_tls_hand 2 success
331f6f9 3 sock.c:274:read_conn          2 5 snd 0/0 rcv 0/2048 n=928 err=0
331f6f9 3 sock.c:285:write_conn         2 5 snd 2672/4096 rcv 0/2048 n=2672 err=0
331fabe 3 sock.c:274:read_conn          2 5 snd 0/4096 rcv 0/2048 n=866 err=0
331fabe 3 sock.c:274:read_conn          2 5 snd 65/4096 rcv 0/2048 n=-1 err=0
331fabe 3 sock.c:285:write_conn         2 5 snd 65/4096 rcv 0/2048 n=65 err=0
331fabe 3 net.c:148:mg_close_conn       2 5 closed
33209e8 3 sock.c:417:accept_conn        3 5 accepted 127.0.0.1:56690 -> 0.0.0.0:8443
33209e8 3 tls_openssl.c:74:mg_tls_init  3 Setting TLS
33209f7 3 tls_openssl.c:143:mg_tls_init 3 SSL accept OK
3320a0b 3 tls_openssl.c:155:mg_tls_hand 3 success
3320a77 3 sock.c:274:read_conn          3 5 snd 0/0 rcv 0/2048 n=978 err=0
3320a9f 3 sock.c:285:write_conn         3 5 snd 1687/2048 rcv 0/2048 n=1687 err=0
3322c42 3 sock.c:417:accept_conn        4 6 accepted 127.0.0.1:49858 -> 0.0.0.0:8443
3322c42 3 tls_openssl.c:74:mg_tls_init  4 Setting TLS
3322c52 3 tls_openssl.c:143:mg_tls_init 4 SSL accept OK
4097FB69CE7F0000:error:0A00009C:SSL routines:ssl3_get_record:http request:../ssl/record/ssl3_record.c:345:
3322c53 1 event.c:21:mg_error           4 6 tls hs: rc -1, err 1
3322c53 3 net.c:148:mg_close_conn       4 6 closed
```

MarlikAlmighty, 2023-10-10:
[https://ibb.co/x1jpHPM](url)

MarlikAlmighty, 2023-10-10:
[image]

MarlikAlmighty, 2023-10-10:
[image]

scaprile, 2023-10-10:
Do you know that the proper protocol identifier would be `wss://`?

MarlikAlmighty, 2023-10-10:
Yes, sure. Bad screenshot. But that doesn't work either. Sorry for wasting your time on me. I just don’t know who to ask anymore.

MarlikAlmighty, 2023-10-10:
Oooh, it worked!!! Sorry to interrupt you, it worked!!! Thank you very much.

MarlikAlmighty, 2023-10-10:
[image]