cesanta/mongoose #2425: Incorrect Header Validation

kenballus, 2023-10-13:
Mongoose does not sufficiently validate request headers in 2 ways:

1. The RFCs specify that empty header names are invalid. Mongoose should therefore either reject requests with empty header names (as FILL_ME_IN do) or silently drop headers with empty names (as FILL_ME_IN do).
2. The RFCs specify that header names may contain only alphanumerics and a few other unreserved characters (see [RFC 9110 section 5.6.2](https://www.rfc-editor.org/rfc/rfc9110.html#name-tokens)) Mongoose allows the following reserved characters within header names: `"`, `(`, `)`, `,`, `/`, `;`, `<`, `=`, `>`, `?`, `@`, `[`, `\`, `]`, `{`, `}`, and the range `\xc0-\xef`.

cpq, 2023-11-03:
Mongoose already rejects empty header names:
https://github.com/cesanta/mongoose/blob/e418da4bd833510e07b4a99f9dc753232d0fc191/src/http.c#L225

W.r.t. header names containing only certain characters: we do not see any issue with allowing any valid utf-8 character. Mongoose does not strictly follow the RFC, that has never been the goal.

kenballus, 2023-11-03:
> Mongoose already rejects empty header names:

Wow; something must have been wrong with my testing code. Sorry for the hassle.

> we do not see any issue with allowing any valid utf-8 character. Mongoose does not strictly follow the RFC, that has never been the goal.

That's a fair assessment, but you should be aware that `\xc0-\xef` are not valid UTF-8 sequences because they are unpaired surrogates.

cpq, 2023-11-03:
@kenballus that's a fair point about \xc0-\xef, I'll take a look!

cpq, 2023-11-25:
Pushed https://github.com/cesanta/mongoose/commit/bd53e4687377af2c7d56ce69e3af289e59654cb0 to disallow invalid utf-8 in method, uri, header names.
Allow invalid utf-8 in header values and protocol

closed 2023-11-25 by cpq