cesanta/mongoose #2432: wss does not seem to be working in 7.12 (ok up to 7.11)

hdabis1234, 2023-10-21:
- My goal is: upgrade connection from http to websocket
- My actions were: Initiate tls handshake. mg_tls_init is working ok having changed the cert/key pointer
- My expectation was: to get MG_EV_HTTP_MSG after MG_EV_ACCEPT as is the case up to 7.11.
- The result I saw: I seem to be getting the http message in MG_EV_READ, I also get MG_EV_TLS_HS. I do not get the MG_EV_HTTP_MSG event.
- My question is: How am I suppose to call mg_ws_upgrade(conn, mg_hm, NULL); to websocket

# Environment

- mongoose version: 7.12
- Compiler/IDE and SDK: msvc cl
- Target hardware/board: PC
- Connectivity chip/module: x86
- Target RTOS/OS (if applicable): Windows. I am using qt 5.15.10 (qwebsocket) and openssl 3.1.2.

scaprile, 2023-10-22:
It does work #2413 
Please follow the docs and examples in that closed issue. If that does not work for you, please be specific and provide example code.

hdabis1234, 2023-10-23:
The example is exactly the same as the one in the link. It does work from a browser. The only bit that is different is qwebsocket (qt 5.15.10). However same qt version works fine with mongoose up to 7.11.

scaprile, 2023-10-23:
Then sniff your network, understand what the client is doing, read the RFC, and if you determine that there is an actual issue with Mongoose then open a detailed issue and document it so we can reproduce it.

closed 2023-10-23 by scaprile