cesanta/mongoose #2447: Unable close accepted connection immediately

dev31337, 2023-11-02:
- My goal is: To close an accepted connection immediately
- My actions were: Set c->is_closing to 1 in MG_EV_ACCEPT event
- My expectation was: The connection should be closed immediately
- The result I saw: The connection is closed in the next poll (so it depends on the poll timeout)
- My question is: Would is be possible to allow an accepted connection to be immediately closed ? (the goal is to reject DDoS attack as quickly as possible)

# Environment

- mongoose version: 7.12
- Compiler/IDE and SDK: GCC / Linux
- Target hardware/board: PC
- Connectivity chip/module: None
- Target RTOS/OS (if applicable): Linux Ubuntu

The reason it is not disconnected immediately is because of the following mongoose.c code:

void mg_mgr_poll(struct mg_mgr *mgr, int ms) {
...
  if (c->is_readable) accept_conn(mgr, c);
...
 if (c->is_closing) close_conn(c);

The issue is that when you call accept_conn() a brand new accepted connection is created, and this is this connection which receives the is_closing  = 1 in the MG_EV_ACCEPT event.

So when mongoose test the is_closing, it tests the is_closing of the listening connection but not the new accepted connection.

A way to avoid this would be that accept_conn() returns the new accepted connection, and then it would be as easy as doing:

void mg_mgr_poll(struct mg_mgr *mgr, int ms) {
...
  if (c->is_readable) **c =** accept_conn(mgr, c);
...
 if (c->is_closing) close_conn(c);

cpq, 2023-11-02:
@dev31337 I guess we should just set the wait timeout to 0 if close flag is set

Could you check that please? 
```c
    if (can_write(c)) MG_EPOLL_MOD(c, 1);
    if (c->is_closing) ms = 1;  // ADD THIS LINE
    max++;
  }
```

dev31337, 2023-11-02:
Sure, it could work but your code is in a #ifdef section (#elif MG_ENABLE_EPOLL) so it would not work for MG_ENABLE_POLL (nor MG_ENABLE_FREERTOS_TCP), while the code above would work in all cases.

scaprile, 2023-11-02:
I think the point is that we should avoid the delay. I personally don't like substituting `c`, it gets confusing and prone to maintenance-induced errors.

dev31337, 2023-11-02:
Sure as you prefer, I tested your code and it works perfectly and makes an immediate disconnection.

scaprile, 2023-11-02:
@cpq Shall I add that line to every connection poll group ? (FreeRTOS, EPOLL, POLL, SELECT)

cpq, 2023-11-02:
> @cpq Shall I add that line to every connection poll group ? (FreeRTOS, EPOLL, POLL, SELECT)

Yes please!

PR #2451 (2023-11-02): Allow fast close
closes #2447
closed 2023-11-02 by scaprile