cesanta/mongoose #2454: Unable to proxy https://mongoose.ws/ using http-reverse-proxy example

jsm174, 2023-11-03:
- My goal is: Learn how to reverse proxy some endpoints
- My actions were: Use the [http-reverse-proxy](https://github.com/cesanta/mongoose/tree/master/examples/http-reverse-proxy) example
- My expectation was: The reverse proxy would work as expected
- The result I saw: Sometimes content is successfully returned to the browser, other times the browser is stuck with pending requests.
- My question is: 

Since I was unable to use my own endpoints, I decided to hit the mongoose.ws website at `http://mongoose.ws`

As suggested, I attempted to capture the traffic in Wireshark.

I'm using the [http-reverse-proxy](https://github.com/cesanta/mongoose/tree/master/examples/http-reverse-proxy) example verbatim except:

```c++
static const char *s_backend_url = "http://mongoose.ws";
static const char *s_listen_url = "http://0.0.0.0:8000";
```

From the video below, you will see several pending requests that never even appear in Wireshark:

https://github.com/cesanta/mongoose/assets/1197137/ef7ca7f7-358a-46f3-8a8d-d51601e32790

I have double checked that the pending requests are not trying to use ssl, ie:
http://mongoose.ws/images/chat.svg loads just fine in a browser.

# Environment

- mongoose version: 7.12
- Compiler/IDE and SDK: Clang 15.0.0 arm64-apple-darwin22.6.0
- Target hardware/board: _?_
- Connectivity chip/module: _?_
- Target RTOS/OS (if applicable): _?_

scaprile, 2023-11-03:
Please _*attach file captures*_ as indicated

jsm174, 2023-11-03:
The new issue template is not mentioning capture files. Do you want the wireshare capture?

scaprile, 2023-11-03:
You opened a discussion and I think I was clear enough: 
> Please put a network sniffer (Wireshark) on each side of the connection and analyze the traffic.
If you have an actual issue, please open an issue, follow the issue template, and attach the capture files, be specific.

Yeah, I was

jsm174, 2023-11-03:
Yes, again I am sorry.

Here is the capture. I waited, and eventually hit stop in the browser, which is the very last one. 

[http-proxy-ws.pcapng.zip](https://github.com/cesanta/mongoose/files/13254099/http-proxy-ws.pcapng.zip)

I can not capture the mongoose.ws side.

jsm174, 2023-11-03:
FWIW, `info.cern.ch` ends up being just two resources, not several resources.

I just test that on my end and have no issues.

scaprile, 2023-11-03:
> FWIW, `info.cern.ch` ends up being just two resources, not several resources.

As told above, we get Cesanta and Mongoose pages (lots of files) with no issues

What is your network ?, who is who ? What is that capture ? I see responses to ports I haven't seen a connection open before.

Well, as told above, the sample works as is. I'm closing this issue.
I suggest that you build with TLS and retry those sites. If you want to reopen this issue, please capture BOTH sides as requested, and we would gladly appreciate you analyze the traffic and discard problems on your side. Please include ALL the dialogs, start fresh, we need to see every bit of every conversation.

closed 2023-11-03 by scaprile

jsm174, 2023-11-03:
Thanks for your time and sorry for aggravating you. 

> What is your network ?, who is who ? What is that capture ? I see responses to ports I haven't seen a connection open before.

I shut down everything on my machine, started the app, captured HTTP in Wireshark, opened up an incognito window in chrome, and went to localhost:8000, waited, and then stopped Wireshark, and then sent to you. 

I rebuilt the app in Ubuntu, and same thing. I even tethered my phone to be off my network.

I'll keep working on it..

jsm174, 2023-11-04:
@scraprile - Worked on this some more, and I finally found the issue!

The tutorial website points to [7.7](https://github.com/cesanta/mongoose/blob/7.7/examples/http-reverse-proxy/main.c#L47-L69) which is missing a line of code compared to master:

``` 
     c->is_resp = 0;  // process further msgs in keep-alive connection
```

[image]

Permalink to master:

https://github.com/cesanta/mongoose/blob/13259f8902ebb7b8eb18724f474c899ef8df613d/examples/http-reverse-proxy/main.c#L68C21-L68C21

Any chance the tutorial website could be updated?

scaprile, 2023-11-04:
Done. Always check current example in lastest release