cesanta/mongoose #2456: Realtime Data with Web Sockets

lplassman, 2023-11-04:
Now that pipe functionality is depreciated when sending data using web sockets, I was wondering if there is any alternative? I looked through the documentation, and all the solutions there appear to be polling only, with no other way to wake up the main event thread to actually send or receive the data. I am working on an application that sends realtime audio and video data over a web socket connection, and previously the pipe feature worked great for this type of application.

scaprile, 2023-11-04:
Can you please explain what is "pipe functionality when sending data using web sockets" ?
We have deprecated the back to back socket pipe and provided a queue instead, as this has been created for multithreaded applications.
Can you please ellaborate on your use case and what you can't do, why, and why you need it ?

lplassman, 2023-11-06:
I just meant pipe functionality was depreciated in general. I was using it to trigger sending to a web socket.
Currently, I have a multi-threaded application which processes audio and video frames, and then, using the new queue functionally, queues it into the main mongoose event process to send to the client over web sockets. Previously, the thread that processes audio and video frames would be able to wake up the main mongoose thread using the pipe. Now, as far as I can tell, there is no way to wake up the mongoose thread from another thread, and I instead have to rely on the polling feature.

scaprile, 2023-11-06:
OK, I think I figured out what you are doing and what you need.
For some reason you can't make the poll interval smaller, and for some reason there is no traffic that would wake up the manager, so the data you've put in the queue to be sent as a response to an HTTP request stays there longer than you'd like. With the back-to-back pipe, as it was built around sockets, you got an immediate wake up when data was placed in the pipe.
Please allow us to discuss internally if there is something we can do.

cpq, 2023-11-25:
@lplassman Luke, thank you.

The solution is to create a socket pair yourself!
Take a look at https://github.com/cesanta/mongoose/commit/122115dbea0735f3c16303599d60c8058d60589a , that's a modified multi-threaded example that creates a helper UDP socket pair and wakes up mongoose thread after sending stuff to the queue.

commit 122115db (2023-11-25): Fix #2456 - wake up mongoose thread
commit 30738f66 (2023-11-25): Merge pull request #2493 from cesanta/wake

Fix #2456 - wake up mongoose thread
closed 2023-11-25 by scaprile