cesanta/mongoose #2457: Issue with timer set up by mg_timer_add / quickly using all the CPU

dev31337, 2023-11-04:
- My goal is: Having a timer trigger regularly
- My actions were: Set up the timer and re-arm it every time
- My expectation was: Nothing special should happen
- The result I saw: Mongoose is using more and more CPU until using all the available CPU
- My question is: Why does it happen ?

# Environment

- mongoose version: 7.12
- Compiler/IDE and SDK: Gcc / Linux
- Target hardware/board: PC
- Connectivity chip/module: None
- Target RTOS/OS (if applicable): Ubuntu

Here is a sample code:

```
static void cb_timer(void *fn_data)
{
    struct mg_mgr *mgr = (struct mg_mgr *)fn_data;
    mg_timer_add(mgr, 10, MG_TIMER_RUN_NOW, cb_timer, mgr);
}

int main(void)
{
    int done = 0;
    struct mg_mgr mgr;
    mg_mgr_init(&mgr);
    mg_timer_add(&mgr, 10, MG_TIMER_RUN_NOW, cb_timer, &mgr);
    while(!done) mg_mgr_poll(&mgr, 5);
    mg_mgr_free(&mgr);
    exit(0);
}
```

If you run this sample code and monitor it with "top" you will see that the process will use more and more CPU every second until using all the available CPU.

It seems that there is a bug with the timer handling by mongoose.

dev31337, 2023-11-04:
It seems the issue is in the function "mg_timer_poll".

I think it should call mg_timer_free() but it seems that it does not.

mg_timer_free() is defined in mongoose but never called anywhere.

dev31337, 2023-11-04:
The code in "mg_timer_poll" is also wrong because if you set a timer with MG_TIMER_RUN_NOW for example with an expiration of 1 second, the timer is _not_ triggered in 1 second but in X milliseconds, X being the poll timer.

The reason is because the code in "mg_timer_poll" does not check the "expired" status of the timer as long as it is a "once" timer:

`if (!once && !expired) continue;`

scaprile, 2023-11-04:
Please follow the guidelines in the [documentation](https://mongoose.ws/documentation),  [examples](https://github.com/cesanta/mongoose/tree/master/examples) and [tutorials](https://mongoose.ws/tutorials/) available.

From https://mongoose.ws/documentation/tutorials/timers/ :

> It is important to keep the event manager polling timeout smaller than the timer interval, because timers are checked at every polling cycle. If there are no network events, the poll function would sleep for the whole polling interval and if it is larger than the timer timeout, a timer invocation could be missed.

dev31337, 2023-11-04:
I of course already knew and followed this guideline, as you can see in my second sample code the polling timeout is 100 ms while the timer interval is 1000 ms. (the first sample code the polling timeout is 5 ms while the timer interval is 10 ms)

Just try any sample code and you will see the issue(s).

Personally I stopped using the timer of mongoose, so it's up to you to decide if you want to fix it or ignore it.

scaprile, 2023-11-05:
I don't think adding a tjmer inside a callback function is something we suggest in the docs. I don't claim it can't be done, just that it was not designed that way.
That is the reason why we did not check your samples.
I beg to differ on your appreciations of the code.

closed 2023-11-05 by scaprile