cesanta/mongoose #2460: Improper handling of empty list elements in `Transfer-Encoding` header values

kenballus, 2023-11-06:
Mongoose doesn't properly handle `Transfer-Encoding` header values with empty list elements. `Transfer-Encoding` is a list-valued header, and RFC 9110 says this about list-valued header parsing:
> A recipient MUST parse and ignore a reasonable number of empty list elements: enough to handle common mistakes by senders that merge values, but not so much that they could be used as a denial-of-service mechanism.

Thus, `Transfer-Encoding: ,chunked` should be equivalent to `Transfer-Encoding: chunked`. Mongoose does not make this distinction, and is thus vulnerable to request smuggling when deployed behind gateway servers that do (and also don't normalize out the comma). There are a few widely-deployed load balancers that exhibit this behavior.

# Environment
- mongoose version: `master`
- Compiler/IDE and SDK: GCC 12.2.1
- Platform: Linux x86_64

kenballus, 2023-11-06:
You can reproduce this bug by starting a Mongoose HTTP server and sending it the follow request:
```
POST / HTTP/1.1\r\n
Host: whatever\r\n
Transfer-Encoding: ,chunked\r\n
Content-Length: 5\r\n
\r\n
0\r\n\r\n
```

If you log the received request's message body, it will be incorrectly reported as `0\r\n\r\n` instead of empty.

cpq, 2023-11-25:
@kenballus thank you!

Is is correct to say that it is `ZERO_OR_MORE_CHARACTERS,chunked` ?
Or strictly `,chunked` ?

kenballus, 2023-11-25:
Sorry; I'm not sure I understand what you're asking.

The standard says these two things:

1. Empty list elements should be ignored. Thus, `Transfer-Encoding: ,chunked` and `Transfer-Encoding: , ,,\t , chunked, , , ` should be equivalent to `Transfer-Encoding: chunked`.
2. Proxies must strip empty elements out of list-valued header values.

Mongoose does not follow (1). When Mongoose is paired with a proxy server that follows (1) but not (2), there is the ability to bypass security policies implemented by the proxy.

The simplest fix for Mongoose is probably to strip commas, spaces, and tabs from either side of the `Transfer-Encoding` header value, then compare what's left (case-insensitive) to `"chunked"` when determining whether a message has a chunked body.

cpq, 2023-12-01:
Thank you.
I am just trying to figure out what's the best algorithm should be.

I think, this one:
1. Assume that the header value is a comma separated list of items
2. Iterate over items
3. For each item, strip off trailing and leading whitespaces
4. Compare the stripped value with `chunked` in a case-insensitive way

Does that sound correct?

kenballus, 2023-12-01:
I think that won't work because chunk-extension values can contain commas within quoted strings. Thus, splitting out the list of items will require implementing something like the full `Transfer-Encoding` ABNF rule from the RFC.

I think the easiest thing to do is just respond 400 to all `Transfer-Encoding` headers with values other than `chunked`, case-insensitive.

commit 4c445453 (2023-12-07): Fix #2460 - sanity check of the Transfer-Encoding: chunked header
commit dbc7b5e7 (2023-12-07): Merge pull request #2509 from cesanta/te

Fix #2460 - sanity check of the Transfer-Encoding: chunked header
closed 2023-12-01 by cpq