cesanta/mongoose #2479: 404 Page Problem

ffreality, 2023-11-17:
- My goal is:
I am doing some experiments about Unreal Engine 5 Async HTTP Server creation with mongoose and FRunnableThreads.
Plugin exposes HTTP message to a blueprint delegate. So, we can parse it and send reply with blueprints without touching C/C++ code. Currently HTTP Static Content Serving and Rest API features are working perfectly.

- My Problem:
When I define a 404 page, I can send API requests but can't access root directory contents. `http://localhost:8081` gives me `Index Of... Mongoose 7.1.2` page and when I want to open directly a .html file, it directs me 404 page.

- My actions were: 
1- I copied mongoose.c and .h to my Unreal Engine 5 plugin's source folder and do necessary binding operations.
2- My code : https://gist.github.com/ffreality/d0428ba3e1e021d82cf4245e36c14586#file-ue_mongoose-cpp-L188

# Environment
- mongoose version: Master from 17th November 2023
- Compiler/IDE and SDK: Visual Studio 2022
- Target hardware/board: Windows x64

scaprile, 2023-11-17:
Please follow the guidelines in the [documentation](https://mongoose.ws/documentation),  [examples](https://github.com/cesanta/mongoose/tree/master/examples) and [tutorials](https://mongoose.ws/tutorials/) available.

Sorry, I won't go fetch your code and check it. Please see the examples and tutorials and if you have specific questions, here we are.

closed 2023-11-17 by scaprile

ffreality, 2023-11-17:
Please read what write again but read it slow.
Did I ask you anything about threads ? and my project is about implementing mongoose to Unreal Engine.
Your thread examples are won't give any results for us.

Question is simple. In your example, you just defined .page404 and thats all. But it doesn't work.

```
struct mg_http_serve_opts HTTP_Options;
memset(&HTTP_Options, 0, sizeof(mg_http_serve_opts));					
HTTP_Options.root_dir = TCHAR_TO_UTF8(*Owner->Server_Path_Root);
HTTP_Options.page404 = TCHAR_TO_UTF8(*Owner->Server_Path_404);
```

scaprile, 2023-11-17:
> Question is simple. In your example, you just defined .page404 and thats all. But it doesn't work.

I beg to differ.
This is the http-server example code (s_root_dir ="." ) :
https://github.com/cesanta/mongoose/blob/6386cdff5037280e069e17dc456c93fd6003ca9e/examples/http-server/main.c#L26-L28

the doc: https://mongoose.ws/documentation/#struct-mg_http_serve_opts
turn former code above to:
```c
    opts.root_dir = s_root_dir;
    opts.ssi_pattern = s_ssi_pattern;
    opts.page404 = "my404.html";
    mg_http_serve_dir(c, hm, &opts);
```
```sh
$ make
cc main.c mongoose.c        -W -Wall -Wextra -g -I.   -DMG_HTTP_DIRLIST_TIME_FMT="%Y/%m/%d %H:%M:%S" -DMG_ENABLE_LINES=1 -DMG_ENABLE_IPV6=1 -DMG_ENABLE_SSI=1  -o example                                    
./example                    
1987610 2 main.c:97:main                Mongoose version : v7.12
1987610 2 main.c:98:main                Listening on     : http://0.0.0.0:8000
1987610 2 main.c:99:main                Web root         : [/.../mongoose/examples/http-server]
1988f57 2 main.c:35:cb                  GET /whatever 200 76
```
```sh
$ curl http://localhost:8000/whatever
<html><head>404</head><body><H1>This is a custom 404 page</H1></body></html>
$
```
```sh
$ ls
example  main.c  Makefile  mongoose.c  mongoose.h  my404.html  README.md
$ cat my404.html 
<html><head>404</head><body><H1>This is a custom 404 page</H1></body></html>
```