cesanta/mongoose #2482: "mg_json_unescape" does not unescape valid json

NETertainer, 2023-11-19:
Simple test:
```
test.c:
#include "mongoose.h"
#include <stdio.h>

int main() {
   char *test = "{\"1\": \"a\\\\nb\"}";
   char* temp = mg_json_get_str(mg_str(test), "$.1");
   printf("%s\n%s\n\n", test, temp);

   test = "{\"1\": \"a\\nb\"}";
   temp = mg_json_get_str(mg_str(test), "$.1");
   printf("%s\n%s\n\n", test, temp);

   return 0;
}
```

Output:
```
$ gcc mongoose.c test.c
$ ./a.out
{"1": "a\\nb"}
a\nb

{"1": "a\nb"}
a
b
```

As reference, output of JSON.parse:
```
console.log(JSON.parse('{ "1": "a\\nb" }')[1])
a
b

console.log(JSON.parse('{ "1": "a\nb" }')[1])
Uncaught SyntaxError: Bad control character in string literal in JSON at position 9 (line 1 column 10)
```

Quick fix for parsing: 
```
mongoose.c:
3195,3196c3195,3196
<     } else if (s.ptr[i] == '\\' && i + 1 < s.len) {
<       char c = json_esc(s.ptr[i + 1], 0);
---
>     } else if (s.ptr[i] == '\\' && s.ptr[i + 1] == '\\' && i + 2 < s.len) {
>       char c = json_esc(s.ptr[i + 2], 0);
3199c3199
<       i++;
---
>       i += 2;
```

New output:
```
$ ./a.out
{"1": "a\\nb"}
a
b

{"1": "a\nb"}
a\nb
```

I guess this affects "mg_pass_string", too.

scaprile, 2023-11-19:
Would you be so kind to fill the issue template ? I'm sure this will be a great reading once you decode what you want to do and what you got, for me, so I don't have to infer that from your code and your outputs. If this is our bug, we'll fix it sooner. Thank you.

NETertainer, 2023-11-19:
Sorry for being lazy :smile:. Updated.

scaprile, 2023-11-19:
The first test string seems to be the escape of an already escaped `\n` , that is `\\n`.
`"{\"1\": \"a\\\\nb\"}"` is C for `{"1": "a\\nb"}`. The double backslash is unescaped and the string returned as `{"1": "a\nb"}`.

The second test string to be parsed contains a backslash and its corresponding character.
`"{\"1\": \"a\\nb\"}"` is C for `{"1": "a\nb"}`, that is what `mg_json_unescape()` sees and correctly processes, so the output would be {"1": "a<newline char>b"}
https://github.com/cesanta/mongoose/blob/6386cdff5037280e069e17dc456c93fd6003ca9e/src/json.c#L292-L293
See real stuff with `mg_hexdump()"

```c
   char *test = "{\"1\": \"a\\\\nb\"}";
   char* temp = mg_json_get_str(mg_str(test), "$.1");
   MG_INFO(("In:"));
   mg_hexdump(test, strlen(test));
   MG_INFO(("Out:"));
   mg_hexdump(temp, strlen(temp));
   

   test = "{\"1\": \"a\\nb\"}";
   temp = mg_json_get_str(mg_str(test), "$.1");
   MG_INFO(("In:"));
   mg_hexdump(test, strlen(test));
   MG_INFO(("Out:"));
   mg_hexdump(temp, strlen(temp));
```
```sh
$ ./fr13 
4d1fb5a 2 fr13.c:7:main                 In:
0000   7b 22 31 22 3a 20 22 61 5c 5c 6e 62 22 7d         {"1": "a\\nb"}  
4d1fb5a 2 fr13.c:9:main                 Out:
0000   61 5c 6e 62                                       a\nb            
4d1fb5a 2 fr13.c:15:main                In:
0000   7b 22 31 22 3a 20 22 61 5c 6e 62 22 7d            {"1": "a\nb"}   
4d1fb5a 2 fr13.c:17:main                Out:
0000   61 0a 62                                          a.b             
```

What I think you are observing is your JavaScript console processing the first \n and escaping it for you when printing it.

I'd like to see your actual use case and whether this has an actual problem that needs to be solved.

NETertainer, 2023-11-19:
What I was trying to point out is that `"a\nb"` is not valid json and the \n should not lead to being parsed as new line, while `"a\\nb"` is valid json (double escaping is strictly required) but is not unescaped at all.

```
JSON.parse('"a\nb"')
Uncaught SyntaxError: 

JSON.parse('"a\\nb"')
'a\nb'
```
I only used console.log to visualize the new line when the valid json gets unescaped in e.g. a browser.

So the current implementation just does not work with valid json and additionally accepts invalid json that can not be used otherwise.

NETertainer, 2023-11-19:
Sorry for bothering. I misinterpreted the output of the browser console. You are totaly right and I got it wrong.

closed 2023-11-19 by NETertainer