cesanta/mongoose #2483: Open file with fat file system give invalid state usage

adrien-cardinale, 2023-11-20:
I try to open a file with mg_fs_open() when f_open is called the hard fault ISR is triggered with invalid state usage. But if I call f_open from main I don't have fault ISR. 

I use `struct mg_fd *fd = mg_fs_open(&mg_fs_fat, "index.htm", MG_FS_READ);` for open file with mongoose.

# Environment

- mongoose version: 7.12
- Compiler/IDE and SDK: Code compose studio TI v20.2.7.LTS
- Target hardware/board: TI f28388d
- Connectivity chip/module: 
- Target RTOS/OS (if applicable): FreeRTOS

scaprile, 2023-11-20:
From "hard fault" and "TI f28388d" I guess you are talking about some ARM chip, maybe an MSP432 ?
Then for "open file in FAT filesystem" I guess you have a file system in flash or some other support, is it read only ? is it read/write ?
Then for "FreeRTOS" and "Code compose studio TI v20.2.7.LTS", what should I infer ? Who is providing FAT ? Who translates your f_open() call to actual hardware access ? How did you configure Mongoose ? What compile options did you use ?
How do you call mg_fs_open() ? What is your simplest piece of code that will reproduce the problem ? Are you calling ALL mg_* functions from one and only one and the same context ?

adrien-cardinale, 2023-11-20:
The chip is a f28388d from texas instrument with a cortex m4. The filesystem is  Generic FAT Filesystem Module with a sd card who is in read and write. I call mg_fs_open with `struct mg_fd *fd = mg_fs_open(&mg_fs_fat, "index.htm", MG_FS_READ);` in main without task.

adrien-cardinale, 2023-11-20:
Do you have a working example with a fat file system?

scaprile, 2023-11-20:
Googling for "f28388d" returns pointers to TMS320F28388D, which is a C2000, a dual core DSP with C28x cores; not a Cortex-M4.
All our examples are "working examples", please give proper details on how you are using Mongoose, we have never used it on a DSP chip and we don't know how you configured it. Given that what you say it is is not what Google says it is, I should also ask for how you configured FreeRTOS.
https://mongoose.ws/documentation/#filesystem
If we can´t infer what could be wrong from your setup and your answers, you will have to debug your hard fault and get information from that.

adrien-cardinale, 2023-11-20:
TMS320F28388D have a cortex-m4 for communication you can look at page 4 here: https://www.ti.com/lit/ds/symlink/tms320f28388d.pdf?ts=1700454591882&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FTMS320F28388D.
I try to make a http server with FreeRTOS, FreeRTOS-TCP and mongoose and I would like store the web file in a sd card. Now I try to make a minimal example for read and write in the sd card with mongoose mg_fs* function. I have already a driver for communicate with sd card from f28388d and the filesysteme basic fonction works well. For mongoose I have use these options:
- MG_ENABLE_FATFS=1
- MG_ARCH=MG_ARCH_FREERTOS

scaprile, 2023-11-20:
Let me summarize the scenario.
You have a Cortex-M4 core, FreeRTOS, some FAT driver you don't tell, an SD card, and Mongoose.
You call mg_fs_open() and whatever runs after this
https://github.com/cesanta/mongoose/blob/6386cdff5037280e069e17dc456c93fd6003ca9e/src/fs_fat.c#L75
triggers a HardFault.
I that is correct, I suggest you inspect that function call, determine where and why that fault is triggered, what is that function expecting that is not given, whether there is some prior initialization that is not being done, etc., and come back to us with that information.

adrien-cardinale, 2023-11-21:
The file systeme is  FatFs - Generic FAT Filesystem module  R0.13a with Texas instrument driver for sd. I have resolved the issue. The struct FIL was declared with 2 different definition in the fat library. ´#define FF_FS_EXFAT 0´ solve the issue

closed 2023-11-21 by adrien-cardinale

scaprile, 2023-11-21:
And I guess that has nothing to do with Mongoose.