cesanta/mongoose #2498: How to use mongoose in Arduino Opta Lite

tdgunes, 2023-11-28:
- My goal is: Use mongoose with Arduino Opta Lite board
- My actions were: I tried to adapt some examples from STM32 Nucleo and Arduino W5500 examples. 
- My expectation was: It'd be easy to adapt the examples. 
- The result I saw: I was able to get some examples compiled. (Basic HTTP echo server) However, I couldn't able to get the TCP/IP part working, as there is not many examples from Arduino Opta Lite regarding low level actions.
- My question is: Am I missing something easy to do, or is it actually a hard task? 

# Environment

- mongoose version: latest
- Compiler/IDE and SDK: Arduino IDE with mbed_opta 4.0.10
- Target hardware/board: Arduino Opta Lite
- Connectivity chip/module: OPTA's STM32H747
- Target RTOS/OS (if applicable): I guess there is either FreeRTOS or ARM's mbed, already included.

# Example code:

```c
#include "mongoose.h"

struct mg_mgr mgr;  // Mongoose event manager

struct mg_tcpip_driver_stm32_data driver_data = {.mdc_cr = 4};
struct mg_tcpip_if mif = {.mac = {2, 0, 1, 2, 3, 5}};  // network interface

void setup() {
  Serial.begin(115200);
  while (!Serial);

  // Set logging function to a serial print
  Serial.print("Booting...");
  mg_log_set_fn([](char ch, void *) { Serial.print(ch); }, NULL);
  mg_mgr_init(&mgr);

  delay(3000);
  MG_INFO(("Starting TCP/IP stack..."));

  mif.driver = &mg_tcpip_driver_stm32;
  mif.driver_data = &driver_data;
  mg_tcpip_init(&mgr, &mif);

  // Start a 5 sec timer, print status message periodically
  mg_timer_add(
      &mgr, 5000, MG_TIMER_REPEAT,
      [](void *) {
        MG_INFO(("ethernet: %s", mg_tcpip_driver_stm32.up(&mif) ? "up" : "down"));
      },
      NULL);

  // Setup HTTP listener. Respond "ok" on any HTTP request
  mg_http_listen(
      &mgr, "http://0.0.0.0",
      [](struct mg_connection *c, int ev, void *ev_data, void *fn_data) {
        if (ev == MG_EV_HTTP_MSG) mg_http_reply(c, 200, "", "ok\n");
      },
      &mgr);
}

void loop() {
  mg_mgr_poll(&mgr, 1);
}
```

To get it compiling, I got this part commented out (mongoose.c lines from 9297-9319): 

```
/ void ETH_IRQHandler(void);
// void ETH_IRQHandler(void) {
//   if (ETH->DMASR & MG_BIT(6)) {           // Frame received, loop
//     ETH->DMASR = MG_BIT(16) | MG_BIT(6);  // Clear flag
//     for (uint32_t i = 0; i < 10; i++) {   // read as they arrive but not forever
//       if (s_rxdesc[s_rxno][0] & MG_BIT(31)) break;  // exit when done
//       if (((s_rxdesc[s_rxno][0] & (MG_BIT(8) | MG_BIT(9))) ==
//            (MG_BIT(8) | MG_BIT(9))) &&
//           !(s_rxdesc[s_rxno][0] & MG_BIT(15))) {  // skip partial/errored frames
//         uint32_t len = ((s_rxdesc[s_rxno][0] >> 16) & (MG_BIT(14) - 1));
//         //  printf("%lx %lu %lx %.8lx\n", s_rxno, len, s_rxdesc[s_rxno][0],
//         //  ETH->DMASR);
//         mg_tcpip_qwrite(s_rxbuf[s_rxno], len > 4 ? len - 4 : len, s_ifp);
//       }
//       s_rxdesc[s_rxno][0] = MG_BIT(31);
//       if (++s_rxno >= ETH_DESC_CNT) s_rxno = 0;
//     }
//   }
//   // Cleanup flags
//   ETH->DMASR = MG_BIT(16)    // NIS, normal interrupt summary
//                | MG_BIT(7);  // Clear possible RBUS while processing
//   ETH->DMARPDR = 0;          // and resume RX
// }
```

The `mongoose_custom.h` file, I tested:

```
#pragma once

#include "Arduino.h"

#include <errno.h>
#include <stdarg.h>
#include <stdio.h>
#include <time.h>

#include "mbedtls_config.h"

#define MG_ARCH MG_ARCH_CUSTOM
#define MG_ENABLE_SOCKET 0
#define MG_ENABLE_TCPIP 1
#define MG_ENABLE_FILE 0           // Disable POSIX filesystem
#define MG_ENABLE_MBEDTLS 1
#define MG_ENABLE_DRIVER_STM32 1

#define mkdir(a, b) (-1)
#define MG_IO_SIZE 128

#define MG_TLS MG_TLS_MBED
#define MG_ENABLE_PACKED_FS 1
```

scaprile, 2023-11-28:
The simplest response I can give you you is that you have to know what you are doing and why you are doing it.
The example does not freeze your board, you are downloading code that is not for that board and it won't work. If you have an STM32H chip, don't download code for an STM32F to it.
The W5500 driver is for a W5500 chip connected via SPI, if you don't have that, it obviously won't work

Please follow the guidelines in the [documentation](https://mongoose.ws/documentation),  [examples](https://github.com/cesanta/mongoose/tree/master/examples) and [tutorials](https://mongoose.ws/tutorials/) available.

https://mongoose.ws/documentation/#nucleo-h743zi2
https://mongoose.ws/documentation/tutorials/stm32/all-make-baremetal-builtin/#compilation-options
NUCLEO-H723ZG 	-DMG_ENABLE_DRIVER_STM32**H**=1
NUCLEO-H743ZI 	-DMG_ENABLE_DRIVER_STM32**H**=1
NUCLEO-H743ZI2 	-DMG_ENABLE_DRIVER_STM32**H**=1
That is the closest to your chip, whether it will work or not, we don't know, we don't have examples for your microcontroller nor your board

closed 2023-11-28 by scaprile

scaprile, 2023-11-28:
This is not a forum to share experiences, you opened an issue. An issue means there is a problem to solve.
There are no public forums for Mongoose, the closest you may have is a discussion https://github.com/cesanta/mongoose/discussions
Embedded is not trial and error, if there is no example for a microcontroller, you have to know what you are doing at the hardware level, which MAC controller the microcontroller has, which PHY the board has, read datasheets and user manuals, debug or even write a driver. For OTA, add to that: flash memory structure and capabilities.
We don't support your microcontroller.