cesanta/mongoose #2512: __sync_synchronize not defined in older GCC versions

danielfrankcom, 2023-12-05:
- My goal is:
    - I want to successfully compile Mongoose using an older g++ version for a legacy project, in order to facilitate the upgrade of that project gradually to a newer compiler version.
- My actions were:
  - Download Mongoose 7.12 source code
  - Attempt to compile using `g++ -c mongoose.c`
- My expectation was:
  - Successful compilation of `.o` file
- The result I saw: 

Compilation failure:
```
g++ -c /tmp/mongoose-7.12/mongoose.c:
/tmp/mongoose-7.12/mongoose.c: In function 'size_t mg_queue_read_len(mg_queue*)':
/tmp/mongoose-7.12/mongoose.c:5849: error: '__sync_synchronize' was not declared in this scope
/tmp/mongoose-7.12/mongoose.c: In function 'void mg_queue_write_len(mg_queue*, size_t)':
/tmp/mongoose-7.12/mongoose.c:5858: error: '__sync_synchronize' was not declared in this scope
```
- My question is:

Can the definition of `MG_MEMORY_BARRIER` check the `__GNUC__` version before attempting to use `__sync_synchronize()`?

The current check is the following:

```c
#if defined(__GNUC__) || defined(__clang__)
```

This is the only barrier to the compilation succeeding, since the current code only checks for the presence of `__GNUC__`, and not the version. I am attempting to compile using g++ 3.2.3 (it's old, I know) and I am able to get everything to work by simply adding a version check like this:

```c
#if (defined(__GNUC__) && (__GNUC__ > 4) || (defined(__GNUC__) && defined(__GNUC_MINOR__) && __GNUC__ == 4 && __GNUC_MINOR__ >= 1)) || defined(__clang__)
```

I have no expectation of ongoing support for such an old compiler version, but the current code happens to be compatible aside from this one small issue.

I have verified that the `__sync_synchronize()` call was added in GCC 4.1.0, and is not available before that. I've provided a `Dockerfile` at the bottom of this request which can be used to test changes before/after support for `__sync_synchronize()` was added.

I am happy to create a PR to modify this definition check if the proposed change is acceptable.

# Environment

- Mongoose version: 7.12
- Compiler/IDE and SDK: GNU g++ version 4.0.4
- Target hardware/board: x86

# Reproduction steps

Any GCC version before 4.1.0 will exhibit this issue. Below is a `Dockerfile` which showcases the issue and allows it to be reproduced as long as you have Docker installed.

Put these lines in a `Dockerfile`, and run `docker build .`

The Docker build will fail when using GCC 4.0.4, but will succeed and produce a `.o` file when using GCC 4.1.0. The environment variable near the top of the file can be modified to change the GCC version.

```Dockerfile
FROM centos:7

# Uncomment one of these to select a GCC version.
ENV GCC_VERSION 4.0.4
#ENV GCC_VERSION 4.1.0

RUN yum install -y \
    binutils \
    bzip2 \
    glibc-devel \
    glibc-devel.i686 \
    make \
    patch \
    unzip \
    wget

RUN rpm -i https://kojipkgs.fedoraproject.org/vol/fedora_koji_archive01/packages/compat-gcc-34/3.4.6/46.fc26/x86_64/compat-gcc-34-3.4.6-46.fc26.x86_64.rpm
RUN ln -s /usr/bin/gcc34 /usr/bin/cc

WORKDIR /tmp
RUN wget --no-check-certificate https://mirrorservice.org/sites/sourceware.org/pub/gcc/releases/gcc-${GCC_VERSION}/gcc-${GCC_VERSION}.tar.bz2
RUN tar -xvf gcc-${GCC_VERSION}.tar.bz2
WORKDIR /tmp/gcc-${GCC_VERSION}

# Patch the GCC source to fix a minor incompatibility with CentOS 7.
RUN wget https://gist.githubusercontent.com/Wheest/bdd5bc7d487061e5c973f0fba4b96c47/raw/31033a6c5e8d7811ae0a81c2e4c9adea2359a822/unwind.patch -O /tmp/unwind.patch
RUN patch -p0 < /tmp/unwind.patch

RUN mkdir /tmp/gcc-${GCC_VERSION}/build
WORKDIR /tmp/gcc-${GCC_VERSION}/build
RUN ../configure --enable-languages=c,c++

RUN make
RUN make install

RUN wget https://github.com/cesanta/mongoose/archive/refs/tags/7.12.tar.gz -O /tmp/7.12.tar.gz
RUN tar -xvzf /tmp/7.12.tar.gz -C /tmp

RUN g++ -c /tmp/mongoose-7.12/mongoose.c

```

cpq, 2023-12-22:
@danielfrankcom Yes please make a PR.

Nitpick: in your check, one `defined(__GNUC__)` seems to be redundant.
Please sign https://cesanta.com/cla.html before we can integrate the PR.

cpq, 2023-12-28:
Apparently I accidentally pushed that change in https://github.com/cesanta/mongoose/commit/01b612cff50133ae68c1e8cb6bdccc94189c80f0#diff-e5e80dd7794e56fd96c9b5c781b500dda60555b7f8e68d7a5d7d3fa9384db6f2

Please verify that it does the job, and reopen this issue if it does not
Thank you for reporting it and suggesting the change!

closed 2023-12-28 by cpq